Koenkk/zigbee2mqtt #25529: Add support for restore state on Aurora AU-A1ZBDSS

karlc28, 2025-01-04:
### Is your feature request related to a problem? Please describe

Currently when there is a power cut and the power is restored, even if for a second the socket changes to the off setting which means I have to physically go and turn it on (My Nuc running HA is plugged into one) Also the brightness of the LED defaults back to maximum

### Describe the solution you'd like

I would like the option for the previous state to be restored. So if the socket was on when power was lost, it is on when the power is restored.
Also if the same could be applied to the led brightness with the previous setting being restored 

### Describe alternatives you've considered

The only alternate I can do is change where I plug HA into

### Additional context

[image]

Pel1can111, 2025-02-20:
I have just set up an automation in home assistant to check socket status every hour and revert switch/brightness back on sockets that remain on permanently. I also have my solar in HA that reports grid status so use the grid power returning as an extra trigger.

As far as I'm aware these sockets do not have the ability to restore to previous status so any solution would need to be run within HA or zigbee2mqtt.

JonDunsdon, 2025-02-21:
Not sure how this would work.
If the HA instance is running on that plug it would be off, so wouldn’t be able to reset the plug. If you have a UPS on the plug running the HA instance (NUC) then you could write an automation to double toggle the switch - it’s double as the plugs report last state (so on) even though it’s actually off …

closed 2025-05-01 by github-actions[bot]