Koenkk/zigbee2mqtt #25532: Device error message after update to 2.0. Status 'UNSUPPORTED_ATTRIBUTE

Sabjul33, 2025-01-04:
### What happened?

After the V2 update, i have an error message for a device. But it seems to be work (i only use single tap and two tap)
Message below : 

error 2025-01-04 22:54:01z2m: Failed to configure 'SMART_BUTTON_SALLE_A_MANGER', attempt 1 (Error: ZCL command 0xa4c13831f7219368/1 genOnOff.write({"tuyaOperationMode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Endpoint.write (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:9) at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/moes.ts:558:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17))

With the V 1.42 ihaven't this message.

Device infos : 
Type de périphérique
EndDevice
Modèle Zigbee
TS004F
Fabricant Zigbee
_TZ3000_ja5osu5g
Description
Smart button

Fabriquant
[Loginovo](https://www.zigbee2mqtt.io/supported-devices/#v=Loginovo)
Modèle
[ZG-101ZL](https://www.zigbee2mqtt.io/devices/ZG-101ZL.html#loginovo-zg-101zl)

I don't remember but I wonder if the manufacturer and model are correct. Won't it have changed with the transition to V2?

the link of the product to aliexpress : 

https://fr.aliexpress.com/item/1005006762476931.html?spm=a2g0o.order_list.order_list_main.76.35ab5e5bq3iWBN&gatewayAdapt=glo2fra

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

ember 7.4.3

### Adapter

 EmberZNet SonOff Usb dongle

### Setup

Z2M Dedicated debian VM

Damianod27, 2025-01-05:
@Sabjul33 You probably need to adjust your settings. Read the Z2M V. 2.0 release notes regarding buttons.

Sabjul33, 2025-01-05:
> @Sabjul33 You probably need to adjust your settings. Read the Z2M V. 2.0 release notes regarding buttons.

Hi, where can i find this informations ? I can't find them.
Thanks,

mrmaximas, 2025-01-05:
@Sabjul33 https://github.com/Koenkk/zigbee2mqtt/discussions/24198

Sabjul33, 2025-01-05:
> @Sabjul33 You probably need to adjust your settings. Read the Z2M V. 2.0 release notes regarding buttons.

I could be interested by some help because when i read the breaking changes, i can't seesomethings about this and i've done the modifications of configuration.yaml before the update. I really wonder if the module is well recognized.
It maybe like this : https://www.zigbee2mqtt.io/devices/IH-K663.html

CRCinAU, 2025-01-05:
I got hit by this too..... How do I then go about using these new things in NodeRed?

I used to use `sensor.front_doorbell_action` - but now that sensor isn't published from Z2M anymore.

Sabjul33, 2025-01-05:
I add a new module in my installation a [ZBMINIR2](https://www.zigbee2mqtt.io/devices/ZBMINIR2.html#sonoff-zbminir2) and when i add it i have the error too and it's seems to work.

error 2025-01-05 17:09:31z2m: Failed to configure 'LAMPE_ALLEE_VOITURES', attempt 3 (Error: ZCL command 0x7cc6b6fffe9cde26/1 customClusterEwelink.read(["radioPower",1,20,21,22,23], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.read (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/sonoff.ts:1394:13) at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback

ppotrec, 2025-01-09:
Update ZBMINIR2 Firmware version to 1.0.4. I had the same error message when I was on 1.0.2, after update it's gone.

bertosch, 2025-02-03:
> Update ZBMINIR2 Firmware version to 1.0.4. I had the same error message when I was on 1.0.2, after update it's gone.

Thanks man :) That bothered me as well!

closed 2025-05-04 by github-actions[bot]