Koenkk/zigbee2mqtt #25534: z2m doesn't start with 2.0 on  docker : "discovery_topic"

cthybert, 2025-01-04:
### What happened?

z2m container doesn't start, I don't even have logs ... while booting on 1.42.0 docker imagne it works

### What did you expect to happen?

to start

### How to reproduce it (minimal and precise)

booting this containers

### Zigbee2MQTT version

2.0

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

on the last docker version

### Debug log

```
Starting Zigbee2MQTT without watchdog.
TypeError: Cannot create property 'discovery_topic' on boolean 'true'
    at setValue (/app/lib/util/settingsMigration.ts:55:33)
    at transferValue (/app/lib/util/settingsMigration.ts:180:13)
    at Object.migrateIfNecessary (/app/lib/util/settingsMigration.ts:535:60)
    at start (/app/index.js:136:27)
Using '/app/data' as data directory
```

and my config : 
```yml
homeassistant: true
permit_join: true
devices: devices.yaml
groups: groups.yaml
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://home.home:1883
  user: lolilol
  password: hthisisnotmypassword
  client_id: zigbee2mqtt_nas
  include_device_information: true
serial:
  port: tcp://192.168.1.123:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
frontend:
  port: 8082
  host: 0.0.0.0
advanced:
  homeassistant_discovery_topic: homeassistant
  log_level: warning
  last_seen: ISO_8601_local
  transmit_power: 20
  network_key:
    - 48
    - 206
    - 149
    - 50
    - 2
    - 172
    - 37
    - 62
    - 213
    - 134
    - 175
    - 204
    - 21
    - 235
    - 88
    - 83
  pan_id: 51800
  ext_pan_id:
    - 6
    - 104
    - 85
    - 134
    - 111
    - 202
    - 13
    - 19
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
```

m0wlheld, 2025-01-05:
#24198 

search for `discovery_topic` …

Koenkk, 2025-01-05:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/25518

closed 2025-01-05 by Koenkk