Koenkk/zigbee2mqtt #25537: Z2M 2.0 + Smlight SLZB-06M stops working after few seconds

asjapulk, 2025-01-05:
### What happened?

Z2M stops working after few seconds:
[2025-01-05 03:05:11] warning:  zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2025-01-05 03:05:11] error:    zh:ember:uart:ash: Received unexpected reset from adapter, with reason=RESET_SOFTWARE.
[2025-01-05 03:05:11] error:    zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-05 03:05:11] error:    zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-05 03:05:11] error:    zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-01-05 03:05:11] info:     zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-01-05 03:05:11] info:     zh:ember:uart:ash:   Total frames: RX=43, TX=73

### How to reproduce it (minimal and precise)

It's a new zigbee installation. z2b conf is as follows: (PS: tried rtscts: false in both places, don't now which is correct)

version: 4
homeassistant:
  enabled: true
advanced:
  transmit_power: 20
  rtscts: false
  log_level: debug
  log_debug_to_mqtt_frontend: true
  network_key:
    - 156
….
….
    - 155
  pan_id: 16647
  ext_pan_id:
    - 69
  …
…
    - 123
mqtt:
  server: mqtt://core-mosquitto:1883
  user: myusername
  password: ********
serial:
  port: tcp://SLZB-06M.local:6638
  adapter: ember
  baudrate: 115200
  rtscts: false
frontend:
  enabled: true
  port: 8099

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20241127

### Adapter

smlight SLZB-06M

### Setup

Home assistant z2m addon, Mosquitto broker, Smlight SLZB-06M via network  (core fw: v2.6.8.dev26, zigbee update: 20241127)

### Debug log

[2025-01-05 02:58:07] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.01-54-24'
[2025-01-05 02:58:07] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 02:58:07] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 02:58:07] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 02:58:07] debug: 	z2m: sd-notify loaded
[... 360 lines cut ...]
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   ACK frames  : RX=4, TX=37
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   NAK frames  : RX=1, TX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-01-05 02:58:30] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-01-05 02:58:30] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-01-05 02:58:30] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-01-05 02:58:30] error: 	z2m: Adapter disconnected, stopping
[2025-01-05 02:58:30] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2025-01-05 02:58:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-05 02:58:30] info: 	z2m: Disconnecting from MQTT server
[2025-01-05 02:58:30] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-05 02:58:30] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-01-05 02:58:30] info: 	z2m: Stopped zigbee-herdsman
[2025-01-05 02:58:30] info: 	z2m: Stopped Zigbee2MQTT

medicus07, 2025-01-05:
I also use the Smlight SLZB-06M Stick via TCP... no developer FW for core or zigbee. In Z2M everything was working fine.

With the new version 2.0, my addon no longer starts at all. The error is: No TCP devices could be found.

Pipitapi, 2025-01-05:
I also had to revert to 1.42.0-2 . Could not connect to the SLZB-06 device. In the morning I had trouble with the heating and thermometers stopping :-(

PalmManiac, 2025-01-05:
This may not help now, but I have exactly the same configuration with the same firmware and everything works great. However, I don't have that many Zigbee devices running yet, so maybe that has something to do with it.

medicus07, 2025-01-05:
I got it working now...

Updateted core and zigbee firmware of SLZB-06 to the latest developer build. After the update to Z2M 2.0.0.1 I have adapted the configuration.yaml:

`homeassistant:
  enabled: true
mqtt:
  server: mqtt://192.168.1.27:1883
serial:
  port: tcp://192.168.1.23:6638
  adapter: zstack
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  last_seen: ISO_8601_local
device_options:
  legacy: false
devices:`

asjapulk, 2025-01-05:
I tried @medicus07 solution, but couldn't get it working:

[2025-01-05 15:11:31] info: 	zh:zstack:znp: Socket connected
[2025-01-05 15:11:31] info: 	zh:zstack:znp: Socket ready
[2025-01-05 15:11:31] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-05 15:11:31] debug: 	zh:zstack:unpi:writer: --> buffer [239]
[2025-01-05 15:11:32] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-05 15:11:32] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-01-05 15:11:32] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: <-- [130,80,58,126,114,58,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: --- parseNext [130,80,58,126,114,58,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: <-- [255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,193,89,126]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: --- parseNext [130,80,58,126,114,58,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143,255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,193,89,126]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: <-- [3,59,161,169,254,42,21,179,162,168,159,126]
[2025-01-05 15:11:35] debug: 	zh:zstack:unpi:parser: --- parseNext [130,80,58,126,114,58,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143,255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,193,89,126,3,59,161,169,254,42,21,179,162,168,159,126]
[2025-01-05 15:11:38] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-01-05 15:11:38] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-01-05 15:11:44] debug: 	zh:zstack:znp: --> SREQ: SYS - ping - {"capabilities":1}
[2025-01-05 15:11:44] debug: 	zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: <-- [132,48,252,126,20,56,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,42,21,179,162,168,159,126,132,48,252,126,20,56,161,169,165,42,21,178,83,148,74,37,170,85,146,73,156,78,38,171,236,206,101,139,252,198,99,137,252,125,94,63,167,235,205,222,111,143]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: --> error Error: Invalid checksum
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: --- parseNext [222,111,143]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: <-- [255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,6,130,126]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: --- parseNext [222,111,143,255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,6,130,126]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: <-- [37,57,161,169,254,42,21,179,162,218,77,126]
[2025-01-05 15:11:46] debug: 	zh:zstack:unpi:parser: --- parseNext [222,111,143,255,199,219,213,210,105,140,70,35,169,236,118,59,165,234,117,130,65,152,76,38,125,51,177,224,112,56,28,14,7,187,229,202,101,128,69,154,77,158,79,159,247,195,217,212,106,53,162,81,144,72,36,6,130,126,37,57,161,169,254,42,21,179,162,218,77,126]
[2025-01-05 15:11:50] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 15:11:50] error: 	z2m: Failed to start zigbee-herdsman

[2025-01-05 15:11:50] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

Pipitapi, 2025-01-05:
medicus07 Works for me too!!!

SLZB-06 Firmware :  20240710 ; v2.6.8.dev26

Zibee2Mqtt Conf:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.2.46:1883
serial:
  port: tcp://192.168.2.47:6638
  adapter: zstack
frontend: null
port: 8099
advanced: null
homeassistant_legacy_entity_attributes: false
homeassistant_legacy_triggers: false
legacy_api: false
legacy_availability_payload: false
last_seen: ISO_8601_local
device_options: null
legacy: false

Nerivec, 2025-01-05:
I see 2 problems described in this thread.

@asjapulk How are you running the SLZB06M? With Ethernet, and a stable network behind it? Seems the adapter is having issues communicating properly.

PS: `rtscts` doesn't matter for TCP adapters.

---

For anyone using TCP and Z2M 2.0.0 not starting, specify the `adapter` in configuration.yaml.
```yaml
serial:
  adapter: XXX
```
As noted in the breaking changes, with the numerous adapters now supported by Z2M, `zstack` is no longer the default to prevent issues for users of non-zstack adapters (_before it always tried to start with the assumption that it was zstack if all else failed, which obviously crashed with non-zstack adapters_).

asjapulk, 2025-01-05:
@Nerivec I use it with ethernet. It's behind the same switch as HA and ping is nice and stable.
The "adapter" is defined. I have tried 
zstack - fails to start at all
ember - starts but then gets an error and dies.
and even ezsp - starts and does not shut down but gets  a lot of errors (especially when I try to add a device to the network). So unusable.

Nerivec, 2025-01-06:
_You have to set `adapter` based on the type of the adapter, if it mismatches, it definitely won't work. For SLZB06M, you have to use `ember` (or legacy `ezsp`). The M variant is EmberZNet-based._

Can you also post the logs from the ESP interface (SMLight UI)?

Make sure to go over the docs on [reducing interferences & such](https://www.zigbee2mqtt.io/advanced/zigbee/02_improve_network_range_and_stability.html), it's usually the big factor in initial troubles (there are numerous posts on the web too).
I've added an onboarding menu to Ember ZLI that generates a `configuration.yaml` based on "ideal" settings for your environment (it scans for existing networks/channels usage around you and tries to avoid potential conflicts). Since you're starting out, it's the perfect time to define good network settings, so you don't have to revamp later.
https://github.com/Nerivec/ember-zli/wiki/Stack#zigbee2mqtt-onboard-auto-configuration

You can also try to downgrade the core firmware through the SMLight flasher.

asjapulk, 2025-01-06:
@Nerivec I moved the smlight to different location (but my wifi frequencies are fairly undersaturated (only 2 networks), changed the ethernet cable. Then flashed back the  v2.3.6.
Still about the same. Ember crashes faster and ezsp doesn't crash but doesn't work either.

ember logs:
[2025-01-06 23:22:56] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-06 23:22:56] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-06 23:22:56] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-06 23:22:56] info: 	zh:ember: Using default stack config.
[2025-01-06 23:22:56] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-06 23:22:56] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-06 23:22:56] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-06 23:22:56] info: 	zh:ember:uart:ash: Socket ready
[2025-01-06 23:22:56] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-06 23:22:57] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-01-06 23:22:57] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-01-06 23:22:57] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-01-06 23:22:57] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-01-06 23:22:57] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.5 [GA]","build":0,"major":7,"minor":4,"patch":5,"special":0,"type":170}
[2025-01-06 23:22:58] info: 	zh:ember: [STACK STATUS] Network up.
[2025-01-06 23:22:58] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-01-06 23:22:58] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1247ms until next broadcast.
[2025-01-06 23:22:58] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-06 23:22:58] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":5,"revision":"7.4.5 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-01-06 23:22:58] info: 	z2m: Currently 0 devices are joined.
[2025-01-06 23:22:58] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-06 23:22:58] info: 	z2m: Connected to MQTT server
[2025-01-06 23:22:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-06 23:22:58] info: 	z2m: Started frontend on port 8099
[2025-01-06 23:22:58] info: 	z2m: Zigbee2MQTT started!
[2025-01-06 23:23:03] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xcc86ecfffec73241/connection_state/config', payload '{"device":{"hw_version":"EmberZNet 7.4.5 [GA]","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0xcc86ecfffec73241_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-01-06 23:23:20] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2025-01-06 23:23:20] error: 	zh:ember:uart:ash: Received unexpected reset from adapter, with reason=RESET_SOFTWARE.
[2025-01-06 23:23:20] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-06 23:23:20] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-06 23:23:20] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Total frames: RX=43, TX=73
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   DATA frames : RX=36, TX=35
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   DATA bytes  : RX=304, TX=349
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   ACK frames  : RX=4, TX=37
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   NAK frames  : RX=1, TX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-01-06 23:23:20] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-01-06 23:23:20] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-01-06 23:23:20] info: 	zh:ember: ======== Ember Adapter Stopped ========
[... cut ...]
[2025-01-06 23:23:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-06 23:23:20] info: 	z2m: Disconnecting from MQTT server
[2025-01-06 23:23:20] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-06 23:23:20] info: 	z2m: Stopped zigbee-herdsman
[2025-01-06 23:23:20] info: 	z2m: Stopped Zigbee2MQTT

SMlight logs from web interface:
[23:22:56] taskZB | New client: 192.168.0.10 id: 1
[23:23:17] taskZB | Client disconnected, id: 0
[23:23:18] taskZB | New client: 192.168.0.10 id: 0
[23:23:20] taskZB | Client disconnected, id: 1

configuration.yaml
[image]

asjapulk, 2025-01-06:
With ezsp it's bit strange.   I think iin some cases it somehow changed the username in configuration to "addons".  Don't know why though. But right now I've had some tries when it does not.

Log with  adapter: ezsp

[2025-01-07 00:07:59] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-07 00:07:59] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-07 00:07:59] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-07 00:07:59] warning: 	zh:ezsp: 'ezsp' driver is deprecated and will only remain to provide support for older firmware (pre 7.4.x). Migration to 'ember' is recommended. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2025-01-07 00:08:04] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-07 00:08:04] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":"5 ","majorrel":"7","minorrel":"4","product":13,"revision":"7.4.5.0 build 0"},"type":"EZSP v13"}'
[2025-01-07 00:08:04] info: 	z2m: Currently 0 devices are joined.
[2025-01-07 00:08:04] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-07 00:08:04] info: 	z2m: Connected to MQTT server
[2025-01-07 00:08:04] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-07 00:08:04] info: 	z2m: Started frontend on port 8099
[2025-01-07 00:08:04] info: 	z2m: Zigbee2MQTT started!
[2025-01-07 00:08:09] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xcc86ecfffec73241/connection_state/config', payload '{"device":{"hw_version":"EZSP v13 7.4.5.0 build 0","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0xcc86ecfffec73241_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-01-07 00:08:10] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xcc86ecfffec73241/restart_required/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"EZSP v13 7.4.5.0 build 0","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"device_class":"problem","enabled_by_default":false,"entity_category":"diagnostic","name":"Restart required","object_id":"zigbee2mqtt_bridge_restart_required","origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0xcc86ecfffec73241_restart_required_zigbee2mqtt","value_template":"{{ value_json.restart_required }}"}'
[2025-01-07 00:08:10] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/button/1221051039810110150109113116116_0xcc86ecfffec73241/restart/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bridge/request/restart","device":{"hw_version":"EZSP v13 7.4.5.0 build 0","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"device_class":"restart","name":"Restart","object_id":"zigbee2mqtt_bridge_restart","origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload_press":"","unique_id":"bridge_0xcc86ecfffec73241_restart_zigbee2mqtt"}'
[2025-01-07 00:08:10] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/select/1221051039810110150109113116116_0xcc86ecfffec73241/log_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_template":"{\"options\": {\"advanced\": {\"log_level\": \"{{ value }}\" } } }","command_topic":"zigbee2mqtt/bridge/request/options","device":{"hw_version":"EZSP v13 7.4.5.0 build 0","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"entity_category":"config","name":"Log level","object_id":"zigbee2mqtt_bridge_log_level","options":["error","warning","info","debug"],"origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/bridge/info","unique_id":"bridge_0xcc86ecfffec73241_log_level_zigbee2mqtt","value_template":"{{ value_json.log_level | lower }}"}'
[2025-01-07 00:08:10] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/1221051039810110150109113116116_0xcc86ecfffec73241/version/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":"EZSP v13 7.4.5.0 build 0","identifiers":["zigbee2mqtt_bridge_0xcc86ecfffec73241"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.0.0"},"entity_category":"diagnostic","icon":"mdi:zigbee","name":"Version","object_id":"zigbee2mqtt_br [... cut ...]
[2025-01-07 00:17:35] error: 	zh:ezsp:uart: Can't send DATA frame (0,1,0): 00000004
[2025-01-07 00:17:46] error: 	z2m: Adapter disconnected, stopping
[2025-01-07 00:17:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-07 00:17:46] info: 	z2m: Disconnecting from MQTT server
[2025-01-07 00:17:46] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-07 00:17:46] info: 	z2m: Stopped zigbee-herdsman
[2025-01-07 00:17:46] info: 	z2m: Stopped Zigbee2MQTT

Nerivec, 2025-01-06:
Looks like data is being corrupted in both cases (ember/ezsp).
Can you try opening `ember-zli stack` with your adapter, and just execute the "get info" menu a few times over a few minutes (to see if it crashes after a while too)?

---

Note that it doesn't take many 2.4GHz networks to interfere with the Zigbee network, just one around the same channel...

[image]

asjapulk, 2025-01-06:
> Can you try opening `ember-zli stack` with your adapter, and just execute the "get info" menu a few times over a few minutes (to see if it crashes after a while too)?

It started printing errors even before I pressed get info. Is it broken then?

ember-zli stack
[2025-01-07 01:47:36.177] info: 	cli: Data folder: /root/ember-zli.
✔ Adapter connection type TCP
✔ Try to discover adapter? Yes
[2025-01-07 01:48:26.768] info: 	cli: Starting mDNS discovery...
✔ Select adapter xxxx (tcp://192.168.0.162:6638)
[2025-01-07 01:48:35.016] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-07 01:48:35.018] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-07 01:48:35.031] info: 	zh:ember:uart:ash: Socket ready
[2025-01-07 01:48:35.033] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-07 01:48:36.150] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-01-07 01:48:36.164] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-01-07 01:48:36.166] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-01-07 01:48:36.202] info: 	ember: NCP EZSP protocol version (13) lower than Host. Switched.
[2025-01-07 01:48:36.271] info: 	ember: NCP version: {"ezsp":13,"revision":"7.4.5 [GA]","build":0,"major":7,"minor":4,"patch":5,"special":0,"type":170}
? Menu
❯ Get stack info
  Get stack config (firmware defaults)
  Get network info
  Scan network
  Backup network
  Restore network
  Leave network
(Use arrow keys to reveal more choices)[2025-01-07 01:48:57.551] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
[2025-01-07 01:48:57.554] error: 	zh:ember:uart:ash: Received unexpected reset from adapter, with reason=RESET_SOFTWARE.
[2025-01-07 01:48:57.555] error: 	zh:ember:uart:ash: ASH disconnected: ASH_ERROR_NCP_RESET | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-07 01:48:57.556] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.569] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.584] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.597] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.611] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.621] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.634] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.649] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.663] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.676] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.700] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.713] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.733] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.749] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.771] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.782] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.793] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.805] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.818] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.830] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.844] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.855] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.866] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.877] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.888] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.901] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.912] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.925] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.936] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.946] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:48:57.961] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[... cut ...]
[2025-01-07 01:49:38.277] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:49:48.301] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:49:48.353] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:49:58.378] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.
[2025-01-07 01:49:58.392] error: 	zh:ember:uart:ash: Error while parsing received frame, status=HOST_FATAL_ERROR.

Nerivec, 2025-01-07:
What do you see in the logs of the SMLight interface (core firmware)?
Your adapter is brand new, never used it before? Could be defective...
_If you are still in the return window, I suggest you use it, safer than to start with a misbehaving adapter._

Tarik2142, 2025-01-09:
@asjapulk Firmware v2.6.8.dev26 has a Zigbee channel scanner in the "Z2M/ZHA" menu. Please run the scan and let us know if it works. This scanner communicates with the Zigbee chip directly.

Nerivec, 2025-01-09:
@Tarik2142 in case it helps, Ember ZLI does not start the network when just in the menu like this (it only does the adapter reset basically, up to `VERSION`).
Looks like the stack is resetting (stuck in a loop?).
The rest of the logs above (the many errors), is only because Ember ZLI doesn't handle such an unexpected state (had never seen it before), i.e. it's just the one `RSTACK` that's the problem:
```
[2025-01-07 01:48:57.551] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac1020b0a527e]
```

asjapulk, 2025-01-14:
I do feel somewhat silly.  Talking to the Smlight support I was informed that only one Z2M/ZHA instance can be connected to the coordinator at a time. 
When I connected the 06M to the HA, it installed the Zigbee addon by default. And I did not know I have to disable it. I thought Z2M uses the Zigbee. Once I disabled the Zigbee addon/device,  Z2M started to work. 

I'm sorry I wasted your time. But then again it might be helpful next time when someone else has similar noobish problem.

closed 2025-01-14 by asjapulk

sorenfalch, 2025-04-03:
https://support.smlight.tech/supportcenter/public/en/blog/z2m_frame_cncelled