Koenkk/zigbee2mqtt #25538: Lutron Aurora dimmer knob (Z3-1BRL) not delivering Brightness value on 2.0.0

apiontek, 2025-01-05:
### What happened?

I have a couple Lutron Aurora dimmer knobs (Z3-1BRL). I had them working with Home Assistant, in a "Legacy" mode.

After updating to Z2M 2.0.0 I'm seeing two problems:

1. Device in Z2M shows "Brightness" under "Exposes" but the value never updates. However, I can see State messages updating under "State." I think the issue is that the payload itself always shows `"brightness": null` -- the value one expects to see as "brightness" is delivered in payload as `"action_level": 44` for example, with `"action": "brightness_move_to_level"`

```json
{
    "linkquality": 152,
    "update": {
        "installed_version": 3080,
        "latest_version": 3080,
        "state": "idle"
    },
    "action": "brightness_move_to_level",
    "action_group": 30374,
    "action_level": 44,
    "action_transition_time": 0.02,
    "brightness": null
}
```

2. There used to be a Device setting to enable a "Legacy" mode. This is no longer available, and reference was removed from [the device page](https://www.zigbee2mqtt.io/devices/Z3-1BRL.html). I assume this is related to the general removal of legacy features in 2.0, and that's fine, but I'm noting here since it requires changing automations.

It's also worth noting because when Legacy mode is enabled, the State payload has the `"brightness"` value, for example:

```json
{
    "brightness": 255,
    "linkquality": 216,
    "update": {
        "installed_version": 3080,
        "latest_version": 3080,
        "state": "idle"
    }
}
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.4 [GA] via EmberZNet

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Docker container

m0wlheld, 2025-01-05:
Could be related to disabled Action events?

#24198

apiontek, 2025-01-05:
I don't think that's related, unless there's something under the hood that would be related. 

I'm talking about the value not getting updated within Z2M itself. It's just one of the exposed properties of the device. State messages include the property but the value never changes there either. The actual value that changes is in the state message with a different name, as documented above. 

As far as Home Assistant, this means that the state value exists in home assistant as a device entity but is never updated, so it's not easy to create an automation. However, I was able to create an MQTT trigger in HA to read the state message directly and get the desired value. But this is kind of going behind the curtain just to make my switch work again since the exposed data in Z2M and HA never changes.

asktgm, 2025-01-05:
Experiencing this as well, exact same behavior. 

@apiontek Can you share how you made the MQTT trigger for a workaround by chance? I'm not very experienced setting that kind of trigger up.

tommyjlong, 2025-01-05:
I'll add my 2 cents regarding the "Home Assistant experimental event entities" [wiki](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-home-assistant-event-entity-experimental) that I have enabled for this device ("Home Assistant legacy action sensors" disabled).

The Discovery process has Z2M telling HA to create an event entity for this device whose event types are "brightness".
When I tap the Lutron Aurora dimmer button, Z2M sends HA an event type "brightness_move_to_level", which does not match the discovered event types "brightness".

So HA flags this event type as invalid and this results in an HA Log (_where Lutron Aurora name is "Fireplace Dimmer"_)
```
WARNING (MainThread) [homeassistant.components.mqtt.event] Invalid event type brightness_move_to_level for event.fireplace_dimmer_action received on topic zigbee2mqtt/Fireplace Dimmer, payload {"event_type":"brightness_move_to_level"}
```
Here is what HA-MQTT (from the Lutron Aurora Device page) actually received from the button tap:
```
Topic: zigbee2mqtt/Fireplace Dimmer
Payload: 
action: brightness_move_to_level
action_group: 64821
action_level: 0
action_transition_time: 0.07
brightness: null
linkquality: 94
update:
  installed_version: 3080
  latest_version: 3080
  state: idle
```

apiontek, 2025-01-05:
@asktgm and for whoever else it might help, here's an example automation YAML I came up with as a workaround. The trigger is based on the [HA Automation Trigger doc section on MQTT Trigger](https://www.home-assistant.io/docs/automation/trigger/#mqtt-trigger). Otherwise the behavior/code is modeled after [this HA blueprint](https://community.home-assistant.io/t/zigbee2mqtt-lutron-aurora-dimmer-control/287379).

Change the `topic:` from `zigbee2mqtt/Upstairs Hall Light Control` to `zigbee2mqtt/Device Name` — this is likely what the name in Z2M and HA appears as, but you may also want to watch the Z2M logs/output or use MQTT Explorer to be sure.

And then of course change both `device_id:` lines and change the `alias:`

The below makes use of the "transition time" from the knob's own payload data, but for me, it was 0.2 and I didn't like how it worked, so for me I comment out `transition_time: "{{ trigger.payload_json.action_transition_time }}"` and uncomment `transition_time: "{{ 0.7 }}"` and find that a better experience. YMMV.

Overall this isn't a perfect solution but it suits my needs.

```yaml
alias: Lutron Aurora Dimmer (Upstairs Hall Lamp)
description: MQTT trigger for Lutron Aurora Dimmer Knob
triggers:
  - trigger: mqtt
    topic: zigbee2mqtt/Upstairs Hall Light Control
    payload: brightness_move_to_level
    value_template: "{{ value_json.action }}"
actions:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ action_level == 0 }}"
        sequence:
          - data:
              transition: "{{ transition_time }}"
            target:
              device_id: 7eccead43f6ade7b1d32845a44887092
            action: light.turn_off
    default:
      - target:
          device_id: 7eccead43f6ade7b1d32845a44887092
        data:
          transition: "{{ transition_time }}"
          brightness: "{{ action_level }}"
        action: light.turn_on
mode: queued
max_exceeded: silent
variables:
  action: "{{ trigger.payload_json.action }}"
  action_level: "{{ trigger.payload_json.action_level }}"
  transition_time: "{{ trigger.payload_json.action_transition_time }}"
  # transition_time: "{{ 0.7 }}"
```

Athep, 2025-01-06:
> @asktgm and for whoever else it might help, here's an example automation YAML I came up with as a workaround. 

Thank you for this. I had the same issue with this device: https://www.zigbee2mqtt.io/devices/4512726.html

Your solution works fine.

robertklep, 2025-02-20:
Same issue with the [INNR RC 250](https://www.zigbee2mqtt.io/devices/RC_250.html).

k8gg, 2025-03-25:
And this only happens after the Zigbee2MQTT version 2.0.0 upgrade.

rebpdx, 2026-09-20:
In case anyone else stumbles across this. The above almost worked for me but I had an issue where it wouldn't always turn the light off because the brightness got reported as 255 when you pressed the button. I also noted the transition times of `0.7` when you press the switch and the `0.2` when you rotate the knob.

To fix this I added a 3rd conditional for checking if the button was pressed reporting transition_time as `0.7` and the light is already on.

```
alias: Lutron Aurora Dimmer (Hallway)
description: MQTT trigger for Lutron Aurora Dimmer Knob
triggers:
  - trigger: mqtt
    options:
      topic: zigbee2mqtt/Hallway Switch
      payload: brightness_move_to_level
      value_template: '{{ value_json.action }}'
actions:
  - choose:
      - conditions:
          - condition: template
            value_template: '{{ action_level == 0 }}'
        sequence:
          - data:
              transition: '{{ transition_time }}'
            target:
              device_id: 7eccead43f6ade7b1d32845a44887092
            action: light.turn_off
      - conditions:
          - condition: light.is_on
            target:
              device_id: 7eccead43f6ade7b1d32845a44887092
            options:
              behavior: any
              for: '00:00:00'
          - condition: template
            value_template: '{{ action_level == 255 }}'
          - condition: template
            value_template: '{{ transition_time == 0.7 }}'
        sequence:
          - data:
              transition: '{{ transition_time }}'
            target:
              device_id: 7eccead43f6ade7b1d32845a44887092
            action: light.turn_off
    default:
      - target:
          device_id: 7eccead43f6ade7b1d32845a44887092
        data:
          transition: '{{ transition_time }}'
          brightness: '{{ action_level }}'
        action: light.turn_on
mode: queued
max_exceeded: silent
variables:
  action: '{{ trigger.payload_json.action }}'
  action_level: '{{ trigger.payload_json.action_level }}'
  transition_time: '{{ trigger.payload_json.action_transition_time }}'
```