Koenkk/zigbee2mqtt #25539: MQTT device action based automation isn't working since 2.0.0

rogervdh, 2025-01-05:
### What happened?

I'm using several switches/buttons (for example TS0044) and the automations stopped working. As mentioned in the release notes, it is recommended to use MQTT device triggers. My automations were already build that way, so I assumed that they would remain working. Unfortunately they didn't. I can't figure out why... 

I have also noticed that new actions (for example pressing the button twice, which wasn't discovered before by Z2M) are not showing in the drop down menu when creating a new automation.

### What did you expect to happen?

That my automation based on MQTT device triggers would still work after updating to 2.0.0.

### How to reproduce it (minimal and precise)

The action is visible in the logbook of the MQTT device:
[image]

The automation doesn't trigger:
[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Add-on on Home Assistant mini PC

### Debug log

Example automation: 

```
alias: Lichtschakelaar | Keuken
description: ""
triggers:
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 1_single
    discovery_id: 0x3425b4fffef55a1e action_1_single
    id: links-boven-enkel
    alias: Links boven, enkel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 2_single
    discovery_id: 0x3425b4fffef55a1e action_2_single
    id: rechts-boven-enkel
    alias: Rechts boven, enkel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 3_single
    discovery_id: 0x3425b4fffef55a1e action_3_single
    id: links-onder-enkel
    alias: Links onder, enkel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 4_single
    discovery_id: 0x3425b4fffef55a1e action_4_single
    id: rechts-onder-enkel
    alias: Rechts onder, enkel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 1_double
    discovery_id: 0x3425b4fffef55a1e action_1_double
    id: links-boven-dubbel
    alias: Links boven, dubbel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 2_double
    discovery_id: 0x3425b4fffef55a1e action_2_double
    id: rechts-boven-dubbel
    alias: Rechts boven, dubbel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 3_double
    discovery_id: 0x3425b4fffef55a1e action_3_double
    id: links-onder-dubbel
    alias: Links onder, dubbel drukken
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 2_hold
    discovery_id: 0x3425b4fffef55a1e action_2_hold
    alias: Rechts boven, lang drukken
    id: rechts-boven-ingedrukt
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 4_double
    discovery_id: 0x3425b4fffef55a1e action_4_double
    alias: Rechts onder, dubbel indrukken
    id: rechts-onder-dubbel
    trigger: device
  - domain: mqtt
    device_id: edb52b2a36f1143e3ea5901e82129813
    type: action
    subtype: 4_hold
    discovery_id: 0x3425b4fffef55a1e action_4_hold
    id: rechts-onder-ingedrukt
    alias: Recht onder, lang drukken
    trigger: device
conditions: []
actions:
  - alias: Wandverlichting in- en uitschakelen
    if:
      - condition: trigger
        id:
          - links-boven-enkel
    then:
      - action: light.toggle
        target:
          entity_id: light.wandverlichting_keuken
        data:
          brightness_pct: 35
  - alias: Spots kookeiland in- en uitschakelen
    if:
      - condition: trigger
        id:
          - rechts-boven-enkel
    then:
      - data:
          brightness_pct: 75
        target:
          entity_id: light.spots_kookeiland
        action: light.toggle
  - alias: Verlichting keukentafel in- en uitschakelen
    if:
      - condition: trigger
        id:
          - links-onder-enkel
    then:
      - target:
          entity_id: switch.keukentafel_keuken
        data: {}
        action: switch.toggle
  - alias: Spots openslaande in- en uitschakelen
    if:
      - condition: trigger
        id:
          - rechts-onder-enkel
    then:
      - data: {}
        target:
          entity_id: light.spots_openslaande_deuren
        action: light.toggle
  - alias: Spots kookeiland op 30% lichtsterkte
    if:
      - condition: trigger
        id:
          - rechts-boven-dubbel
    then:
      - data:
          brightness_pct: 30
        target:
          entity_id: light.spots_kookeiland
        action: light.turn_on
  - alias: Spots kookeiland 100% lichttsterkte
    if:
      - condition: trigger
        id:
          - rechts-boven-ingedrukt
        alias: When triggered by rechts-boven-ingedrukt
    then:
      - data:
          brightness_pct: 100
        target:
          entity_id: light.spots_kookeiland
        action: light.turn_on
  - alias: Keukenverlichting uit
    if:
      - condition: trigger
        id:
          - rechts-onder-ingedrukt
    then:
      - target:
          entity_id: scene.keuken_verlichting_uit
        metadata: {}
        action: scene.turn_on
        data: {}
  - if:
      - condition: trigger
        id:
          - rechts-onder-dubbel
    then:
      - target:
          entity_id: scene.woonkamer_keuken_alle_verlichting_uit
        metadata: {}
        action: scene.turn_on
        data: {}
    alias: Woonkamer en keukenverlichting uit
mode: single
```

rchiileea, 2025-01-05:
Same behaviour with aqara WXCJKG13LM opple wireless switches too, visible in z2m show up actions in z2m and show in ha, but all automations with switch broken, tried removal and re add on a couple of them and set up new automation and they don’t work.

rollback to previous version of z2m fixed them

DeminLi, 2025-01-05:
Same with [Aqara wxkg06lm](https://www.zigbee2mqtt.io/devices/WXKG06LM.html#aqara-wxkg06lm)

DeminLi, 2025-01-05:
found the answer: https://github.com/Koenkk/zigbee2mqtt/issues/25500#issuecomment-2571344098

m0wlheld, 2025-01-05:
> All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml.
> ```
> homeassistant:
>   legacy_action_sensor: true
> ```

https://github.com/Koenkk/zigbee2mqtt/discussions/24198

rogervdh, 2025-01-05:
Well, I am already using MQTT device triggers (as recommended). I don't use the legacy action sensors in my automations. So the MQTT device trigger based automation was perfectly working with Z2M 1.42 and it stopped with 2.0.0.

rogervdh, 2025-01-05:
I solved the problem. :-) I had to add the following lines to the Z2M configuration.yaml:

```
homeassistant:
    enabled: true
```

rogervdh, 2025-01-05:
Solved

closed 2025-01-05 by rogervdh

SirJohnDoe, 2025-01-05:
> Well, I am already using MQTT device triggers (as recommended). I don't use the legacy action sensors in my automations. So the MQTT device trigger based automation was perfectly working with Z2M 1.42.0 and it stopped with 2.0.0.

Have this same issue, and the suggested solution by changing the config didnt work.

rchiileea, 2025-01-05:
Closed to soon as I am the same as above and was already using triggers