Koenkk/zigbee2mqtt #25542: Z2M doesn't appear to start

Tuttle13, 2025-01-05:
### What happened?

I'm desperately hoping someone can help me.
I have been using Z2M with HA on my raspberry Pi for approx 18 months, never had a single issue. 

Today I noticed none of my Zigbee devices were responding. I clicked on the Z2M tab and it just says "The Add-on is not running" I click retry or no, nothing happens.

So I go into installed addons, open Z2M and click "start". There's a brief moment of activity in the CPU usage, and then goes to 0 (along with RAM usage).

Again, I haven't touched a single thing in 18 months and had zero issues. (Perhaps Z2M auto updated?)

So I check the logs, and each time I click run, it just does this and then stops:
[19:26:37] INFO: Preparing to start...
[19:26:37] INFO: Socat not enabled
[19:26:39] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 19:26:45] info: z2m: Logging to console, file (filename: log.log)
[2025-01-05 19:26:45] info: z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 19:26:45] info: z2m: Starting zigbee-herdsman (3.2.1)

At the time I noticed it has stopped running, my HA was still on 1.xxx (no auto update and haven't done any updates in past 4 months)
Thinking maybe it needed updating, I have now updated both parts of HA, retstarted and still the same issue.

I've the added the extra config text to the Z2M Config file ,
None of those things already appear elsewhere in my config file (neither advanced or device options), so I copy and pasted in at the bottom:
advanced:
homeassistant_legacy_entity_attributes: false
homeassistant_legacy_triggers: false
legacy_api: false
legacy_availability_payload: false
device_options:
legacy: false

I am at a loss - as you can probably tell, im not a natural at this, but I do heavily rely on my automations based around my home Zigbee devices.

I would be so thankful if someone could assist me on regaining Zigbee control.
Thanks

Edit: now when I try to start it from inside addons, the log just shows a refreshing "500 Internal Server Error"

I'm so lost :(

[image]
[image]
[image]
[image]
[image]

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20250103.0

### Adapter

Conbee

### Setup

Pi

Andybbl, 2025-01-05:
Mine has done the same thing this morning 🙁

tuut84, 2025-01-05:
Same for me. I did all the things mentioned above as well + reinstalled the addon ( retain config data ) but still fails to start

With debug on, i don;t get more info than this

Starting Zigbee2MQTT without watchdog.
[2025-01-05 09:09:05] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.08-17-31'
[2025-01-05 09:09:05] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 09:09:05] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 09:09:05] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 09:09:05] debug: 	z2m: sd-notify loaded
[2025-01-05 09:09:05] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 09:09:05] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 09:09:05] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2226674","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2226674-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyUSB0","manufacturer":"RFXCOM","serialNumber":"A11HEJOE","pnpId":"usb-RFXCOM_RFXtrx433_A11HEJOE-if00-port0","vendorId":"0403","productId":"6001"}]
[2025-01-05 09:09:05] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE221174","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2226674-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
[2025-01-05 09:09:06] debug: 	zh:deconz:driver: Set delay to 0
[2025-01-05 09:09:06] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 09:09:06] debug: 	zh:deconz:driver: Opening with /dev/ttyACM0

RusNimex, 2025-01-05:
Произошло автоматическое обновление до версии 2. У меня тоже перестал запускаться контейнер. Нужно откатиться до 1.42. Это делается в настройках HA. Найдите резервную копию Z2MQTT она будет называться, примерно так `addon_45df7312_zigbee2mqtt_1.42.0-2` и выберите "восстановить". Все заработает снова.

> ### Что случилось?
> Я отчаянно надеюсь, что кто-то сможет мне помочь. Я использую Z2M с HA на моем Raspberry Pi уже около 18 месяцев, и никогда не было ни одной проблемы.
> 
> Сегодня я заметил, что ни одно из моих устройств Zigbee не отвечает. Я нажал на вкладку Z2M, и там просто написано «Дополнение не запущено». Я нажимаю «Повторить» или «Нет», ничего не происходит.
> 
> Итак, я захожу в установленные дополнения, открываю Z2M и нажимаю «запустить». На короткий момент происходит активность в использовании ЦП, а затем она падает до 0 (вместе с использованием ОЗУ).
> 
> Опять же, я не трогал ничего за 18 месяцев и не имел никаких проблем. (Возможно, Z2M автоматически обновился?)
> 
> Итак, я проверяю логи, и каждый раз, когда я нажимаю «Запустить», он просто делает это и затем останавливается: [19:26:37] ИНФОРМАЦИЯ: Подготовка к запуску... [19:26:37] ИНФОРМАЦИЯ: Socat не включен [19:26:39] ИНФОРМАЦИЯ: Запуск Zigbee2MQTT... Запуск Zigbee2MQTT без сторожевого таймера. [2025-01-05 19:26:45] информация: z2m: Регистрация в консоли, файл (имя файла: log.log) [2025-01-05 19:26:45] информация: z2m: Запуск Zigbee2MQTT версии 2.0.0 (коммит #неизвестен) [2025-01-05 19:26:45] информация: z2m: Запуск zigbee-herdsman (3.2.1)
> 
> Когда я заметил, что он перестал работать, мой HA все еще был на версии 1.xxx (без автоматического обновления и не делал никаких обновлений за последние 4 месяца). Подумав, что, возможно, его нужно обновить, я обновил обе части HA, перезапустил, но проблема осталась прежней.
> 
> Я добавил дополнительный текст конфигурации в файл конфигурации Z2M. Ничего из этого больше не встречается в моем файле конфигурации (ни в дополнительных параметрах, ни в параметрах устройства), поэтому я скопировал и вставил в конец: advanced: homeassistant_legacy_entity_attributes: false homeassistant_legacy_triggers: false legacy_api: false legacy_availability_payload: false device_options: legacy: false
> 
> Я в растерянности — как вы, вероятно, поняли, я не прирожденный мастер в этом деле, но я во многом полагаюсь на свою автоматику, основанную на моих домашних устройствах ZigBee.
> 
> Я был бы очень благодарен, если бы кто-нибудь помог мне восстановить контроль над Zigbee. Спасибо
> 
> Редактировать: теперь, когда я пытаюсь запустить его из дополнений, в журнале просто появляется обновляющаяся надпись «500 Internal Server Error»
> 
> Я так запуталась :(
> 
> ![Скриншот_20250105_200250_Домашний помощник](https://private-user-images.githubusercontent.com/69562495/400189603-06d7b41f-fe0a-4b20-bf1c-a70049d5ed86.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNzEyOTMsIm5iZiI6MTczNjA3MDk5MywicGF0aCI6Ii82OTU2MjQ5NS80MDAxODk2MDMtMDZkN2I0MWYtZmUwYS00YjIwLWJmMWMtYTcwMDQ5ZDVlZDg2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDA5NTYzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNmM2I0NWEzNTljNDcyMjRkNzgyZDBlN2M1Y2Q5NzRhZWI1MGU3NWVhYzk0NDhlMTkzYTI1OGM5NzczYjRkZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wQkBG_CjIKuLa6wFUSSurs7jXr8KUPbksnkUeL53a6s) ![Скриншот_20250105_200245_Домашний помощник](https://private-user-images.githubusercontent.com/69562495/400189610-a9f39328-b545-4390-93d9-a791c41c33fe.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNzEyOTMsIm5iZiI6MTczNjA3MDk5MywicGF0aCI6Ii82OTU2MjQ5NS80MDAxODk2MTAtYTlmMzkzMjgtYjU0NS00MzkwLTkzZDktYTc5MWM0MWMzM2ZlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDA5NTYzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVkNGRlNzc5YTc5Njk2MzExODNkYjFhZGIwMWIxMDQ0N2VmMzA1NzE1YzcyMWQzYjA5ZWI4MDE2MmM4MWIzNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.BJdShUq1n-dZ2Z3VXOsp-I6Bd9Bz9cnga2ox3M2zZHk) ![Скриншот_20250105_192243_Домашний помощник](https://private-user-images.githubusercontent.com/69562495/400189611-0f5b8559-b1e9-4df2-ab62-6fb75f55b538.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNzEyOTMsIm5iZiI6MTczNjA3MDk5MywicGF0aCI6Ii82OTU2MjQ5NS80MDAxODk2MTEtMGY1Yjg1NTktYjFlOS00ZGYyLWFiNjItNmZiNzVmNTViNTM4LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDA5NTYzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJiOTk1YzQwNjg2OTIxOGIyY2E1N2Q4MzUzYzNmMjcxODg1ZDk4ZTk5Y2E2MTBjZTA1YTcyNGE0MWY3ODkxNjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MfO501CobJAmmmOEN6kWopBwHmq8AnJuBC3lqDsM9XM) ![Скриншот_20250105_192241_Домашний помощник](https://private-user-images.githubusercontent.com/69562495/400189612-3b2a9a65-d338-41b1-abba-f878477a9fed.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNzEyOTMsIm5iZiI6MTczN [... cut ...]
> ### Как это воспроизвести (минимально и точно)
> _Нет ответа_
> 
> ### Версия Zigbee2MQTT
> 2.0.0-1
> 
> ### Версия прошивки адаптера
> 20250103.0
> 
> ### Адаптер
> Конби
> 
> ### Настраивать
> Пи
> 
> ### Журнал отладки
> _Нет ответа_

Произошло автоматическое обновление до версии 2. У меня тоже перестал запускаться контейнер. Нужно откатиться до 1.42. Это делается в настройках HA. Найдите резервную копию Z2MQTT она будет называться, примерно так `addon_45df7312_zigbee2mqtt_1.42.0-2` и выберите "восстановить". Все заработает снова.

Andybbl, 2025-01-05:
I have sorted it

looking at other posts

on the z2mqtt window click config
In the serial window under you dongle info add the line

adapter: zstack
Mine then restarted and worked

tuut84, 2025-01-05:
I saw that as well in multiple post however the topic starter and me are using Deconz which is required for Conbee

Next determine what adapter you are using by going to your [adapter page](https://www.zigbee2mqtt.io/guide/adapters/). Possible adapters are zstack, ember, deconz, zigate or zboss.

git-greatguy, 2025-01-05:
same here, adding "adapter" config make it start again.

m0wlheld, 2025-01-05:
> Specify valid 'adapter' and 'port' in your configuration.

#24198

m0wlheld, 2025-01-05:
> [Tuttle13](https://github.com/Tuttle13)
> 
> There was an automatic upgrade to version 2. My container also stopped starting. We need to roll back to 1.42. This is done in the HA settings. Find the backup copy of Z2MQTT, it will be called something like `addon_45df7312_zigbee2mqtt_1.42.0-2` and select "restore". Everything will work again.

Better read the release notes and the [Breaking Changes](#24198), otherwise you won't be update to more recent versions.

m0wlheld, 2025-01-05:
> I saw that as well in multiple post however the topic starter and me are using Deconz which is required for Conbee
> 
> Next determine what adapter you are using by going to your [adapter page](https://www.zigbee2mqtt.io/guide/adapters/). Possible adapters are zstack, ember, deconz, zigate or zboss.

So, you added `adapter: deconz` to your `serial:` section and it worked (as documented)?

tuut84, 2025-01-05:
for me it didn't

serial:
  port: /dev/ttyACM0
  adapter: deconz

also tried replacing port based on /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2226674-if00

m0wlheld, 2025-01-05:
> [19:26:37] INFO: Preparing to start...
> [19:26:37] INFO: Socat not enabled
> [19:26:39] INFO: Starting Zigbee2MQTT...
> Starting Zigbee2MQTT without watchdog.
> [2025-01-05 19:26:45] info: z2m: Logging to [...]

So, it is working now or not? The Z2M log do not indicate a failure.

Also, please use code formatting ([image]) for copied configuration files and logs, as indendation needs to stay intact. Do not use screenshots of them, please.

m0wlheld, 2025-01-05:
Formatted like this:

```yaml
serial:
  port: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2226674-if00
  adapter: deconz
```

Tuttle13, 2025-01-05:
Thanks to everyone who suggested restoring the 1.xx backup - that worked (and for those with similar issues, my HA core is still updated to 2.xx)

Not sure what will subsequently break due to this 1.xx/2.xx mismatch, but I'll take my chances of it means all my Zigbee devices are available.

- and I did/still do have the device inputted (always had "adapter: deconz" in the config, so that wasn't the issue as far as I can tell..

*also if this helps anyone else, don't forget to turn off auto updates!

yell3D, 2025-01-05:
same here, `adapter: deconz`. Added the other options from the 2.0 page in in HA and config yaml does nothing. rolling back to 1.42.0-2 and disabling auto update. super unpleasant morning

tuut84, 2025-01-05:
@m0wlheld i have updated my comments with the code blocks.

I'm not rolling back yet as i still have some time to try things out without anyone at home complaining that lights and other things are not working :-)

roobbb, 2025-01-05:
Hello.

I also suffered the same issue. But I have a ZigStar Ethernet adapter (zstack based). The docker container didn't start after upgrading to V2. Log said:
```
...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 10:27:15] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 10:27:15] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-05 10:27:15] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 10:27:15] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 10:27:15] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 10:27:15] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 10:27:15] error: 	z2m: Exiting...
[2025-01-05 10:27:15] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
...
```
As mentioned above I just added "adapter: zstack" to my configuration.yaml:
```
...
serial:
  port: tcp://192.168.0.65:6638
  adapter: zstack
frontend:
  enabled: true
...
```

Now it's running:
```
...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 11:30:17] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 11:30:17] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-05 11:30:17] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 11:30:18] info: 	zh:zstack:znp: Opening TCP socket with 192.168.0.65:6638
[2025-01-05 11:30:18] info: 	zh:zstack:znp: Socket connected
[2025-01-05 11:30:18] info: 	zh:zstack:znp: Socket ready
[2025-01-05 11:30:18] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-05 11:30:19] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
...
```

Thank you and kind regards 😀
roobbb

m0wlheld, 2025-01-05:
> same here, `adapter: deconz`. Added the other options from the 2.0 page in in HA and config yaml does nothing. rolling back to 1.42.0-2 and disabling auto update. super unpleasant morning

Maybe you want to create an issue at the [Home Assistant Zigbee2MQTT add-on repository](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt), because plain Zigbee2MQTT (this repository) is working as expected, assuming changes according to #24198 have been applied.

yell3D, 2025-01-05:
> Maybe you want to create an issue at the [Home Assistant Zigbee2MQTT add-on repository](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt), because plain Zigbee2MQTT (this repository) is working as [...]

so you confirm that `adapter: deconz` works with plain/standalone Z2M?
(in my case Conbee 2 on a rpi4)

m0wlheld, 2025-01-05:
> so you confirm that `adapter: deconz` works with plain/standalone Z2M? (in my case Conbee 2 on a rpi4)

No, I can't confirm, as I use [zstack](https://www.zigbee2mqtt.io/guide/adapters/zstack.html) coordinator devices, but adding `adapter: zstack` to my configuration.yaml kept Z2M working when updating from 1.42.0 to 2.0.0

yell3D, 2025-01-05:
> No, I can't confirm, as I use [zstack](https://www.zigbee2mqtt.io/guide/adapters/zstack.html) coordinator devices, but adding `adapter: zstack` to my configuration.yaml kept Z2M working when [...]

Whats the point of you telling us Conbee users to add `adapter` to our configuration that we always needed ?

> because plain Zigbee2MQTT (this repository) is working as expected

How can you claim this while using a different coordinator and then forward us to the Homeassistant addon?

m0wlheld, 2025-01-05:
> Whats the point of you telling us Conbee users to add `adapter` to our configuration that we always needed ?

In https://github.com/Koenkk/zigbee2mqtt/issues/25542#issuecomment-2571577372, you said you added it for Z2M 2.0.0?

Tuttle13, 2025-01-05:
I think yell3D is in the same boat at me - we already had "adapter: deconz" in our Config way before 2.0
Ie it's always been there and so there's no way this can be the solution as it's already sitting there.

tuut84, 2025-01-05:
i have created [https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/691](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/691) . not sure if it will be flagged as duplicate topics but lets see how it goes

m0wlheld, 2025-01-05:
> I think yell3D is in the same boat at me - we already had "adapter: deconz" in our Config way before 2.0 Ie it's always been there and so there's no way this can be the solution as it's already sitting there.

Understood. Actually, in your original post I don't even see an error message coming from Zigbee2MQTT. So it could be related to the HASS Zigbee2MQTT Add-On with Conbee/Raspbee only. @tuut84  just opened issue https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/691 over there.

gabi1324, 2025-01-05:
Hi! I managed to make the addon start but it still didn't work. When trying to add a new device nothing happened. I deleted everything and I tried restoring from an older backup. Now it sees the device as 0x0000000000000000 and it doesn't interview it successfully. Any help would be appreciated.

GuyVerDonck, 2025-01-05:
After update to Z2M 2.0.0 and implementing fix "adapter: zstack" in zigbee configuration yaml Z2M worked again...
Except illuminance from IKEA, HUE and TUYA devices is not reported to HA.

In Z2M it is correctly identified

[image]

In HA the sensor.......illumination is no longer available ?!

[image]

Have a lot of automations based on illumination.
Any suggestions or solutions please ?

m0wlheld, 2025-01-05:
> Any suggestions or solutions please ?

Replace _illuminance_lux by _illuminance as mentioned by #24198 . Might also be required to activate the new "illuminance only" entities in HA.

ibex72, 2025-01-05:
I have a Sonoff Zigbee 3.0 Plus (TI CC2652P + CP2102N)  and today the same problem. Z2M doesn`t starts.
For me works to add adapter zstack in serial:

port: /dev/ttyUSB0
adapter: zstack

GuyVerDonck, 2025-01-05:
> I have a Sonoff Zigbee 3.0 Plus (TI CC2652P + CP2102N) and today the same problem. Z2M doesn`t starts. For me works to add adapter zstack in serial:
> 
> port: /dev/ttyUSB0 adapter: zstack

As stated in my first sentence ;-)

GuyVerDonck, 2025-01-05:
> > Any suggestions or solutions please ?
> 
> Replace _illuminance_lux by _illuminance as mentioned by #24198 . Might also be required to activate the new "illuminance only" entities in HA.

SOLVED: thanks for the hint and reference !

jvelzz, 2025-01-05:
I have a cod.m ZigBee Coordinator CC2652P7 (CZC-1.0) and today had the same problem. Z2M didn't want to start and I spent a lot of hours trying to figure out why.

Coming here to check if anyone else had the issue and found this issue. Adding the line below to my configuration.yaml as was mentioned multiple times here fixed it for me, thank you guys.

`adapter: zstack`

grandras, 2025-01-05:
Same happened with me....
Auto update was on. :(
When I woke up in the moring, all zigbee devices were down..

There was nothing helpful in the logs.
After the " zh:zstack:znp: Opening SerialPort " entry, nothing happens. The add-on is not running. Then the next line is when I tried to start it again, and again... but hothing happened.

I also rolled back to previous version from backup. Now it works. 

```
[11:16:08] INFO: Preparing to start...
[11:16:09] INFO: Socat not enabled
[11:16:10] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 11:16:16] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.11-14-30'
[2025-01-05 11:16:16] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 11:16:16] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 11:16:16] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 11:16:16] debug: 	z2m: sd-notify loaded
[2025-01-05 11:16:16] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 11:16:16] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 11:16:16] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6e08ca940460ec11af0b2e5f25bfaa52","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e08ca940460ec11af0b2e5f25bfaa52-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-05 11:16:16] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6e08ca940460ec11af0b2e5f25bfaa52","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e08ca940460ec11af0b2e5f25bfaa52-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-01-05 11:16:16] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 11:16:16] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[11:18:02] INFO: Preparing to start...
[11:18:02] INFO: Socat not enabled
[11:18:04] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 11:18:09] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.11-14-42'
[2025-01-05 11:18:09] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 11:18:09] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 11:18:09] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 11:18:09] debug: 	z2m: sd-notify loaded
[2025-01-05 11:18:09] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 11:18:09] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyUSB0\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-05 11:18:09] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6e08ca940460ec11af0b2e5f25bfaa52","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e08ca940460ec11af0b2e5f25bfaa52-if00-port0","vendorId":"10c4","productId":"ea60"}]
[2025-01-05 11:18:09] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"ITead","serialNumber":"6e08ca940460ec11af0b2e5f25bfaa52","pnpId":"usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_6e08ca940460ec11af0b2e5f25bfaa52-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: path=/dev/ttyUSB0, score=4
[2025-01-05 11:18:09] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-05 11:18:09] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[11:20:28] INFO: Preparing to start...
```

m0wlheld, 2025-01-05:
> I have a cod.m ZigBee Coordinator CC2652P7 (CZC-1.0) and today had the same problem

That is not the same problem and the solutions (add `adapter:`) settings is part of the Zigbee2MQTT release notes and the associated list of breaking changes. 

Problem here is some setups not able to communicate with the coordinator device, despite the fact that their configuration is 2.0.0 ready.

tuut84, 2025-01-05:
The current edge build seems to work for me , the addon is starting again.

Koenkk, 2025-01-05:
@grandras that issue is fixed in addon 2.0.0-2!

grandras, 2025-01-05:
> @grandras that issue is fixed in addon 2.0.0-2!

Thank you! You are doing a great job!

Tuttle13, 2025-01-05:
@Koenkk does this mean I should attempt (again) to upgrade to 2.0.0-2? I don't use the edge version, just the regular Zigbee2MQTT addon.

Just to re-clarify, I had a working system prior to 2.0, and have always had the adapter and port lines of code in my config file, which was there obviously well before 2.0 came about.
(my port: line looked a bit different to any of the install instructions, but from memory it was the only way I could get it to recognize the USB Conbee stick?)

If this doesn't make sense, it's it my original screenshots up top (apologies again for including my.code in this format)

Cheerd

Koenkk, 2025-01-06:
@Tuttle13 try to check for updates again, 2.0.0-2 should be there now.

haegar33, 2025-01-15:
I have a Raspbee II adapter sitting in my Raspberry 3. Thus config is of course 
`serial:
  adapter: deconz
  baudrate: 38400
  port: /dev/ttyAMA0`

Last week this config broke for unknown reasons. I ruled out the following options:

- under voltage message -> replaced powersupply -> no effect
- incompatible version 2.0? -> downgrade to previous V1.41 -> no effect

For my case it looks like there is no communication at all:
```
[2025-01-15 14:06:51] debug: 	zh:deconz:driver: Opening with /dev/ttyAMA0
[2025-01-15 14:06:51] debug: 	zh:deconz:driver: Serialport opened
[2025-01-15 14:06:51] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 5
[2025-01-15 14:07:01] debug: 	zh:deconz:driver: Timeout for request - CMD: 0xa seqNr: 1
[2025-01-15 14:07:01] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-15 14:07:01] error: 	z2m: Failed to start zigbee
[2025-01-15 14:07:01] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2025-01-15 14:07:01] error: 	z2m: Exiting...
[2025-01-15 14:07:01] error: 	z2m: Error: TIMEOUT
    at Driver.processBusyQueue (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:515:28)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:168:40)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

/opt/zigbee2mqtt/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/opt/zigbee2mqtt/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/opt/zigbee2mqtt/node_modules/winston/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/opt/zigbee2mqtt/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/opt/zigbee2mqtt/lib/util/logger.ts:198:25)
    at Logger.debug (/opt/zigbee2mqtt/lib/util/logger.ts:219:14)
    at Driver.processQueue (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:484:28)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:163:22)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
 ELIFECYCLE  Command failed with exit code 1.
```

closed 2025-03-25 by github-actions[bot]