Koenkk/zigbee2mqtt #25544: ACTION entity on my buttons are no longer exposed to HA

Martin-Lucan, 2025-01-05:
### What happened?

Recently 1-3 day ago my buttons stopped exposing ACTION entity to HA. Not sure exactly where the issue is.
I am using HA, Mosquito and zigbee2mqt all on docker. 
I tried updating everything, restarting my machine or also re-adding my button with diffrent ID. Nothing worked. 
a
[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.1.1.0 build 273

### Adapter

EZSP v9

### Setup

Docker - HA - Mosquitto - Ziqbee2mqtt - Raspberry pi 5

janpfischer, 2025-01-05:
See this. This helped me as well:

https://github.com/Koenkk/zigbee2mqtt/issues/25539#issuecomment-2571539201

ageurtse, 2025-01-05:
did you read this one [https://github.com/Koenkk/zigbee2mqtt/discussions/24198](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

and what is happening when you go to zigbee2mqtt->settings->Home assistant intergration and you enable Home Assistant legacy action sensors, then restart. for me that solved it.

[image]

m0wlheld, 2025-01-05:
> It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead.

Go for this, because legacy support will end some day.

soonic6, 2025-01-05:
and whats about automations/blueprints which is working with entities?

lug-gh, 2025-01-05:
> > It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead.
> 
> Go for this, because legacy support will end some day.

Are there any instructions for this somewhere?

m0wlheld, 2025-01-05:
> > > It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead.
> > 
> > 
> > Go for this, because legacy support will end some day.
> 
> Are there any instructions for this somewhere?

Yes, the blue "MQTT device trigger" text is a link ...

lug-gh, 2025-01-05:
> Yes, the blue "MQTT device trigger" text is a link ...

Unfortunately, there is not much information there. It only says that they will be “discovered”. Where are they discovered? You can no longer see them in the GUI, or how should I understand that?

m0wlheld, 2025-01-05:
> You can no longer see them in the GUI, or how should I understand that?

There is an automation example right below the headline.

lug-gh, 2025-01-05:
> > You can no longer see them in the GUI, or how should I understand that?
> 
> There is an automation example right below the headline.

This issue has nothing to do with automation, nor did I write anything about automation.

m0wlheld, 2025-01-05:
Okay, maybe I am on the wrong track. Could you please elaborate why you originally create comment https://github.com/Koenkk/zigbee2mqtt/issues/25544#issuecomment-2571573944 ?

lug-gh, 2025-01-05:
> Okay, maybe I am on the wrong track. Could you please elaborate why you originally create comment [#25544 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25544#issuecomment-2571573944) ?

Because I have the same problem as @Martin-Lucan. And I was hoping to get more information about it somewhere. I've only been using zigbee2mqtt for two weeks, so I still need to build up the knowledge.

m0wlheld, 2025-01-05:
@Martin-Lucan's issue (actually not an issue) is caused by a change, that has been introduced with recent release of Zigbee2MQTT 2.0.0 . The effects of this chance and counter-measures are documented in the [Zigbee2MQTT Release Notes](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) and the linked issue #24198: You can either choose to stay with legacy support (which will probably not last forever) or use [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended). 

Triggers may be used by Home Assistant automations, that's why I considered https://github.com/Koenkk/zigbee2mqtt/issues/25544#issuecomment-2571577709 to be helpful.

mattiaskagstrom, 2025-01-05:
How do i access the mqtt trigger in node-red when it is not exposed as an entity on the device?

paegan999, 2025-01-05:
Suddenly overnight i have the same problem, never had it before, but tonight it happened for me too.
I've turned on the legacy support in the addon and it was fixed.
But what i do not understand, why does this happen overnight? (no update done)
Today i updated the core, in the hope it would fix the issue, but not, so i went searching and found this issue and solution overhere.

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-05 by Koenkk

mattiaskagstrom, 2025-01-05:
> This is one of the Zigbee2MQTT 2.0.0 breaking changes, from #24198

Are these triggers available some how in node-red?
This seems un-googlable, i only get results on how to trigger flows from other flows

Vangen92, 2025-01-19:
> Yes, the blue "MQTT device trigger" text is a link ...

@m0wlheld: I think what @lug-gh means is that he could use some help with what he specifically needs to do. I also don't quite understand the guide [here](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended)... 
So if I have a device (say a smart button) where the action entity isn't showing by default in that MQQT device in HA (eventhough it is showing in the Zigbee2MQTT integration), all I have to do is use that entity in an automation? And then it will just appear in that MQQT device?

heejew, 2025-01-25:
OMG, these Zigbee2MQTT's breaking changes...
A few hours researching of my problem after reboot.

Yeah, my mistake also, I'm using the latest build. But...this was established precisely according to guides and recommendations a few years ago.

And... What benefit do I get from "no legacy" for these breaking changes? I don't understand.

guaycuru, 2025-02-04:
I have the same problem and, as others, I don't understand the recommended solution to use MQTT device trigger. What do I have to do to use my MQTT button click as an action trigger in a HA automation without using the legacy mode?