Koenkk/zigbee2mqtt #25546: IKEA Styrbar E2001/E2002 left button doesn't work with latest firmware 2.4.16

rohitpid, 2025-01-05:
### What happened?

I have read through this thread: https://github.com/Koenkk/zigbee2mqtt/issues/22458 which discusses getting newer firmwares than 1.0.024 to work with z2m. I have a styrbar upgraded to 2.4.16 firmware and the left button doesn't work but the right button does. I get log messages in the z2m logs for the right button but nothing for the left button.

I have tried:
* re-pairing the remote many times
* removing the battery and pressing bind and putting the battery in
* checking logs for button press events
* my 1.0.024 switch uses the same automation as 2.4.16 switch and the left button action works fine there

Can anyone help? Everything works fine for my other switch on 1.0.024 firmware. Is there a way to downgrade firmware? Can the newer firmware be made to work?

### What did you expect to happen?

The left button press should return log messages and turn on the lights bound to the action.

### How to reproduce it (minimal and precise)

Press the left button of the styrbar remote

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

7.4.4

### Adapter

ember

### Setup

HA Green

Koenkk, 2025-01-14:
It may help to force remove the device, **restart z2m** and re-pair the device.

rohitpid, 2025-01-15:
@Koenkk I have done that many times without any success.

Greendragoonjr, 2025-01-16:
Heya, 

I have kinda the similar problem (see [Issue 22458 ](https://github.com/Koenkk/zigbee2mqtt/issues/22458) )

For me the Styrbar remote seems correctly binded with ikea color bulbs. The left and right button press are correctly detected by ZM. And immediately that is. But they will not chnage the bulbs color between their 8 default set. But the up and down buttons works fine.

I also made multiple rebind, force remove, reset the remote AND bulbs outside of HA and rebind them within Z2M. I now have a usable situation, but not being able to arrows to change the colors is a bit annoying. I can do it within HA but still faster if using the remote.

Here are my logs when pressing the left and right button: 

[image]

By the way, I don't know why but i can't have the reading of the battery level of the remote. The battery level is reading "Null" value.

Here is how the remote is binded with the coordinator and light bulbs 

[image]

Personally i don't want to try downgrading the firmware. it seems quite complicated and i'm not confident being able to do it without breaking stuff

MDMAN, 2025-01-22:
I have the same problem. The LEFT and RIGHT Button are not working. But when I press the LEFT or RIGHT Button the Z2M show me the pressed Button "light on".

What can we do?

I have installed also the last firmeware 2.4.16. The Update was successfully... But bevor I have updated the Styrbar, the problem was also with the Left and Right-Button.

newlund, 2025-02-02:
> EDIT:
> 
> I habe Restart the Service under "Configuration --> Tools --> Restart" and now, I can use the LEFT and RIGHT-Button perfect.

Uh, how? By restarting Z2M?

I have exactly the same problem described by [Greendragoonjr](https://github.com/Greendragoonjr)

I have two Styrbar [E2001/E2002](https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002)

One with 2.4.11 that works fine with left/right button color change (bound to group). I have not upgraded this one to 2.4.16 due to this problem.
One with 2.4.16 that can't control the color with left/right button. I have tried bound to group and bound directly to the lamp.

newlund, 2025-02-02:
Yay! I finally managed to downgrade my remote to 1.0.024 by adding this in a index.json file in the zigbee2mqtt config directory:

```
[
  {
    "fileVersion": 65572,
    "fileSize": 194002,
    "manufacturerCode": 4476,
    "imageType": 4555,
    "force": true,
    "sha512": "732ed02c39087dfffbe1d6e7adaf23851ff8dc695fe1e0c85daa5b7e1a3bea85796181d6da02b695685bc35470cf0c69cb06cd73c4bb51c67f89a198b1dac78f",
    "url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.19.026/bin/10078031-zingo_kt_styrbar_remote-1.0.024.ota.ota.signed"
  }
]
``` 
After downgrading I could bind the remote to a group and left/right color changing works! :)

Now I'm trying to update it to 2.4.11 again using this in index.json:

```
[
  {
    "fileVersion": 33816593,
    "fileSize": 269638,
    "manufacturerCode": 4476,
    "imageType": 4555,
    "force": true,
    "sha512": "77c56399abe9fef1ce176c73e347be029686f7f0406639c31a80f56dfe634df176c8f216f3c40f95c01e2968244d57c09b72698fc618c92c99153820b7086292",
    "url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/styrbar_dimmer_zingo-0x11cb-2.4.11-prod.ota.ota.signed"
  }
]
``` 

Add this to Z2M configuration.yaml in order to use the added index.json file and restart zigbe2mqtt

```

ota:
  zigbee_ota_override_index_location: index.json
  disable_automatic_update_check: true
``` 

Cred to misternic
[https://github.com/Koenkk/zigbee2mqtt/discussions/23243#discussioncomment-10038494](url)

rohitpid, 2025-02-03:
@newlund I followed your instructions but when I click on check for update on the OTA tab I get "No new updates". What am I missing? I created an index.json in the z2m directory and added the lines to configuration.yaml in the VSCode Extension for homeassistant.

rohitpid, 2025-02-03:
@newlund OMG thank you so much. You revived this remote. Is there any way to make this a special option for this remote for the general public in z2m without having to go through special configuration. This particular downgrade for this remote will help a lot of people. @Koenkk

Taxom, 2025-02-03:
Hi, I'm trying, but for some reason, it's not working.

I created the index.json file with the specified content in the /homeassistant/zigbee2mqtt folder.
I added the specified lines to the end of /homeassistant/zigbee2mqtt/configuration.yaml and restarted zigbee2mqtt, but the device page still shows that there are no updates available.

[image]

I'm definitely missing something, but I can't figure out what ((

Taxom, 2025-02-03:
info 2025-02-03 08:15:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe2c2bec', payload '{"action":"off","battery":85,"identify":null,"linkquality":255,"update":{"installed_version":33816598,"latest_version":65572,"state":"available"}}'
info 2025-02-03 08:15:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe2c2bec/action', payload 'off'
info 2025-02-03 08:15:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe2c2bec', payload '{"action":"on","battery":85,"identify":null,"linkquality":252,"update":{"installed_version":33816598,"latest_version":65572,"state":"available"}}'
info 2025-02-03 08:15:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe2c2bec/action', payload 'on'
info 2025-02-03 08:15:30z2m: Updating '0x540f57fffe2c2bec' to latest firmware
info 2025-02-03 08:16:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x540f57fffe2c2bec', payload '{"battery":85,"identify":null,"linkquality":252,"update":{"installed_version":33816598,"latest_version":65572,"state":"available"}}'
info 2025-02-03 08:16:40z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of '0x540f57fffe2c2bec' failed (Device didn't respond to OTA request)","status":"error","transaction":"uirtu-4"}'
error 2025-02-03 08:16:40z2m: Update of '0x540f57fffe2c2bec' failed (Device didn't respond to OTA request)

newlund, 2025-02-03:
Right after you have pressed "Check for new updates", click a button on the remote to wake it up.

Greendragoonjr, 2025-02-03:
Heya, 

I followed the instructions of @newlund. Thanks for that, the rollback was actually more easy to do than expected. But i don't think i would have managed to make the index.json file properly.

So now both my Styrbar E2001/E2002 remote are back in 2.4.11. 

But.... the left and right arrows still don't work. I didn't try to rebind them after unbinding them first tho so maybe that would help.

On a positive note, i got the battery level reading back so yeah i guess 😅.

Taxom, 2025-02-04:
I thought this was just my problem too. I also rolled back to 1.0.024 and then to 2.4.11, but it didn’t help. I still can’t bind the switch to the lamp so that the side buttons work (at least for color temperature).
If I have time tomorrow, I’ll try a second switch to see if it works for changing colors (RGB lamp).

Taxom, 2025-02-04:
So, my tests showed that if both the remote and the lamp are connected to Home Assistant, direct control of the lamp color for "TRADFRI bulb E12 CWS 450lm" (LED1925G6 fw1.0.021) only works with firmware 1.0.024 on the "STYRBAR remote control" (E2001/E2002) and does not work on 2.4.11 or 2.4.16.

Unfortunately, direct control of the color temperature for "TRADFRI bulb E26 WS globeopal 1100lm" (LED2003G10 fw1.0.012) does not work with any firmware version.

Thanks, @newlund , for sharing the instructions! That helped me solve the issue with my bedside lamp, which I can finally integrate into Home Assistant.

Taxom, 2025-02-04:
To control the color temperature of the "TRADFRI bulb E26 WS globeopal 1100lm" via Home Assistant, I used this script:

`
```
alias: control the color temperature STYRBAR
description: ""
triggers:
  - topic: zigbee2mqtt/torchiere_living_switch/action
    payload: arrow_right_click
    trigger: mqtt
  - topic: zigbee2mqtt/torchiere_living_switch/action
    payload: arrow_left_click
    trigger: mqtt
actions:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ trigger.payload == 'arrow_right_click' }}"
        sequence:
          - data:
              topic: zigbee2mqtt/torchiere_living/set
              payload: |
                {
                  "color_temp": {{ ((state_attr('light.torchiere_living', 'color_temp') | int + 10) - 250) % (454 - 250 + 1) + 250 }}
                }
            action: mqtt.publish
      - conditions:
          - condition: template
            value_template: "{{ trigger.payload == 'arrow_left_click' }}"
        sequence:
          - data:
              topic: zigbee2mqtt/torchiere_living/set
              payload: |
                {
                  "color_temp": {{ ((state_attr('light.torchiere_living', 'color_temp') | int - 10) - 250) % (454 - 250 + 1) + 250 }}
                }
            action: mqtt.publish
```
`
I hope this helps someone.

newlund, 2025-02-04:
It should work to control the lamp colors with the remote left/right buttons with 2.4.11.
But! It does not work if binding the remote directly to the lamp. If you create a group (with one lamp or several) and bind the remote to that instead it should work.
Same thing with 2.4.16 but that also removed the ability to bind to group properly...

Taxom, 2025-02-04:
Thanks, I'll try it later when I have time. Thanks to you, updating the firmware is now easy, but the flashing process itself still takes a lot of time.

rohitpid, 2025-02-04:
So I am using 1.0.024 again and I can control my lights. Does anyone have 2.4.16 working with the left and right buttons or is that not working? When I was on 2.4.16 I tried binding the scenes but I couldn't get it to work.

newlund, 2025-02-04:
> So I am using 1.0.024 again and I can control my lights. Does anyone have 2.4.16 working with the left and right buttons or is that not working? When I was on 2.4.16 I tried binding the scenes but I couldn't get it to work.

You can probably upgrade to 2.4.11 if you bind the remote to group instead of device as I wrote in previous message.
But 2.4.16 is no good... :)

kribjo, 2025-02-04:
> So I am using 1.0.024 again and I can control my lights. Does anyone have 2.4.16 working with the left and right buttons or is that not working? When I was on 2.4.16 I tried binding the scenes but I couldn't get it to work.

My two STYRBAR E2001 controls work fine on 2.4.16 and with one TRÅDFRI LED2201G8 each, with Home Assistant and Z2M.
This means that they do turn off/on, dim up/down and change color temp just as they are supposed to with the standard button controls.
It was working for me on 1.x and then I upgraded to 2.4.16 and everything broke. After spending a lot of time and energy trying to fix it, things finally worked out after I took the leap and updated Z2M from 1.42.0 to 2.0.0. I had to make sure that "Settings -> Home  Assistant integration -> Home Assistant legacy action sensors" was enabled. In that settings tab I also have the main "Enabled" box checked, but "Home Assistant experimental event entities" is unchecked.

In the Bind tab of each controller it is set up like this:
Source endpoint: 1
Destination: Coordinator
Destination endpoint: 1
Clusters: LevelCtrl, OnOff, genPollCtrl, PowerCfg, Scenes

I use no other bindings than the one above.

Some more details in my comment here: https://github.com/Koenkk/zigbee2mqtt/issues/21963#issuecomment-2597019399

I have refrained from commenting here before, because I am afraid that I might be missing some nuance of the troubles reported here that would mean that my situation is not relevant. Perhaps it is so?

newlund, 2025-02-04:
Ah, perhaps the problem in 2.4.16 is caused by "Home Assistant legacy action sensors" being disabled.
But, I was under the impression that it should be disabled in Z2M 2.0.0 onwards?

newlund, 2025-02-04:
But still, 2.4.16 does not work with groups. So I will stay on 2.4.11 for now...
And I don't have to enable any legacy setting that might get removed in a later version of Z2M

kribjo, 2025-02-04:
> Ah, perhaps the problem in 2.4.16 is caused by "Home Assistant legacy action sensors" being disabled. But, I was under the impression that it should be disabled in Z2M 2.0.0 onwards?
> 
> Edit: No it was these settings that need to be disabled in Z2M 2.0.0:
> 
> advanced: homeassistant_legacy_entity_attributes: false homeassistant_legacy_triggers: false legacy_api: false legacy_availability_payload: false device_options: legacy: false

Work is being done to update relevant blueprints to use MQTT device triggers instead of legacy action sensors, conforming to the changes with Z2M 2.x. Until having switched to such updated versions, when using Z2M 2.x+, "legacy action sensors" need to be checked/true to not break the existing blueprints.

More information about that here:
- https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0
- https://github.com/Koenkk/zigbee2mqtt/discussions/24198
- https://github.com/yarafie/HA-BluePrints
- https://github.com/EPMatt/awesome-ha-blueprints/issues/622
- https://github.com/lsismeiro/awesome-ha-blueprints/issues/26

closed 2025-04-13 by github-actions[bot]

ireun, 2025-10-07:
Could someone help me to downgrade Styrbar E2002 to older version?
1. I'm using docker
2. in the `/app/data/` folder i've created 'index.json` file with this:
```json
[
  {
    "fileVersion": 33816593,
    "fileSize": 269638,
    "manufacturerCode": 4476,
    "imageType": 4555,
    "force": true,
    "sha512": "77c56399abe9fef1ce176c73e347be029686f7f0406639c31a80f56dfe634df176c8f216f3c40f95c01e2968244d57c09b72698fc618c92c99153820b7086292",
    "url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/styrbar_dimmer_zingo-0x11cb-2.4.11-prod.ota.ota.signed"
  }
]
```
3. in `/app/data/configuration.yaml` at the end i've added this:
```yaml
ota:
  zigbee_ota_override_index_location: /app/data/index.json
  disable_automatic_update_check: true
```
4. I've restarted zigbee2mqtt
5. after checking for OTA updates it shows this:

[image]

I'm using Z2M 2.6.0.

roberga, 2026-01-11:
I had the same problems with bindings to Trådfri lamps for left/right buttons for Styrbar E2002 after upgrading to frmware 2.4.16 and Styrbar E2313 on firmware 2.4.17. I managed to solve it in the following way:

1. Upgrade firmware
2. Interview the Styrbar device again after firmware upgrade. This added the genScenes binding to the device 
3. Bind genScenes to the lamp. This fixed the left/right buttons