Koenkk/zigbee2mqtt #25548: WZ1 ZigBee & RF 2CH LED Controller - WZ1 as ZigBee-RF converter to match RF LED controller or dimming driver - not working

mkbodanu4, 2025-01-05:
### What happened?

I bought an LED controller with a brand name Lisxchda, that looks exactly like WZ1 from Skydance.

For unknown reason, it detected by z2m as Tuya [L1(ZW)](https://www.zigbee2mqtt.io/devices/L1(ZW).html#tuya-l1(zw)), ZigBee model TS0501B, Manufacturer _TZB210_rkgngb5o

Anyway, I can control light, and dimming works well too. After I paired the RF control, I could also control light and dimming, but any action on the RF controller didn't impact the state in z2m.

After checking the manuals, I found that ZigBee-RF converter mode must be enabled to get it working correctly. So, I tried to switch to this mode without any success. So I connected the LED controller to the Tuya hub and tried the same, and now anything I do with the RF controller changes the state within the Tuya app. I don't want to use the Tuya hub, so having this functionality work in z2m will be great.

### What did you expect to happen?

ZigBee-RF converter mode to work correctly.

### How to reproduce it (minimal and precise)

After adding the LED controller into z2m and pairing it with the RF remote, a short press on the RF remote and the on-off state in z2m must enable ZigBee-RF converter mode.

Same actions within the Tuya app - pressing an on-off button at the same time as pressing the on-off button on the RF remote enables ZigBee-RF converter mode

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

7.4.5 [GA]

### Adapter

SLZB-06M with ember driver

### Setup

Arch Linux on Dell Wyse 5070

migor7, 2025-01-05:
I can confirm this problem

martin131, 2025-03-07:
Same problem with this device https://www.t-led.cz/p/prijimac-dimled-zigbee-pr-2k-69060
Do you have any solution please?

closed 2025-05-15 by github-actions[bot]