Koenkk/zigbee2mqtt #25554: Version 2.0.0. Z2M_WATCHDOG=default no longer works

IgnacioHR, 2025-01-05:
### What happened?

Updating to 2.0.0 when you have your system configured with Z2M_WATCHDOG=default causes a failure and a continuous start loop.

### What did you expect to happen?

service normal startup

### How to reproduce it (minimal and precise)

I've installed zigbee2mqtt service in a separate VM in proxmox. It is started by `systemd` using the following unit

```
# /etc/systemd/system/zigbee2mqtt.service
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Environment=Z2M_WATCHDOG=default
Type=notify
AssertCapability=CAP_NET_BIND_SERVICE
ExecStart=/usr/local/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling sys>
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=zigbee2mqtt

[Install]
WantedBy=multi-user.target
```

This was working with version 1.42 and doesn't work in version 2.0.0 maybe because the service is no longer informing systemd when it is really started.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

N/A

### Adapter

N/A

### Setup

N/A

### Debug log

I've restored a backup. Sorry, I've no logs available

m0wlheld, 2025-01-05:
There have been a lot of changes and some are breaking changes documented by #24198 . I'd guess that you are affected by this and the cause of failure is not watchdog, but somewhere else.

Please scan the z2m logs or journalctl to identify.

IgnacioHR, 2025-01-05:
Thank you for jumping in @m0wlheld 

Last time, I migrated from 1.X to 2.0 by executing `update.sh` I discovered I'd to install `pnpm` so I installed as described in their documentation so I continue executing all steps defined in `update.sh` one by one and and ended up in a system that couldn't start.

Now I've followed a different path that has been successfull:
1. Created `/opt/zigbee2mqtt2.0` directory
2. Followed instructions as described [https://www.zigbee2mqtt.io/guide/installation/01_linux.html](https://www.zigbee2mqtt.io/guide/installation/01_linux.html) to install a fresh copy of version 2.0
3. Copy files `/opt/zigbee2mqtt/data/configuration.yaml` `/opt/zigbee2mqtt/data/database.db` `/opt/zigbee2mqtt/data/state.json` to `/opt/zigbee2mqtt2.0/data`
4. Create directory `/opt/zigbee2mqtt2.0/data/log` (optional)
5. Stop running version 1.X using `systemctl stop zigbee2mqtt`
6. Rename directory `/opt/zigbee2mqtt2.0` to `/opt/zigbee2mqtt` 
7. Start version 2.0 using `systemctl start zigbee2mqtt`

And the system started up version 2.0 without any issues

closed 2025-01-05 by IgnacioHR