Koenkk/zigbee2mqtt #25555: Aqara RTCGQ11LM (lumi.sensor_motion.aq2), luminance is broken in 2.0

antst, 2025-01-05:
### What happened?

After upgrade to 2.0, it does not report luminance anymore and in HA it is indicated as disabled.

### What did you expect to happen?

to work as it was in 1.4

### How to reproduce it (minimal and precise)

RTCGQ11LM + Z2M 2.0.0

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

supervised HA+Z2M add-on

kashub, 2025-01-05:
A little info here, but to me sounds like it's there, just renamed, as per the information about breaking changes here: https://github.com/Koenkk/zigbee2mqtt/discussions/24198#:~:text=/UNLOCK.-,Renamed%20illuminance_lux%20to%20illuminance,-%2C%20this%20breaks%20in

m0wlheld, 2025-01-05:
> After upgrade to 2.0, it does not report luminance anymore and in HA it is indicated as disabled.

It does. Please read #24198 and adjust your entities.

closed 2025-01-05 by Koenkk

antst, 2025-01-05:
> > After upgrade to 2.0, it does not report luminance anymore and in HA it is indicated as disabled.
> 
> It does. Please read #24198 and adjust your entities.

My bad, after update, I had to enable manually luminance in HA

khoanong, 2025-01-07:
> > > After upgrade to 2.0, it does not report luminance anymore and in HA it is indicated as disabled.
> > 
> > 
> > It does. Please read #24198 and adjust your entities.
> 
> My bad, after update, I had to enable manually luminance in HA

How can do it ?