Koenkk/zigbee2mqtt #25556: Devices cannot be added to a group

thueske, 2025-01-05:
### What happened?

Since the update to Zigbee2MQTT 2.0.0, I can no longer add devices to an existing or new group. However, the devices as such function without any problems.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Docker

### Debug log

[image]

thueske, 2025-01-07:
I could cause the problem by adding devices to reused group IDs (happens automatically). If I create a group with an ID that has never existed before, I have no problems.

Koenkk, 2025-01-07:
Could you provide the debug log + your data/database.db?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

stefanmd023, 2025-01-17:
+1 to this issue

tungmeister, 2025-02-01:
I'm also seeing this with a group, did you find a solution?

rickvdl, 2025-02-10:
Same issue here. The workaround I found was renaming the old group to something else, and then creating a new group with another ID.

colmben, 2025-02-10:
Same for me - I wonder how they are being added back to the DB after deleting them?

closed 2025-04-19 by github-actions[bot]