Koenkk/zigbee2mqtt #25557: Zigbee2MQTT in docker on Synology NAS running DSM 7.2 can no longer see /dev/ttyUSB0

matt05d, 2025-01-05:
### What happened?

Zigbee2MQTT in docker on Synology NAS running DSM 7.2 can no longer see /dev/ttyUSB0

I believe the issue started after updating to the latest Synology DSM release:
Version: 7.2.2-72806 Update 2

Prior to this, no issues.
I ran a new container with the latest-dev image.
I saw I did not have the adapter line in my config so added that.
Then I got the same error I was seeing on latest:

zh:ember:uart:ash: Failed to init port with error Error: Error: No such file or directory, cannot open /dev/ttyUSB0

See all other startup logs in the docker logs attached.

I confirmed my container manager runs as root and executed a write test to the ttyUSB0 which was successful:
```
Matt@Home_Storage:/dev$ test -w /dev/ttyUSB0 && echo success || echo failure
success
```

### What did you expect to happen?

Successful start up

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

Unknown

### Adapter

Sonoff 3.0 Zigbee Dongle

### Setup

Container in Synology NAS

### Debug log

<html>
<body>

date | stream | content
-- | -- | --
2025/01/05 12:49:35 | stdout | at start (/app/index.js:161:5)
2025/01/05 12:49:35 | stdout | at Controller.start (/app/lib/controller.ts:142:13)
2025/01/05 12:49:35 | stdout | at Zigbee.start (/app/lib/zigbee.ts:69:27)
2025/01/05 12:49:35 | stdout | at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
2025/01/05 12:49:35 | stdout | at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
2025/01/05 12:49:35 | stdout | at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  z2m: Exiting...
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  z2m: Failed to start zigbee-herdsman
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  z2m: Error while starting zigbee-herdsman
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [31merror[39m:  zh:ember:uart:ash: Failed to init port with error Error: Error: No such file or directory, cannot open /dev/ttyUSB0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash: ======== ASH stopped ========
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   ACK timeouts    : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Out of sequence : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Retry dupes     : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Out of buffers  : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Bad ACK numbers : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Bad lengths     : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Bad controls    : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Length > maximum: RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Length < minimum: RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Comm errors     : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   CRC errors      : RX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   ACK frames  : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Retry frames: RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   DATA bytes  : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   DATA frames : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Cancelled   : RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash:   Total frames: RX=0, TX=0
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash: ASH COUNTERS since last clear:
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember:ezsp: ======== EZSP starting ========
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember: ======== Ember Adapter Starting ========
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  zh:ember: Using default stack config.
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  z2m: Starting zigbee-herdsman (3.2.1)
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  z2m: Starting Zigbee2MQTT version 2.0.0-dev (commit #43a5b24dcc22af3ec4c0bb76b45a2706f898421f)
2025/01/05 12:49:35 | stdout | [2025-01-05 12:49:35] [32minfo[39m:  z2m: Logging to console, file (filename: log.log)
2025/01/05 12:49:33 | stdout | Starting Zigbee2MQTT without watchdog.
2025/01/05 12:49:33 | stdout | Using '/app/data' as data directory
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364
2025/01/05 12:47:21 | stdout | at start (/app/index.js:161:5)
2025/01/05 12:47:21 | stdout | at Controller.start (/app/lib/controller.ts:142:13)
2025/01/05 12:47:21 | stdout | at Zigbee.start (/app/lib/zigbee.ts:69:27)
2025/01/05 12:47:21 | stdout | at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
2025/01/05 12:47:21 | stdout | at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
2025/01/05 12:47:21 | stdout | at processTicksAndRejections (node:internal/process/task_queues:105:5)
2025/01/05 12:47:21 | stdout | at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: Exiting...
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: Failed to start zigbee-herdsman
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [31merror[39m:  z2m: Error while starting zigbee-herdsman
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [32minfo[39m:  z2m: Starting zigbee-herdsman (3.2.1)
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [32minfo[39m:  z2m: Starting Zigbee2MQTT version 2.0.0-dev (commit #43a5b24dcc22af3ec4c0bb76b45a2706f898421f)
2025/01/05 12:47:21 | stdout | [2025-01-05 12:47:21] [32minfo[39m:  z2m: Logging to console, file (filename: log.log)
2025/01/05 12:47:17 | stdout | Starting Zigbee2MQTT without watchdog.
2025/01/05 12:47:16 | stdout | Creating configuration file...
2025/01/05 12:47:16 | stdout | Using '/app/data' as data directory

</body>
</html>

kashub, 2025-01-05:
This is a synology issue, not Z2M - this should help you: https://github.com/robertklep/dsm7-usb-serial-drivers

matt05d, 2025-01-05:
Hi @kashub, thanks for the reply

That was an issue when I first set up ZigBee, however it manifests as ttyUSB0 not showing up when the device is plugged in and after adding the drivers using the link you sent it shows up. 
TtyUSB0 is shown and I can write to it with root as shown from the success output in my post.
So while this may still be a Synology issue, it's not that one and I'm not sure why I would be able to write to it and see it but Z2M can't if it's a Synology issue

kashub, 2025-01-05:
> Hi @kashub, thanks for the reply
>
> That was an issue when I first set up ZigBee, however it manifests as ttyUSB0 not showing up when the device is plugged in and after adding the drivers using the [...]

Sorry, I didn't read it all properly by the look of it. It still seems synology issue though, not z2m. I'm having "fun" with the usb in synology after every update i do, so maybe this one is yet waiting for me ;)

matt05d, 2025-01-05:
Ah okay thanks, well aside from the first time having to setup the startup script for drivers this is my first issue but I have sent a ticket in to Synology anyway just in case!

ralphklos, 2025-01-05:
i can comfirm this issue and do not believe this is a Synology issue.
Drivers for usb are configured right, and the device does show in the output. as of version 2 it does not work anymore, going back to version 1.42.0 makes things working again. Not sure if it's due to Sonoff firmware that is not compatible anymore with v2 ?

matt05d, 2025-01-06:
Okay seems like I can get past the not discovering ttyUSB0 by starting the container from the command line and explicitly passing it the device:

`docker run --name Zigbee2MQTT --network host --device /dev/ttyUSB0 -v /volume1/docker/zigbee2mqtt/data:/app/data:rw -e TZ=Europe/London --restart always -d koenkk/zigbee2mqtt:latest-dev`

This isn't something I've had to do before so is at least one change that I think is associated with a recent Synology update.

However, it unfortunately didn't get me out of the woods.
On 1.42, latest and latest-dev I now get this behaviour:

`2025/01/06 09:35:34 | stdout | at start (/app/index.js:161:5)
-- | -- | --
2025/01/06 09:35:34 | stdout | at Controller.start (/app/lib/controller.ts:142:13)
2025/01/06 09:35:34 | stdout | at Zigbee.start (/app/lib/zigbee.ts:69:27)
2025/01/06 09:35:34 | stdout | at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
2025/01/06 09:35:34 | stdout | at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
2025/01/06 09:35:34 | stdout | at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
2025/01/06 09:35:34 | stdout | [2025-01-06 09:35:34] [31merror[39m:  z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
2025/01/06 09:35:34 | stdout | [2025-01-06 09:35:34] [31merror[39m:  z2m: Exiting...
2025/01/06 09:35:34 | stdout | [2025-01-06 09:35:34] [31merror[39m:  z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
2025/01/06 09:35:34 | stdout | [2025-01-06 09:35:34] [31merror[39m:  z2m: Failed to start zigbee-herdsman
2025/01/06 09:35:34 | stdout | [2025-01-06 09:35:34] [31merror[39m:  z2m: Error while starting zigbee-herdsman
2025/01/06 09:35:31 | stdout | [2025-01-06 09:35:31] [32minfo[39m:  zh:ember:uart:ash: ======== ASH starting ========
2025/01/06 09:35:31 | stdout | [2025-01-06 09:35:31] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/06 09:35:29 | stdout | [2025-01-06 09:35:29] [32minfo[39m:  zh:ember:uart:ash: ======== ASH starting ========
2025/01/06 09:35:29 | stdout | [2025-01-06 09:35:29] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/06 09:35:26 | stdout | [2025-01-06 09:35:26] [32minfo[39m:  zh:ember:uart:ash: ======== ASH starting ========
2025/01/06 09:35:26 | stdout | [2025-01-06 09:35:26] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/06 09:35:24 | stdout | [2025-01-06 09:35:24] [32minfo[39m:  zh:ember:uart:ash: ======== ASH starting ========
2025/01/06 09:35:24 | stdout | [2025-01-06 09:35:24] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember:uart:ash: ======== ASH starting ========
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember:uart:ash: Serial port opened
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember:uart:ash: ======== ASH Adapter reset ========
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember:ezsp: ======== EZSP starting ========
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember: ======== Ember Adapter Starting ========
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  zh:ember: Using default stack config.
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  z2m: Starting zigbee-herdsman (3.2.1)
2025/01/06 09:35:22 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  z2m: Starting Zigbee2MQTT version 2.0.0-dev (commit #43a5b24dcc22af3ec4c0bb76b45a2706f898421f)
2025/01/06 09:35:21 | stdout | [2025-01-06 09:35:21] [32minfo[39m:  z2m: Logging to console, file (filename: log.log)
2025/01/06 09:35:19 | stdout | Starting Zigbee2MQTT without watchdog.
2025/01/06 09:35:19 | stdout | Using '/app/data' as data directory
2025/01/06 09:35:11 | stdout | at start (/app/index.js:161:5)
2025/01/06 09:35:11 | stdout | at Controller.start (/app/lib/controller.ts:142:13)
2025/01/06 09:35:11 | stdout | at Zigbee.start (/app/lib/zigbee.ts:69:27)
2025/01/06 09:35:11 | stdout | at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
2025/01/06 09:35:11 | stdout | at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
2025/01/06 09:35:11 | stdout | at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
2025/01/06 09:35:11 | stdout | [2025-01-06 09:35:11] [31merror[39m:  z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
2025/01/06 09:35:11 | stdout | [2025-01-06 09:35:11] [31merror[39m:  z2m: Exiting...
[... cut ...]
2025/01/06 09:34:57 | stdout | [2025-01-06 09:34:57] [32minfo[39m:  z2m: Starting Zigbee2MQTT version 2.0.0-dev (commit #43a5b24dcc22af3ec4c0bb76b45a2706f898421f)
2025/01/06 09:34:57 | stdout | [2025-01-06 09:34:57] [32minfo[39m:  z2m: Logging to console, file (filename: log.log)
2025/01/06 09:34:43 | stdout | Starting Zigbee2MQTT without watchdog.
2025/01/06 09:34:42 | stdout | Using '/app/data' as data directory`

So it seems to be able to open the serial port but doesn't get much further.

cub0head, 2025-01-06:
Today faced the same problem. For me it helped to change from "adapter: auto" to "adapter: zstack" in configuration.yaml

cub0head, 2025-01-06:
Hope it helps
[configuration.yaml.txt](https://github.com/user-attachments/files/18324711/configuration.yaml.txt)

closed 2025-03-15 by github-actions[bot]