Koenkk/zigbee2mqtt #25558: Unable to Set Log Level to ‘warn’ in Zigbee2MQTT Integration

zhuykovkb, 2025-01-05:
### What happened?

When attempting to change the log level to warn in Zigbee2MQTT via the Home Assistant interface at
http://ha_url/45df7312_zigbee2mqtt/ingress -> Logs -> Log level, the log level does not update, and an error appears in the logs:

```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:125
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))  
First occurred: 13:38:21 (2 occurrences)  
Last logged: 13:56:32  

Invalid option for select.zigbee2mqtt_bridge_log_level: 'warn' (valid options: ['error', 'warning', 'info', 'debug'])  

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

sonoff dongle P

### Setup

z2m addon in ha supervised

m0wlheld, 2025-01-05:
> Invalid option for select.zigbee2mqtt_bridge_log_level: 'warn' (valid options: ['error', 'warning', 'info', 'debug'])  

Quite obvious, isn't it? If you cannot pick "warning" from the Home Assistant Add-On interface, please report this over [there](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt).

closed 2025-01-05 by zhuykovkb

uffevedenbrant, 2025-01-06:
For me zigbee2mqtt crashes , most likely after this message. I think that I changed the log levels while doing some tests with a sensor and now it seems impossible to find the offending configuration. Ill try to remove all configuration including the configuration directory to see if that helps.. ( tried a "grep warn *" there to find the possible config line but so far no success.. )

Starting Zigbee2MQTT without watchdog.
AssertionError [ERR_ASSERTION]: 'warn' is not valid log_level, use one of 'error, warning, info, debug'
    at Logger.init (/app/lib/util/logger.ts:41:15)
    at new Controller (/app/lib/controller.ts:79:16)
    at start (/app/index.js:159:18)

uffevedenbrant, 2025-01-06:
Sorry.. Did a new check and found the offending configuration.. 
configuration.yaml

➜  zigbee2mqtt grep warn *
configuration.yaml:  log_level: warn
➜  zigbee2mqtt vi configuration.yaml 
➜  zigbee2mqtt sync               

This was changed via the zigbee2mqtt Web-GUI.. I.e. probably something there that sets "warn" instead of "warning" that makes zigbee2mqtt later to crash with the previous error..

Should i report this as a new issue or should it be kept here?
As this might be a regexp issue on the log module as well if someone changed log level from accepting "warn" earlier instead of the full word "warning"

Koenkk, 2025-01-06:
The `warn` log level has been removed, update it to `warning` in your `configuration.yaml`, I just fixed the frontend in https://github.com/Koenkk/zigbee2mqtt/pull/25634

uffevedenbrant, 2025-01-06:
To clarify..

The setting to be changed exists in config/zigbee2mqtt/configuration.yaml if anyone finds this post and have this issue.

Thanks for the quick fix!

zhuykovkb, 2025-01-06:
thank you @Koenkk!

vgdh, 2025-08-16:
> anyone finds this post and have this issue.
> 
> Thanks for the quick fix!

Why is the config on two different locations at the same time?
1. config/zigbee2mqtt/configuration.yaml
2. In the addon configuration page?

And logging configuration only works from the first location.