Koenkk/zigbee2mqtt #25560: You forgot some actions (double_both, hold_right, hold_both) Aqara

Bagunda, 2025-01-05:
### What happened?

Switch [Aqara WXKG02LM_rev2](https://www.zigbee2mqtt.io/devices/WXKG02LM_rev2.html#aqara-wxkg02lm-rev2).
Actions `double_both`, `hold_right`, `hold_both` are not in the list "trigger".
Yesterday I updated z2m to version 2.0.0.
Before that - actions `double_both`, `hold_right`, `hold_both` worked.

[image]

And in "[Aqara WXKG07LM](https://www.zigbee2mqtt.io/devices/WXKG07LM.html#aqara-wxkg07lm)" there are not enough "actions" and the names are mixed up:
| IN PAGE      | IN LIST OF TRIGGER IN HA |
|--------------|--------------------------|
| single_left  | left, single_lef         |
| single_right | right                    |
| single_both  | both                     |
| double_left  | left_double              |
| double_right | right_double             |
| double_both  | both_double              |
| hold_left    |                          |
| hold_right   |                          |
| hold_both    |                          |

### What did you expect to happen?

I expected all `actions` to be available in the `trigger` list. Or that the page would describe that they don't work after the update.

### How to reproduce it (minimal and precise)

Create a new automation, in the "When" block select the "Device" trigger, in the "Device" list select the WXKG02LM_rev2 switch. Expand the "Trigger" list. And there you can see the missing "Actions"

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

0.6.10

### Adapter

ZigStar

### Setup

Add-on on Home Assistant Core 2025.1.0 Supervisor 2024.12.3 x86_64

KossikCZ, 2025-01-05:
@Bagunda Did you try executing those actions on your Aqara switch? I noticed my Aqara devices show up the actions/triggers once I actually executed them on the switch/button.

Bagunda, 2025-01-09:
I confirm. If you call the necessary actions on the switch, they appear in the list. I think that this should be indicated in the instructions.
[image]

Koenkk, 2025-01-10:
It's described here: https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended

> The [MQTT device triggers](https://www.home-assistant.io/integrations/device_trigger.mqtt/) are discovered by Zigbee2MQTT once the event is triggered on the device at least once.

closed 2025-01-10 by Koenkk