Koenkk/zigbee2mqtt #25561: Aqara SRTS-A01: child_lock on/off values have been changed from true/false to LOCK/UNLOCK does not work !!

DaN66O, 2025-01-05:
### What happened?

Here
https://github.com/Koenkk/zigbee2mqtt/discussions/24198
it is written about this change.
But it doesn't work.
Interview and Reconfigure doesn't help
There is an error in the log and the switch is not happening.
`error 2025-01-05 16:36:21z2m: Publish 'set' 'child_lock' to 'radiator_koridor2' failed: 'Error: Expected boolean, got: string'`

### What did you expect to happen?

Normal operation of LOCK/UNLOCK

### Zigbee2MQTT version

2.0.0 commit: 060ae99

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

plain

### Debug log

error 2025-01-05 16:36:21z2m: Publish 'set' 'child_lock' to 'radiator_koridor2' failed: 'Error: Expected boolean, got: string'

LedLeoHA, 2025-01-05:
Adding that i have the same issue.
- Right now in Home assistant, if "Child lock" is turned on, it just switches off
- In Z2M, trying to turn it on returns error z2m: Publish 'set' 'child_lock' to 'Office Radiator' failed: 'Error: Expected boolean, got: string'

freeDom-, 2025-01-08:
Same problem with all Aqara SRTS-A01 HVACs.

Koenkk, 2025-01-08:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-08 by Koenkk