Koenkk/zigbee2mqtt #25562: Update from 1.42.0 to 2.0.0 not working

Eistee82, 2025-01-05:
### What happened?

I get an error when updating:

[image]

### What did you expect to happen?

no error

### How to reproduce it (minimal and precise)

I have a debian linux installation
using update.sh like in the doc

### Zigbee2MQTT version

 1.42.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

plain

### Debug log

Checking Zigbee2MQTT status...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating bc6eccf6..060ae99c
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting

m0wlheld, 2025-01-05:
Please read the releases notes, your „issue“ is covered there.

Sabjul33, 2025-01-05:
Answer here : 😉

Bare-metal installations (no Docker/Home Assistant add-on)
For Git based installations: removed configuration.yaml from repo, this will result in the following error when updating:
```
error: Your local changes to the following files would be overwritten by merge:
  data/configuration.yaml
```
To fix this, update by executing:
```
git checkout data/configuration.example.yaml
mv data/configuration.yaml data/configuration.yaml.bak
./update.sh # (update as you usually do)
mv data/configuration.yaml.bak data/configuration.yaml
```

Marcoletto, 2025-01-05:
> Please read the releases notes, your „issue“ is covered there.

I've read the notes, I' ve changed the configuration.yaml and with the 1.42.0-2 everythings work... when I update @ 2.0.0-1 the Z2M does not work.

I' don't know how to solve

closed 2025-01-05 by Eistee82