Koenkk/zigbee2mqtt #25567: Z2M not starting entirely

indi81, 2025-01-05:
### What happened?

After updating my Z2M seems to have issues starting entirely. I have gotten it to run ocassionaly but most of the time it just sits there, no response from devices, no access to the UI and no errors in the logs.

### What did you expect to happen?

It would complete its startup and be functional.

### How to reproduce it (minimal and precise)

Stop, start and/or restart of Z2M in Home Asssistant.

### Zigbee2MQTT version

2.0.0 commit unknown

### Adapter firmware version

7.4.4

### Adapter

Skyconnecct

### Setup

Home Assistant OS add-on

### Debug log

Short excerpt of the last rows from of a forced stop, logging one error regarding the frontend,

```
[2025-01-05 15:32:37] info: 	z2m: Currently 74 devices are joined.
[2025-01-05 15:32:37] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-05 15:32:38] info: 	z2m: Connected to MQTT server
[2025-01-05 15:32:38] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-05 15:35:13] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
[2025-01-05 15:35:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-05 15:35:13] info: 	z2m: Disconnecting from MQTT server
[2025-01-05 15:35:13] info: 	z2m: Stopping zigbee-herdsman...
[2025-01-05 15:35:21] info: 	zh:controller: Wrote coordinator backup to '/config/zigbee2mqtt/coordinator_backup.json'
[2025-01-05 15:35:21] info: 	zh:zstack:znp: closing
[2025-01-05 15:35:21] info: 	zh:zstack:znp: Port closed
[2025-01-05 15:35:21] info: 	z2m: Stopped zigbee-herdsman
[2025-01-05 15:35:21] info: 	z2m: Stopped Zigbee2MQTT
```

indi81, 2025-01-05:
Startup, it just hangs after the "bridge online" part,

```
[15:37:07] INFO: Preparing to start...
[15:37:07] INFO: Socat not enabled
[15:37:07] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 15:37:12] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 15:37:12] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 15:37:12] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 15:37:12] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Silicon Labs","serialNumber":"0001","pnpId":"usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","vendorId":"10c4","productId":"ea60"} => zstack: 4
[2025-01-05 15:37:12] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2025-01-05 15:37:12] info: 	zh:zstack:znp: Serialport opened
[2025-01-05 15:37:13] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-05 15:37:13] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
[2025-01-05 15:37:13] info: 	z2m: aqara.motion3 (0x00158d000836d2c8): RTCGQ11LM - Aqara Motion sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.door1 (0x00158d0008a9fedc): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aeotec.range.extender2 (0xdc8e95fffe92ab33): WG001 - Aeotec Range extender Zi (Router)
[2025-01-05 15:37:13] info: 	z2m: aqara.vibration1 (0x00158d000807dd56): DJT11LM - Aqara Vibration sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.button1 (0x00158d0008f31f21): WXKG11LM - Aqara Wireless mini switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.dial.8 (0x001788010d3501f9): 8719514440937/8719514440999 - Philips Hue Tap dial switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: danfoss.ally.termostat1 (0xdc8e95fffe216c39): 014G2461 - Danfoss Ally thermostat (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: mhcozy.4relay.1 (0xa4c1380df5d55417): TYWB 4ch-RF - MHCOZY 4 channel relay (Router)
[2025-01-05 15:37:13] info: 	z2m: danfoss.ally.termostat2 (0x70ac08fffe6e5d30): 014G2461 - Danfoss Ally thermostat (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote3 (0x943469fffe7969b4): E1743 - IKEA TRADFRI on/off switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: gledopto.led1 (0x60a423fffe042192): GL-C-008P - Gledopto Zigbee LED Controller RGB+CCT (pro) (Router)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.dial1 (0x001788010d34f4b3): 8719514440937/8719514440999 - Philips Hue Tap dial switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.motion4 (0x00158d00083a1465): RTCGQ11LM - Aqara Motion sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.vibration2 (0x00158d000805a79d): DJT11LM - Aqara Vibration sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote1 (0xf4b3b1fffe9a3a84): E1743 - IKEA TRADFRI on/off switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.motionp1.1 (0x54ef4410004e4227): RTCGQ14LM - Aqara Motion sensor P1 (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.motionp1.2 (0x54ef4410004e4dfa): RTCGQ14LM - Aqara Motion sensor P1 (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aeotec.range.extender1 (0xdc8e95fffe956fb9): WG001 - Aeotec Range extender Zi (Router)
[2025-01-05 15:37:13] info: 	z2m: aqara.dorr2 (0x00158d0008a9f210): MCCGQ11LM - Aqara Door and window sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: idlock.door1 (0x04cd15fffe7c0b3d): 0402946 - Datek Zigbee module for ID lock (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote2 (0x943469fffe3d8c43): E1743 - IKEA TRADFRI on/off switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote4 (0xcc86ecfffe920de1): E1766 - IKEA TRADFRI open/close remote (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.fyrtur.blind2 (0x847127fffe8f2d9c): E1757 - IKEA FYRTUR roller blind, block-out (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.fyrtur.blind3 (0xcc86ecfffe19e1f0): E1757 - IKEA FYRTUR roller blind, block-out (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.fyrtur.blind1 (0xcc86ecfffe929a41): E1757 - IKEA FYRTUR roller blind, block-out (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote5 (0xcc86ecfffe95975c): E1766 - IKEA TRADFRI open/close remote (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: ikea.tradfri.remote6 (0xf4b3b1fffe9e039a): E1743 - IKEA TRADFRI on/off switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.e14.2 (0x001788010b2e0b09): 9290024406 - Philips Hue P45 light bulb (Router)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.e14.1 (0x001788010c307a17): 9290024406 - Philips Hue P45 light bulb (Router)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.dial.3 (0x001788010d359c93): 8719514440937/8719514440999 - Philips Hue Tap dial switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.dial.4 (0x001788010d35ad7c): 8719514440937/8719514440999 - Philips Hue Tap dial switch (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: plusled.led.driver2 (0x94deb8fffe7f4a1d): HK-ZD-CCT-A - Sunricher 50W Zigbee CCT LED driver (constant current) (Router)
[2025-01-05 15:37:13] info: 	z2m: philips.hue.motion1 (0x001788010cd6da54): 9290030675 - Philips Hue motion sensor (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: aqara.motion1 (0x00158d000836de23): RTCGQ11LM - Aqara Motion sensor (EndDevice)
[... cut ...]
[2025-01-05 15:37:13] info: 	z2m: nous.powerplug1 (0xa4c138ab98481bd6): A1Z - Nous Smart plug (with power monitoring) (Router)
[2025-01-05 15:37:13] info: 	z2m: fingerbot.dator.nordlo (0xa4c138189399184c): TS0001_fingerbot - Tuya Zigbee fingerbot plus (EndDevice)
[2025-01-05 15:37:13] info: 	z2m: Currently 74 devices are joined.
[2025-01-05 15:37:13] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-05 15:37:13] info: 	z2m: Connected to MQTT server
[2025-01-05 15:37:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
```

indi81, 2025-01-05:
If it eventually works it could be several minutes (1-5 minutes) until it comes online wih the following lines on the same second,

info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/901/availability', payload '{"state":"online"}'
info: 	z2m: Started frontend on port 8099

RyoShinzo, 2025-01-05:
What device do you use? I have a Raspberry Pi 3b+ and can't update from 1.41. If I do, Z2M will crash every few minutes. Probably a resource issue? Because of this and other things, I will upgrade to a Beelink EQ14 soon.

indi81, 2025-01-05:
> What device do you use? I have a Raspberry Pi 3b+ and can't update from 1.41. If I do, Z2M will crash every few minutes. Probably a resource issue? Because of this and other things, I will upgrade to a Beelink EQ14 soon.

I have Raspberry Pi 4 with 4GB. Never had any issues before and when Z2m is running eventually it works perfectly fine, just that it seems to lock up on something during startup. 

CPU idles at around 9-15% and RAM usage is 1.7GB.

solidfox, 2025-01-05:
Seeing the same behaviour with an RPi 5 8GB.

solidfox, 2025-01-05:
[45df7312_zigbee2mqtt_2025-01-05T16-22-49.986Z.log](https://github.com/user-attachments/files/18310891/45df7312_zigbee2mqtt_2025-01-05T16-22-49.986Z.log)

maxandersen, 2025-01-05:
same issue for me.

I added the zstack suggested adapter line and other homeassistant settigns to get it to get pass device detection - tubezb ethernet adapter.

But now hitting this with error:

```
[2025-01-05 20:48:42] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /app/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/app/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.stop (/app/lib/controller.ts:233:9)
    at stop (/app/index.js:172:5)
    at process.handleQuit (/app/index.js:180:13))
```

indi81, 2025-01-06:
I've changed my original post, I am using Sonoff_Zigbee_3.0_USB_Dongle_Plus and not SkyConnect on this one.

zstack is used as adapter.

indi81, 2025-01-06:
Whenever it starts it seems stable, its just when the Z2M starts it seems to lock up. I'd say most of the time it locks up, very rare it starts.

indi81, 2025-01-06:
After updating to 2.0.0.2 which says it has a crash fix for RPi4 it now seems to be solid again. Tried restarting 4 times and it has always started without any longer delays (a couple of seconds only).

closed 2025-01-06 by indi81