Koenkk/zigbee2mqtt #25568: TS0044 not working

Jos-77, 2025-01-05:
### What happened?

after update to 2.0 TS0044 not working any longer. Only showing battery and link. Action is N/A.

Please fix

### What did you expect to happen?

actions still work

### How to reproduce it (minimal and precise)

After update no actionbuttons working. Showing as N/A

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

 20221226

### Adapter

sonoff zigbee usb

### Setup

Home Assistant OS 

### Debug log

[2025-01-05 15:31:15] info: 	z2m: Studeren Tuya 12 scene (0x70ac08fffe287167): TS0044 - Tuya Wireless switch with 4 buttons (EndDevice)
[2025-01-05 15:31:16] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Studeren Tuya 12 scene', payload '{"battery":97,"linkquality":25,"voltage":2900}'

m0wlheld, 2025-01-05:
Please, read the release notes and the breaking changes at #24198. After that, you know how to resolve this „issue“.

gianniocchipinti, 2025-01-05:
Hi

have you solved?

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - All action sensors are now disabled by default (`sensor.*_action` entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your `configuration.yaml`.
>   ```yaml
>   homeassistant:
>     legacy_action_sensor: true
>   ```

closed 2025-01-05 by Koenkk

gianniocchipinti, 2025-01-05:
[image]
Thank you. How do I implement this?

Via MQTT device trigger (recommended)

Device ID???
where do I find the device ID?

gianniocchipinti, 2025-01-05:
> This is one of the Zigbee2MQTT 2.0.0 breaking changes, from #24198

[image]
Thank you. How do I implement this?

Via MQTT device trigger (recommended)

Device ID???
where do I find the device ID?

Hardy6itsix, 2025-01-05:
i read the release notes and the breaking changes at https://github.com/Koenkk/zigbee2mqtt/discussions/24198....

but still don't understand how to fix this issue. Could someone help to fix it.

Jos-77, 2025-01-06:
@Koenkk I really don't understand why something that worked perfect is removed. doing it by the recommended way it means I would have to config 10 devices with each 12 functions and some other switches with 4 functions. So over 150 button actions.

Already tried the legacy_action_ sensor: true function, did isn't working. Also replaced config file with new one, no succes. Z2M was not running. At last I tried a complete reinstall after deleting all files. Also not working.

Since today the update to 2.0.. is not available in Home Assistant OS. Why??

gianniocchipinti, 2025-01-06:
solved
[image]

CocoLeo3680, 2025-03-09:
> solved [image]
> 
> [image]

Many thanks!

Same problem solved :)