Koenkk/zigbee2mqtt #25569: Pairing button doesn't work

MasterChief1335, 2025-01-05:
### What happened?

I can't click the pairing button

error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')

### What did you expect to happen?

That it works?! 🤡🙄

### How to reproduce it (minimal and precise)

The error doesn't even go away🤡🙄

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

current

### Adapter

Conbee

### Setup

Proxmox HP400 G7 SSF

### Debug log

[15:54:33] INFO: Preparing to start...
[15:54:33] INFO: Socat not enabled
[15:54:33] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 15:54:35] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 15:54:35] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 15:54:35] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 15:54:35] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-05 15:54:35] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"product":0,"revision":"0x26780700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2025-01-05 15:54:35] info: 	z2m: Currently 0 devices are joined.
[2025-01-05 15:54:35] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-05 15:54:35] info: 	z2m: Connected to MQTT server
[2025-01-05 15:54:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-05 15:54:35] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-05 15:54:35] info: 	z2m: Started frontend on port 8099
[2025-01-05 15:54:35] error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at HomeAssistant.getBridgeEntity (/app/lib/extension/homeassistant.ts:1918:78)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:439:28)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-05 15:54:35] info: 	z2m: Zigbee2MQTT started!
[2025-01-05 15:54:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'

MasterChief1335, 2025-01-06:
?

MasterChief1335, 2025-01-07:
Don't roll over with help......

mduplifting, 2025-01-13:
Exactly the same with rpi 4b & conbee II :(

Did you figured out something?

MrGennae, 2025-01-26:
Same error here! Have you found a fix for that ?
I'm using a conbee2 key and HAOS

giovannigenna, 2025-01-26:
Same error here! Have you found a fix for that ?
I'm using a conbee2 key and HAOS

closed 2025-04-04 by github-actions[bot]