Koenkk/zigbee2mqtt #25570: migration creates v1 v2 v3 yaml and 1-2, 2-3, and 3-4 logs (everything works though)

y8s, 2025-01-05:
### What happened?

I was running latest-dev docker 1.41.0 and hit the update button in dockge (i.e. pulled the new version) and then realized I probably don't need the dev version so I edited the compose file and removed -dev to run just the latest tag. I stopped and restarted the container and saw the file list included:

```
-rw-r--r-- 1 pi   pi   2.2K Jan  5 10:03 configuration.yaml
-rw-r--r-- 1 root root 2.2K Jan  5 10:03 configuration_backup_v3.yaml
-rw-r--r-- 1 root root  120 Jan  5 10:03 migration-3-to-4.log
-rw-r--r-- 1 root root  303 Jan  5 10:03 migration-2-to-3.log
-rw-r--r-- 1 root root 2.2K Jan  5 10:03 configuration_backup_v2.yaml
-rw-r--r-- 1 root root  475 Jan  5 10:03 migration-1-to-2.log
-rw-r--r-- 1 root root 2.4K Jan  5 10:03 configuration_backup_v1.yaml

```
Which was funny. the logs contain only a few lines each, just spread across them:

```
==> migration-1-to-2.log <==
[ADDITION] Migrated settings to version 2

[REMOVAL] The permit_join setting has been removed, use the frontend or MQTT to permit joining.

[REMOVAL] Removed configuring group members through configuration.yaml (groups.xyz.devices setting). This will not impact current group members; however, you will no longer be able to add or remove devices from a group through the configuration.yaml.

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
==> migration-2-to-3.log <==
[CHANGE] Migrated settings to version 3

[SPECIAL] Property 'homeassistant' is now always an object.

[SPECIAL] Property 'frontend' is now always an object.

[SPECIAL] Property 'availability' is now always an object.

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
==> migration-3-to-4.log <==
[CHANGE] Migrated settings to version 4

For more details, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198
```

### What did you expect to happen?

just to see a single backup of my configuration.yaml and one migration-1-2.log

### How to reproduce it (minimal and precise)

see above. update docker and then stop and restart docker (using latest-dev tag)

### Zigbee2MQTT version

1.41.0 --> 2.0.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

eWelink Zigbee 3.0 Gateway ZB-GW03 (tasmota 14.2.0.4)

### Setup

running in docker on raspberry pi, managed through dockge

### Debug log

not really useful?

m0wlheld, 2025-01-05:
Just remove them after upgrade, why is this an issue?

y8s, 2025-01-05:
> Just remove them after upgrade, why is this an issue?

This is unintended behavior. It's ambiguous. I'm letting the devs know.

Koenkk, 2025-01-05:
That's expected, due to many changes we had 4 config migrations when going from 1.42.0 to 2.0.0, in the future this will likely be 1 migration per version increment (and for most versions there will be nothing at all).

y8s, 2025-01-05:
Thanks @Koenkk . I was confused because the from/to versions were weird. I would have understood if it was `migration-1.41.0-1.42.0.log` and then `migration-1.42.0-2.0.0.log` or something.

Koenkk, 2025-01-05:
That's because the config version does not have to align with the z2m version (since the migration can be per commit, otherwise you cannot migrate people running the dev branch)

closed 2025-03-14 by github-actions[bot]