Koenkk/zigbee2mqtt #25571: Upgrading to 2.0.0 disables illuminance property of Hue motion sensors

lawrencedudley, 2025-01-05:
### What happened?

Upgraded to 2.0.0

Before the upgrade, illuminance worked fine.

After the upgrade, illuminance properties showed as disabled. They can be enabled again in Home Assistant and appear to work but I reckon this is probably undesired behaviour.

[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

7.4.4

### Adapter

Skyconnect/ZBT-1

### Setup

Add-on on home assistant

m0wlheld, 2025-01-05:
> What happened?

You did ignore the release notes….

Nihvel, 2025-01-05:
Happened to[ Xiaomi GZCGQ01LM](https://www.zigbee2mqtt.io/devices/GZCGQ01LM.html#xiaomi-gzcgq01lm) too, but the release notes do not mention that an illuminance sensor gets its only useful sensor... disabled.

https://github.com/Koenkk/zigbee2mqtt/releases

Here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8304

m0wlheld, 2025-01-05:
The release notes have a link to the breaking changes #24198 (the red one at the top). Following this link, the following Information is provided:

> Renamed illuminance_lux to illuminance, this breaks in two ways: illuminance_lux has been removed and the value of illuminance changes. Home Assistant users might have to manually enable this sensor via Home Assistant. This affects the following devices: 83633205, MOSZB-140, MOSZB-153, EFEKTA_iAQ3, EFEKTA_PWS_Max, EFEKTA_PWS_MaxPro, EFEKTA_eON29wz, EFEKTA_eFlower_Pro, EFEKTA_iAQ, E2134, POK004, POK011, ROB_200-070-0, HK-SENSOR-4IN1-A, AU-A1ZBPIRS, RADON TriTech ZB, TS0601_illuminance_temperature_humidity_sensor_1, TS0202_3, S-LUX-ZB, TS0222_light, THE01860A, TS0222, LCZ030, MIR-HE200-TY, XFY-CGQ-ZIGB, TS0601_smart_human_presence_sensor_1, Y1_IN, ZY-M100-S_1, ZY-M100-S_2, WZ-M100, ZG-205ZL, ZG-204ZM, YXZBRB58, ZY-M100-24G, ZY-M100-24GV2, ZY-M100-24GV3, MTG075-ZB-RL, RT_ZCZ03Z, ZY-M100-S_3, ZG-205Z, 8750001213, HSE2927E, ZB003-X, 07502L, JS-MC-SENSOR-ZB, KK-ES-J01W, KMPCIL_RES005, RTCGQ11LM, RTCGQ12LM, b-parasite, 7A-SS-ZABC-H0, 5AA-SS-ZA-H0, RTCGQ14LM, RTCGQ15LM, ZNCLBL01LM, **GZCGQ01LM**, GZCGQ11LM, 07047L, 9290012607, 9290019758, 9290030675, 9290030674, 545D6306, CCT595011, ZSS-ZK-THL, 4512770, TERNCY-PP01, MBD-S, CTM_MBD_Dim, 3RSNL02043Z, WB-MSW-ZIGBEE v.3, MS100, USM-300ZB, PIR313-E.

lawrencedudley, 2025-01-05:
Ah yes, my bad! Marginally annoying but fair enough. I'll close this.

closed 2025-01-05 by lawrencedudley

lawrencedudley, 2025-01-05:
I know breaking changes do happen but this one does feel avoidable.

No big stress for me, I know what I'm doing and fixing stuff up was pretty quick.

My neighbour who now uses home assistant (with z2m) thanks to me (🙄) is going to be totally lost with this change that I think could have been mitigated quite easily.