Koenkk/zigbee2mqtt #25573: Tyua Single button, IH-K663, after Z2M 2.0 upgrade the "hold" action is missing

KHV8, 2025-01-05:
### What happened?

Using Home Assistant.

In Z2M 1.42 the switch have "single", "double", "hold" action, as part of the action sensor

After upgrade to Z2M 2.0, only "single" and "double" actions are send to the event entity.

### What did you expect to happen?

That all three previous states would be present

### How to reproduce it (minimal and precise)

N/A

### Zigbee2MQTT version

2.0.0.1

### Adapter firmware version

160724

### Adapter

SLZB-06P7

### Setup

HAOS

### Debug log

No relevant debug log

m0wlheld, 2025-01-05:
Action sensors are gone, as stated in the release notes. See #24198 for details and required actions.

KHV8, 2025-01-05:
Sorry if the description is not clear, my bad. I know there are no sensors going forward.

I would expect the "event" to show the same states as the previous sensor. It only show "single" and "double". The "hold" state is gone. This state was used in the old sensor.

Mordevolt, 2025-01-11:
Expirience same issue with Tuya TS004F. It is not producing event for hold action correctly.
In Z2mqtt log event is detected:

info 2025-01-11 14:40:00z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x84fd27fffe803813/action_4_hold/config', payload '{"automation_type":"trigger","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x84fd27fffe803813"],"manufacturer":"Tuya","model":"Wireless switch with 4 buttons","model_id":"TS004F","name":"4 Button Switch","via_device":"zigbee2mqtt_bridge_0x00124b0024bfc9c6"},"origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"https://www.zigbee2mqtt.io"},"payload":"4_hold","subtype":"4_hold","topic":"zigbee2mqtt/4 Button Switch/action","type":"action"}'
info 2025-01-11 14:40:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/4 Button Switch/action', payload '4_hold'

But in HA only legacy sensor detects "4_hold", new event sensor detects just "hold", without button reference:
[image]

Mordevolt, 2025-01-11:
Even in list for all available events it is just plain "hold", without reference to exact button:
[image]

seems like not all Action (enum) exposed in HA correctly

closed 2025-03-20 by github-actions[bot]