Koenkk/zigbee2mqtt #25574: Philips RDM001 Wal Switch - action seems no longer exposed on V2.0.0

luigui2906, 2025-01-05:
### What happened?

I have been using the Philips Hue RDM001 Wall Switch for all switches in my home since about a year now. I created an automation  that listens to sensor action and if left press on switch, a light is toggled, automation below.

When :
```
entity_id:
  - sensor.smart_mate_philips_kitchen_lights_action
to: left_press_release
enabled: true
trigger: state
````

Then do:
```
metadata: {}
data:
  transition: 0
target:
  entity_id: light.bulb_e27_kitchen
action: light.toggle
````

However, this stop working since I updated to V2.0.0 as the action variable seems not to be exposed, I get error on the automation (`"Error in describing trigger: Cannot read properties of undefined (reading 'entity_id'"`). See image below.

[image]

However; the variable is present on zigbee2mqtt and I can see the state change, see image below.

[image]

I decided to roll back to V1.42.0 and everything works back again.

### What did you expect to happen?

Automation continues to work on v2.0.0 on same way as it was working on V1.42.0

### How to reproduce it (minimal and precise)

Add sensor to Zigbee2MQTT and create automation on both versions and compare behavior

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

Sonoff  ZBDongle-P

### Setup

Raspberry Pi with docker container

### Debug log

Sorry, I forgot to download the log before the rollback

m0wlheld, 2025-01-05:
Not an issue, see #24198

m0wlheld, 2025-01-05:
#25573

luigui2906, 2025-01-05:
Thank you for pointing me on the right direction !! It works now with V2.0.0. For everyone that has the same issue, I changed automation from :

```
# AUTOMATION FOR V1.42.0

alias: Kitchen - Countertop Light - Toggle Switch
description: Kitchen countertop lights toggle on/off by selected switches
mode: single
triggers:
  - entity_id:
      - sensor.smart_mate_philips_kitchen_lights_action
    to: left_press_release
    enabled: true
    trigger: state
conditions: []
actions:
  - metadata: {}
    data:
      transition: 0
    target:
      entity_id: light.bulb_e27_kitchen
    action: light.toggle
```

to:

```
# AUTOMATION FOR V2.0.0

alias: Kitchen - Countertop Light - Toggle Switch
description: Kitchen countertop lights toggle on/off by selected switches
mode: single
triggers:
  - trigger: device
    domain: mqtt
    device_id: bbbfc0f589c36c02d260668bbb9b192a
    type: action
    subtype: left_press_release
actions:
  - metadata: {}
    data:
      transition: 0
    target:
      entity_id: light.bulb_e27_kitchen
    action: light.toggle
```

closing now the issue. Thanks for the quick reply and the hard work.

closed 2025-01-05 by luigui2906