Koenkk/zigbee2mqtt #25575: High CPU usage after upgrade to v2.0.0 on Home Assistant

GJurriens, 2025-01-05:
### What happened?

CPU usage is high after the upgrade. Over 30% on an Intel N100 cpu running bare metal Home Assistant.
This probably also causes the reactions of light to be slow.

### What did you expect to happen?

Cpu usage would stay the same and lights would stay response.

### How to reproduce it (minimal and precise)

I just updated in Home Assistant

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Addon on home assistant on an Intel N100 system with 8gb mem running from ssd.

GJurriens, 2025-01-05:
I can upload debug logging later if needed.

m0wlheld, 2025-01-05:
Can you post the output of the following command, showing top 5 processes sorted by CPU usage? 

```bash
ps aux --sort=-pcpu | head -n 6
```

GJurriens, 2025-01-05:
Yes, I can.
`➜  ~ ps aux --sort=-pcpu | head -n 6                                           
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         387  0.0  0.0   4260  3752 pts/1    Ss   19:03   0:00 zsh -l
root         384  0.0  0.0   2796  2232 ?        S    19:03   0:00 tmux -u new -A -s homeassistant zsh -l
root         337  0.0  0.0  15600  2816 ?        Ssl  01:46   0:00 ttyd -d1 -i hassio --writable -p 63840 tmux -u new -A -s homeassistant zsh -l
root         338  0.0  0.0   6116  3968 ?        Ss   01:46   0:00 sshd: /usr/sbin/sshd -D -e [listener] 0 of 10-100 startups
root           1  0.0  0.0    440     0 ?        Ss   01:46   0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
➜  ~ `
Hope this helps out. I also tried running it from within the Zigbee2MQTT container, but that did not work I think.
`➜  ~ ps aux --sort=-pcpu | head -n 6                                           
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         387  0.0  0.0   4260  3752 pts/1    Ss   19:03   0:00 zsh -l
root         384  0.0  0.0   2796  2232 ?        S    19:03   0:00 tmux -u new -A -s homeassistant zsh -l
root         337  0.0  0.0  15600  2816 ?        Ssl  01:46   0:00 ttyd -d1 -i hassio --writable -p 63840 tmux -u new -A -s homeassistant zsh -l
root         338  0.0  0.0   6116  3968 ?        Ss   01:46   0:00 sshd: /usr/sbin/sshd -D -e [listener] 0 of 10-100 startups
root           1  0.0  0.0    440     0 ?        Ss   01:46   0:00 /package/admin/s6/command/s6-svscan -d4 -- /run/service
➜  ~ `

m0wlheld, 2025-01-05:
So, you are running HASSIO, the Home Assistant OS with the Zigbee2MQTT Add-On?

GJurriens, 2025-01-05:
Yes, that is correct.

m0wlheld, 2025-01-05:
I was asking, because I don‘t see any increase in CPU usage with using Zigbee2MQTT standalone (this repository). 

Could be related to the recent Home Assistant update (2025.1) or the Home Assistant Zigbee2MQTT add-on itself. 

Recommend filing an issue [there](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt)

GJurriens, 2025-01-05:
I'll do that and report back here if I know more.

GJurriens, 2025-01-06:
So far no response on the Home Assistant Addon github page.
But I did manage to get some more info on the high CPU usage in the container.
If I check the usage with top inside the container, I get this result:
Mem: 7340724K used, 574316K free, 148788K shrd, 167228K buff, 3142388K cached
CPU:  33% usr  12% sys   0% nic  53% idle   0% io   0% irq   0% sirq
Load average: 1.81 1.89 1.86 3/1215 110
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
    7     1 root     R    11.2g 147%   1  32% node index.js
  103     0 root     S     1708   0%   2   0% sh
  110   103 root     R     1640   0%   2   0% top
    1     0 root     S      860   0%   0   0% /sbin/tini -- /docker-entrypoint.sh

That seems to indicate that the NodeJS process is taking up the CPU cycles within the container.
Is there anything else I can do to help troubleshoot this?

closed 2025-01-06 by GJurriens