Koenkk/zigbee2mqtt #25578: how to go back to version 1.42

rl15453, 2025-01-05:
### What happened?

node: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by node)

### Zigbee2MQTT version

1.42

### Adapter firmware version

20221102

### Adapter

cc2652p

### Setup

pi 4

### Debug log

Found 16 errors in 13 files.

Errors  Files
     4  lib/controller.ts:9
     1  lib/extension/bind.ts:7
     1  lib/extension/bridge.ts:6
     1  lib/extension/configure.ts:4
     1  lib/extension/externalJS.ts:8
     1  lib/extension/frontend.ts:16
     1  lib/extension/groups.ts:7
     1  lib/extension/homeassistant.ts:4
     1  lib/extension/networkMap.ts:4
     1  lib/extension/otaUpdate.ts:8
     1  lib/extension/publish.ts:4
     1  lib/extension/receive.ts:5
     1  lib/zigbee.ts:6
 ELIFECYCLE  Command failed with exit code 1.

rl15453, 2025-01-06:
solved, I found on
https://github.com/Koenkk/zigbee2mqtt/issues/1517

m0wlheld, 2025-01-06:
Can you please close this issue?

closed 2025-01-07 by rl15453