Koenkk/zigbee2mqtt #25580: Z2M fails to start even after restoring from backup

baabercrombie, 2025-01-05:
### What happened?

updated to V2, everything was fine. Ran out of space on HD overnight and now Z2m won't start. I cleared a bunch of space but Z2M refuses to start. I keep getting an error

[10:49:10] INFO: Preparing to start...
curl: (28) Failed to connect to supervisor port 80 after 133498 ms: Could not connect to server
[10:51:23] ERROR: Something went wrong contacting the API
[10:51:23] FATAL: 
[10:51:23] FATAL: A required add-on configuration option is missing!
[10:51:23] FATAL: 
[10:51:23] FATAL: Please set a value for the 'data_path' option.
[10:51:23] FATAL: 
[10:51:23] FATAL: If unsure, check the add-on manual for more information.
[10:51:23] FATAL: 

I have repaired and restarted supervisor from CLI, restored backup to 1.42, unistalled and reinstalled Z2M. Edited .Yaml to make sure that everything was up to date with requests of 2.0. I have no idea whats going on.

### Zigbee2MQTT version

2.0

### Adapter firmware version

unknown

### Adapter

Sonoff 3.0 ZBDongle-P

### Setup

OS on Nuc

### Debug log

[45df7312_zigbee2mqtt_2025-01-05T17-20-59.723Z.log](https://github.com/user-attachments/files/18311036/45df7312_zigbee2mqtt_2025-01-05T17-20-59.723Z.log)

m0wlheld, 2025-01-05:
I think you are in the wrong spot, this looks like an issue with the HASSIO Zigbee2MQTT add-on. 

See https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

djurny, 2025-01-05:
Hi there,
You might be helped with reading https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/662.
Grt,

baabercrombie, 2025-01-05:
> I think you are in the wrong spot, this looks like an issue with the HASSIO Zigbee2MQTT add-on.
> 
> See https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

You might be right. Ill try posting there.

baabercrombie, 2025-01-05:
> Hi there, You might be helped with reading [zigbee2mqtt/hassio-zigbee2mqtt#662](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/662). Grt,

this is on the right track but I have tried these fixes and still nothing has changed

closed 2025-03-14 by github-actions[bot]