Koenkk/zigbee2mqtt #25581: Tuya ZG-227Z _TZE200_a8sdabtg wrongly corrected in Zigbee 2.0.0 although battery level measurement is correct

nephrotranz, 2025-01-05:
### What happened?

I also own three of these devices and they all showed a battery level of 100% since 23rd of January 2024 until the update to Z2M 2.0.0.
After the update the battery levels of all three devices suddenly went down to 50% and - although I read the release note -  I found this a bit suspicious.

The update was done because of this thread:

[https://github.com/Koenkk/zigbee2mqtt/issues/24120]

For a bit of "research" I did put a brand new battery into one of the devices and waited 24 hours now but the battery level is still at 50%. 
At the same time, because I was curious, i did put the old battery from this device into a Hue dimmer v1 and this one went up from 34% to 100% within a few hours.

So I took a second try and changed the batteries from another ZG-227Z wíth another Hue Dimmer v1. After some hours the battery level of the Hue dimmer went up from approx 40% to 100%. The ZG-227Z is still showing 50% although the battery must be under 40%

**So the battery level really is still at 100% after nearly one year.**

In the database.db my devices also show a voltage value which is 29 or 30 for all three batteries. As 3 V is the voltage of a CR2450 battery at 100% that also shows that they are measuring correctly.

Additionally I did put one of the thermometers into our freezer last August and the battery level went down to 10% when it was at approx. -19°C for a fews days, so it was definitely working, it went back to 100% after some hours at +20°C.

I can only imagine that @mleone87 had some faulty devices or something like that or these devices are simply shitty when the battery life is really coming to an end and do not show it,

I also would like to point out that the file was changed because of this line in the database.db:
`"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}`

but my Hue dimmers and my Sonoff SZNB-01 switches are showing exact the same line in database.db so I am sure there i nothing wrong with it.

### What did you expect to happen?

Still showing a battery level of 100%

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle P

### Setup

HA add-on on Raspberry Pi 4

KipK, 2025-01-06:
same Issue here, 

Battery was reported ok pre 2.0.0:

[image]

We can see on the graph that it jumped to 50% just after the 2.0.0 update.

gpriestland, 2025-01-11:
I also see the same issue.  All Tuya ZG 227Z sensors started to show 50% battery level shortly after updating to 2.0.0

[image]

Not sure if this is relevant but I also since bought a light level sensor - Tuya TS0222_light and this always reads 20% battery - I have tried swapping batteries between other sensors and it has always reported 20%.

I have 2x Tuya ZTH01/ZTH02 Temperature / Humidity sensors, 2x Tuya WSD500A Temperature / Humidity sensors and 3x Tuya TS020y leak sensors that appear to be working.  By working I mean that the battery level is still at 100% or is reading the same value after the upgrade 2.0.0 as it was before.

nyctomanica, 2025-01-11:
[image]

Same issue here, all of my Tuya ZG-227Z sensors dropped to 50% after the Z2M 2.0 update.

I've only recently changed the batteries in these, and can confirm that the reporting worked perfectly before the update.
They previously went down as expected from 100% to 10%, hence the battery swaps.

Reported voltages on all in database.db are 3.0v (30), which matches my multimeter.

I also have Tuya TRV601 radiator valves using 3x AA batteries which still report their battery percentage properly, so it seems isolated to just these sensors.

redmoskito-web-solutions, 2025-01-11:
Same here: 3 devices TS0601 were on battery level 100%. Since the updare all are on 50%.

KipK, 2025-01-12:
@Koenkk , can this [PR](https://github.com/Koenkk/zigbee2mqtt/issues/24120) be reverted?

Koenkk, 2025-01-13:
@KipK done!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-13 by Koenkk

remizik, 2025-02-12:
Hi,
Is this supposed to be fixed ?
All my devices report 100% battery for months now : 

[image]

[image]

nephrotranz, 2025-02-12:
> Hi, Is this supposed to be fixed ? All my devices report 100% battery for months now :
> 
Mine too - and as I wrote, I took those batteries out and did put them into my Hue dimmer switches and they are really all still at 100% after more than a year.

remizik, 2025-02-12:
> > Hi, Is this supposed to be fixed ? All my devices report 100% battery for months now :
> 
> Mine too - and as I wrote, I took those batteries out and did put them into my Hue dimmer switches and they are really all still at 100% after more than a year.

Very weird if these are actually at 100% ! One of them is outside and has been sitting in negative temps during winter :)
I guess time will tell

meiser79, 2026-02-12:
Hi, I just received the ZG-204ZM presence sensor. I inserted fully charged AA batteries (IKEA LADDA, AAA, 1,2V / 750mAh) and the device shows exactly 50%.

[image]

Is this issue related to this bugreport?

Thanks,
meiser