Koenkk/zigbee2mqtt #25582: Got notification message from PID xxxx, but reception only permitted for main PID xxxx

lazi3b0y, 2025-01-05:
### What happened?

z2m keeps terminating because of timeout after upgrading to 2.0.0, with these messages under the systemctl status logs:

1. Got notification message from PID xxxx, but reception only permitted for main PID xxxx
2. start operation timed out. Terminating.
3. Failed with result 'timeout'

### Zigbee2MQTT version

2.0.0 commit: 060ae99

### Adapter firmware version

7.4.5 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E

### Setup

Proxmox privileged lxc container

### Debug log

```
2025-01-05T16:59:27.260836+00:00 zigbee2mqtt systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
2025-01-05T16:59:27.379490+00:00 zigbee2mqtt node[4276]: Starting Zigbee2MQTT without watchdog.
2025-01-05T16:59:29.186473+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011z2m: Logging to console, file (filename: log.log)
2025-01-05T16:59:29.205184+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99)
2025-01-05T16:59:29.211274+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011z2m: Starting zigbee-herdsman (3.2.1)
2025-01-05T16:59:29.444397+00:00 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Got notification message from PID 4296, but reception only permitted for main PID 4276
2025-01-05T16:59:29.483852+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember: Using default stack config.
2025-01-05T16:59:29.486054+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember: ======== Ember Adapter Starting ========
2025-01-05T16:59:29.486990+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember:ezsp: ======== EZSP starting ========
2025-01-05T16:59:29.487476+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember:uart:ash: ======== ASH Adapter reset ========
2025-01-05T16:59:29.488549+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
2025-01-05T16:59:29.498135+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember:uart:ash: Serial port opened
2025-01-05T16:59:29.498204+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[32minfo#033[39m: #011zh:ember:uart:ash: ======== ASH starting ========
2025-01-05T16:59:29.499823+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[31merror#033[39m: #011zh:ember:uart:ash: Received frame with CRC error
2025-01-05T16:59:29.502486+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[31merror#033[39m: #011zh:ember:uart:ash: Received frame with CRC error
2025-01-05T16:59:29.502779+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:29] #033[31merror#033[39m: #011zh:ember:uart:ash: Received ERROR from adapter while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
2025-01-05T16:59:30.592778+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember:uart:ash: ======== ASH connected ========
2025-01-05T16:59:30.610854+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember:uart:ash: ======== ASH started ========
2025-01-05T16:59:30.614087+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember:ezsp: ======== EZSP started ========
2025-01-05T16:59:30.625818+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
2025-01-05T16:59:30.632685+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.5 [GA]","build":0,"major":7,"minor":4,"patch":5,"special":0,"type":170}
2025-01-05T16:59:30.774709+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember: [STACK STATUS] Network up.
2025-01-05T16:59:30.793858+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember: [INIT TC] Adapter network matches config.
2025-01-05T16:59:30.805384+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
2025-01-05T16:59:30.822886+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: zigbee-herdsman started (resumed)
2025-01-05T16:59:30.823886+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":5,"revision":"7.4.5 [GA]","special":0,"type":170},"type":"EmberZNet"}'
2025-01-05T16:59:30.825011+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Food cellar motion sensor (0x001788010917b2a4): 9290012607 - Philips Hue motion sensor (EndDevice)
2025-01-05T16:59:30.825412+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Philips Hue motion sensor (0x000425fffffd8af1): 9290012607 - Philips Hue motion sensor (EndDevice)
2025-01-05T16:59:30.825944+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: 0x001788010b09290f (0x001788010b09290f): 9290012607 - Philips Hue motion sensor (EndDevice)
2025-01-05T16:59:30.826315+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Currently 3 devices are joined.
2025-01-05T16:59:30.826829+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Connecting to MQTT server at mqtt://192.168.86.24:1883
2025-01-05T16:59:30.874993+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m: Connected to MQTT server
2025-01-05T16:59:30.875701+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
2025-01-05T16:59:30.936938+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Food cellar motion sensor/availability', payload '{"state":"online"}'
2025-01-05T16:59:30.981178+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:30] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue motion sensor/availability', payload '{"state":"online"}'
2025-01-05T16:59:31.024111+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010b09290f/availability', payload '{"state":"online"}'
2025-01-05T16:59:31.067313+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Office Spotlights/availability', payload '{"state":"online"}'
2025-01-05T16:59:31.110333+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Spotlights/availability', payload '{"state":"online"}'
2025-01-05T16:59:31.153397+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Lights/availability', payload '{"state":"online"}'
2025-01-05T16:59:31.203129+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m: Started frontend on port 8080
2025-01-05T16:59:31.260186+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Food cellar motion sensor', payload '{"battery":44,"illuminance":48,"led_indication":null,"linkquality":148,"motion_sensitivity":null,"occupancy":false,"occupancy_timeout":0,"temperature":14.85,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"}}'
2025-01-05T16:59:31.262217+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Philips Hue motion sensor', payload '{"battery":100,"illuminance":0,"led_indication":null,"linkquality":null,"motion_sensitivity":null,"occupancy":false,"occupancy_timeout":0,"temperature":22.31,"update":{"installed_version":1107323831,"latest_version":1107323831,"state":"idle"}}'
2025-01-05T16:59:31.262514+00:00 zigbee2mqtt node[4276]: [2025-01-05 16:59:31] #033[32minfo#033[39m: #011z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010b09290f', payload '{"battery":100,"illuminance":0,"led_indication":null,"linkqu [... cut ...]
2025-01-05T17:00:58.146024+00:00 zigbee2mqtt node[4276]: [2025-01-05 17:00:58] #033[32minfo#033[39m: #011zh:ember: ======== Ember Adapter Stopped ========
2025-01-05T17:00:58.146384+00:00 zigbee2mqtt node[4276]: [2025-01-05 17:00:58] #033[32minfo#033[39m: #011z2m: Stopped zigbee-herdsman
2025-01-05T17:00:58.146556+00:00 zigbee2mqtt node[4276]: [2025-01-05 17:00:58] #033[32minfo#033[39m: #011z2m: Stopped Zigbee2MQTT
2025-01-05T17:00:58.164666+00:00 zigbee2mqtt systemd[1]: zigbee2mqtt.service: Failed with result 'timeout'.
2025-01-05T17:00:58.165049+00:00 zigbee2mqtt systemd[1]: Failed to start zigbee2mqtt.service - zigbee2mqtt.
```

lazi3b0y, 2025-01-05:
I commented out `Type=notify` and `WatchdogSec=10s` and added `Type=simple` instead in my service to get around it. But still not sure why its happening

lazi3b0y, 2025-01-05:
Adding `NotifyAccess=all` to the [Service] section of the service seems to get rid of the ```Got notification message from PID xxxx, but reception only permitted for main PID xxxx```
problem but the service keeps timing out anyway for some reason, I just assumed that it was a problem with notifications of some sort, but maybe not?

Trebacz, 2025-01-11:
> I commented out `Type=notify` and `WatchdogSec=10s` and added `Type=simple` in my service instead to get around it. But still not sure why its not working with the original parameters.

This also solved it for me running Zigbee2mqtt on Ubuntu 24.04.1. The service would end and restart every 2 minutes. I also got messages "error:  z2m: Not connected to MQTT server!" when I could clearly see that MQTT messages were being sent and received on the MQTT localhost server.

closed 2025-03-20 by github-actions[bot]