Koenkk/zigbee2mqtt #25583: Unable to update to 2.0 - aborts update

SteEllis, 2025-01-05:
### What happened?

Started the update and the installation was aborted  as follows

zigbee-user@zigbee2mqtt-host:/opt/zigbee2mqtt $ ./update.sh
Checking Zigbee2MQTT status...
Stopping Zigbee2MQTT...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating 861cba63..060ae99c
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.example.yaml
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting

### What did you expect to happen?

I have updated many times without issue.

### How to reproduce it (minimal and precise)

just need to remove the data-backup and rerun the update.sh 

### Zigbee2MQTT version

1.42.0 commit: 861cba63

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Installed on a Z2M dedicated raspberry pi - connected to HAOS over network

x86Hephestus, 2025-01-05:
You need to rename your configuration.yaml to configuration.yaml.bak then run .update.sh
Once update is complete you can rename configuration.yaml.bak to configuration.yaml and start

m0wlheld, 2025-01-05:
This has been reported multiple times and is covered by #24198

Koenkk, 2025-01-05:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, from https://github.com/Koenkk/zigbee2mqtt/discussions/24198

> - For Git based installations: removed `configuration.yaml` from repo, this will result in the following error when updating:
>    ```
>    error: Your local changes to the following files would be overwritten by merge:
>      data/configuration.yaml
>    ```
>    To fix this, update by executing:
>    ```sh
>    git checkout data/configuration.example.yaml
>    mv data/configuration.yaml data/configuration.yaml.bak
>    ./update.sh # (update as you usually do)
>    mv data/configuration.yaml.bak data/configuration.yaml
>    ```
>- `pnpm` is now used instead of `npm`, the [installation](https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/docs/guide/installation/01_linux.md#installing) instructions and  `update.sh` script has been updated accordingly. For existing installations you need to execute `npm install -g pnpm` before updating.

closed 2025-01-05 by Koenkk

SteEllis, 2025-01-05:
> This has been reported multiple times and is covered by #24198
Thanks m0wlheld,
I have got it up and working now, maybe not the best way. But I had to do a clean install and restore the data after another rpm build error. I just have to clean up the illuminance_lux variables in HAOS.