Koenkk/zigbee2mqtt #25584: After Update problem with LAN gateway.

IkaSGC, 2025-01-05:
### What happened?

Instant error after update. Restore to the original version fixes the problem immediately. 

I've researched the forum which is now flooded with similar problems, but everyone is addressing the USB adapter, I haven't found much advice on the LAN gateway. How to solve this problem ?? Is there any repair ? 

LOG: 
[18:59:53] INFO: Preparing to start...
[18:59:53] INFO: Socat not enabled
[18:59:53] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-05 18:59:55] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-05.18-59-28'
[2025-01-05 18:59:55] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-05 18:59:55] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-05 18:59:55] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-05 18:59:55] debug: 	z2m: sd-notify loaded
[2025-01-05 18:59:55] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-05 18:59:55] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":16592,\"extendedPanID\":[102,155,160,233,196,13,159,127],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"path\":\"tcp://192.168.2.5:6638\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-01-05 18:59:55] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-05 18:59:55] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-05 18:59:55] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-05 18:59:55] error: 	z2m: Exiting...
[2025-01-05 18:59:55] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

### Zigbee2MQTT version

1.42.0-2 -> 2.0.0-1

### Adapter firmware version

NotFound

### Adapter

ZigStar UZG-01

### Setup

Home Assistant addon.

m0wlheld, 2025-01-05:
> Specify valid 'adapter' and 'port' in your configuration.

#24198

IkaSGC, 2025-01-05:
Not sure how.... Port is there and cant found how to specify adapter. 

`serial:
  port: tcp://192.168.2.5:6638
  baudrate: 115200
  disable_led: false
advanced:
  transmit_power: 20
`

m0wlheld, 2025-01-05:
See #24198

IkaSGC, 2025-01-05:
Bingo..... 
Need to add: adapter: zstack

But not into setting in addon... But not in the addon settings in HA, but directly in:

zigbee2mqtt/configuration.yaml

closed 2025-01-05 by IkaSGC

m0wlheld, 2025-01-05:
This repository is not about the HASSIO add-on ..

IkaSGC, 2025-01-05:
This is possible, but the documentation link goes directly here. The solution was here, but well hidden. I found the truth after 40 minutes of searching and crossing several "tickets".