Koenkk/zigbee2mqtt #25586: Exit code 134 on 1.X.X to 2.X.X update execution

x86Hephestus, 2025-01-05:
### What happened?

I run z2m bare metal on a rPi Zero W, node version 18.19.1, pnpm version 9.15.2. I was trying to execute the update from 1.X.X to 2.0.0 and it is failing with exit code 134. My memory utilization is at 55.9M/429M at idle. I see that there are some people saying to increase the memory allocated to node with node arg "--max_old_space_size=XXX" so I injected the line into package.json but the update command still failed.
```
/opt/zigbee2mqtt $ ./update.sh
Checking Zigbee2MQTT status...
Updating...
Already up to date.
Installing dependencies...
Lockfile is up to date, resolution step is skipped
Already up to date
Done in 53.6s
Building...

> zigbee2mqtt@2.0.0 build /opt/zigbee2mqtt
> tsc && node --max_old_space_size=350 index.js writehash

<--- Last few GCs --->

[4114:0x545fa28]   472809 ms: Scavenge 123.5 (128.9) -> 123.1 (129.4) MB, 22.2 / 0.0 ms  (average mu = 0.279, current mu = 0.296) allocation failure;
[4114:0x545fa28]   478513 ms: Mark-sweep 123.5 (129.4) -> 122.9 (129.7) MB, 5574.2 / 0.0 ms  (average mu = 0.186, current mu = 0.074) allocation failure; scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory
Aborted
 ELIFECYCLE  Command failed with exit code 134.
```

### What did you expect to happen?

Update should build successfully

### How to reproduce it (minimal and precise)

Build z2m on Rasberry Pi Zero W with node version 18.19.1 and pnpm version 9.15.2.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

latest

### Adapter

Sonoff E

### Setup

Bare Metal on Pi Zero W

x86Hephestus, 2025-01-05:
I was able to resolve this issue by disabling all other services running on the pi, restarting, and rerunning update script. 2.0.0 is now built and executing. All other services were restored after the upgrade process with no issues.

m0wlheld, 2025-01-05:
So,  mark this issue as closed, please?

closed 2025-01-05 by x86Hephestus