Koenkk/zigbee2mqtt #25587: Ikea E1745 OTA disables motion detection

RubenKelevra, 2025-01-05:
### What happened?

I've got a message today regarding a OTA firmware update for my E1745 from Ikea (motion sensor). I've updated one of my devices.

After the update the device wasn't sending motion updates anymore, but it showed just 14% battery status (two CR2032). So I replaced them, but no luck, it still won't set any "occupation" status anymore. That's not just in HA but also in the Add-On UI not showing any updates. 

The old batteries had still 2.9 V, so they should have still worked fine for a while. 

No other device is affected, so it's not the 2.0 update, as far as I can tell.

I've also noticed that the E1745 is not on the changelog list and the firmware version number jumped from 2.x to 24.x.

Are we sure that's the right firmware?

@Koenkk sorry for the ping, but I think we may want to stop pushing this update until someone else confirms that it's working for them? 🤔 

### How to reproduce it (minimal and precise)

- Update from `20190308 / 2.0.022` to `20220928 / 24.4.5` via Z2M's OTA function

### Zigbee2MQTT version

2.0

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick 

### Setup

Add-on on HA on a Raspberry 4B

WoBBeLnl, 2025-01-05:
Had some same weird issueswith the E1745  after updating to 2.0.0. My sensor was up to date but I received an update for it which I installed. After updating this message showed up: 
[image]

I had to repair my device and everything started working again EXCEPT for group binding. It only sends out to group 0 and not the previous 123 anymore.

`z2m: Received Zigbee message from 'Trapkast TRADFRI motion sensor', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":1800}' from endpoint 1 with groupID 0`

Koenkk, 2025-01-05:
Removed it!

JornDL, 2025-01-06:
I have updated to 24.4.5 and have no issues

RubenKelevra, 2025-01-06:
@Koenkk is there a way to force the older firmware onto that device somehow?

Or maybe force install the same firmware again onto it? Maybe the low battery status of my device was the culprit here and something went wrong flashing it? 🤔

ashimokawa, 2025-01-06:
I also updated, and it still works well.

stefangries, 2025-01-07:
Having the same problem with the sensor not sending motion updates anymore. Now downgrading the firmware using this command, hoping that it helps: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#downgrade-1

stefangries, 2025-01-08:
Mine detect no motion anymore after the update.

adizanni, 2025-01-09:
Also mine was not detecting motion after the firmware update. I'm trying to re-pair but it fails the interview

adizanni, 2025-01-09:
Small update: after a few trials I managed to re-pair the device and now the occupancy works....

stefangries, 2025-01-09:
Can confirm that deleting and repairing the device solves the problem.

RubenKelevra, 2025-01-10:
I fixed the pairing instructions in the wiki, which were wrong. 10 seconds press on the pairing button is ZLL pairing. (https://github.com/Koenkk/zigbee2mqtt.io/pull/3414)

I've tried to delete and readd the sensor, no idea why that would be necessary for a simple stability update thought...

Anyway, the interview failed on both attempts:

[image]

stefangries, 2025-01-10:
@RubenKelevra Yes, that's a bit tricky. What worked for me was to remove both batteries from the device, then press the interview button, and then immediately reinsert (at least) one battery. Then the interview went through.

Joska59, 2025-01-10:
@stefangries Wow, I laughed at first, but then it worked the same way with the first of 2 stubborn update refusers here. But that's more WooDoo than IT :-)

stefangries, 2025-01-10:
@Joska59 Nice! I think the reason is, that the device is asleep most of the time and does not receive messages. After inserting the batteries the device is awake for a short time.

RubenKelevra, 2025-01-10:
I successfully could finish the interview by pressing the link button 4 times again and manually starting the interview in z2m. Took 2 tries.

Overall this firmware seems to be a lot more buggy than the previous one :/

Koenkk, 2025-01-11:
@RubenKelevra we don't have the old firmware so unfortunately cannot downgrade.

RubenKelevra, 2025-01-11:
Hmmm... that's unfortunate. Is someone familiar with the team at Ikea?

Maybe we can just ask them for the older firmware or they can fix the new firmware? 🤔

drueppela, 2025-01-11:
@RubenKelevra & @Koenkk : 

I have at least one still original packed which I bought a long time ago. I can donate and ship it.

RubenKelevra, 2025-01-11:
@drueppela I think there's some misunderstanding, we would require Ikea to publish the older firmware as a file.

While technically possible to read out the firmware from a device I don't think that's worth the effort. I also got a device I did not update yet. Thanks for the offer, anyway!

romamix, 2025-01-14:
My device lost any binding support after the update until I reset it: removing the batteries does the trick to successfully re-pair. Motion detection works, but only until I bind the sensor to my inovelli 2-1 switch: once bonded, no motion is detected in HA. However, the bond works and the light turns on.

romamix, 2025-01-14:
And I think I found a workaround: in the Bind menu I found that there are a lot of clusters that are not bonded to the coordinator. Binding them all does the trick: the light still comes on/off without any automations, and sensor states are updated in Home Assistant.

RubenKelevra, 2025-01-14:
@Koenkk is this something we can do automatically if the new firmware is detected?

Koenkk, 2025-01-14:
@RubenKelevra yes, but we have to find out which cluster to bind, could you try to bind the `genOnOff` cluster to the coordinator? You may need to wake-up the device right before binding it.

RubenKelevra, 2025-01-14:
@Koenkk I've already removed the device and re-pair-ed it. Now it works. So the question is, what's wrong after just updating it? 🤔

Koenkk, 2025-01-15:
Do you have any binding showing up in the z2m frontend for this device?

RubenKelevra, 2025-01-15:
@Koenkk I guess they would show up here, right?

[image]

Koenkk, 2025-01-17:
Yes, interesting that there are none (you can ignore `genPollCtrl`, so I'm not sure what fixed it.

tmidtboell, 2025-02-27:
The update of below device is still not working:
Device info
TRADFRI motion sensor (E1525/E1745)
by IKEA
Connected via [Zigbee2MQTT Bridge]
Firmware: 2.0.022
Hardware: 1

Hall motion
3 days ago
Update available
Installed version
537011747
Latest version
604241925

bakgaard, 2025-02-27:
@tmidtboell - I got three of those upgraded following this method:

- Open the webpage for Z2M
- Open the tab "OTA", and search for the already known device (either `E1525` or `E1745`).
- Click 4 times on the "pair" button on the device - this will put it in repairing mode.
- Click on the two other buttons next to the "pair"-button - back and forth - as this will keep the device alive. Click once every second or so.
- While clicking back and forth, see check when the device comes back online in Z2M
- While clicking back and forth still, click on `Update` in the "OTA" pane next to the device until you see it saying "Update started..." (or something along those lines).

I hope it helps you

kverko25, 2025-03-26:
I also had the same issue: after updating the motion sensor, things stopped working as before.
- One sensor died mid-update, low battery. I changed the battery and managed to re-interview it, but now it is no longer showing the OTA update as available.
- Three other sensors did update, and they were no longer switching the lights on/off. They were paired to groups, not to the devices directly. After the update, I paired the motion sensors directly to the lights and they started turning on/off as before. However, I could no longer see the occupancy status updating.
- I followed the suggestion I saw here and did the binding of all available clusters between the motion sensor and the coordinator, and I can again see the occupancy status changing. I'm not sure which cluster was responsible for things to start working again.

If needed, I can investigate further or send logs to help pinpoint the issue. Might be a good idea to update the docs to reflect the fact that after this update it seems that the pairing needs to be directly to the device instead of to the group (which was required before).

closed 2025-06-02 by github-actions[bot]