Koenkk/zigbee2mqtt #25591: Aqara WXKG11LM - action not working

lolekanabolek, 2025-01-05:
### What happened?

After upgrade to Z2M 2.0, action is unavailable
[image]

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20221226

### Adapter

Tead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

intel nuc, docker

m0wlheld, 2025-01-05:
#24198

mxbvn, 2025-01-05:
Same here, I understand that this issue stems from the breaking changes in version 2, but I don't see how to resolve it cleanly.

m0wlheld, 2025-01-05:
From #24198:

> All action sensors are now disabled by default (sensor.*_action entities). It's recommended to use the [MQTT device trigger](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt-device-trigger-recommended) instead. In case you really need the action sensors, add the following to your configuration.yaml.
homeassistant:
  legacy_action_sensor: true

closed 2025-01-05 by Koenkk