Koenkk/zigbee2mqtt #25593: IKEA TRADFRI motion sensor E1525/E1745 OTA update fails

drueppela, 2025-01-05:
### What happened?

IKEA TRADFRI motion sensor E1525/E1745 OTA update fails

Bad EG - IKEA TRADFRI motion sensor E1525/E1745 - MQTT -
17 hours ago
Update available
Installed version
537011747
Latest version
604241925

Eingang EG - IKEA TRADFRI motion sensor E1525/E1745 - MQTT -
17 hours ago
Update available
Installed version
537011747
Latest version
604241925

### What did you expect to happen?

I expected the update to be successful 

### How to reproduce it (minimal and precise)

The errors occur every  time I try to update.

### Zigbee2MQTT version

Zigbee2MQTT Current version: 2.0.0-1

### Adapter firmware version

7.4.1 [GA]

### Adapter

Firmware: core: v2.5.8 / zigbee: 20240510 / ember

### Setup

Hardware: Intel - Hypervisor: Proxmox - Home Assistant  Core 2025.1.0 Supervisor 2024.12.3 Operating System 14.1 Frontend 20250103.0

### Debug log

debug 2025-01-05 21:13:19zhc:lumi: RTCGQ14LM: unknown key 19 with value 0
debug 2025-01-05 21:13:22zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:13:22zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
debug 2025-01-05 21:13:22zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2](frmRx=2) Added to rxQueue
debug 2025-01-05 21:13:22zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=3)
debug 2025-01-05 21:13:22zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=194 Len=22]
debug 2025-01-05 21:13:22zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":232}], [messageTag=255]
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=3](frmRx=3) Added to rxQueue
debug 2025-01-05 21:13:24zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=3)
debug 2025-01-05 21:13:24zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=194 Len=20]
debug 2025-01-05 21:13:24zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=43787], [sourceEui=0x54ef44100081b459], [lastHopLqi=192], [lastHopRssi=-52], [relayCount=1], [relayList=33972]
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
debug 2025-01-05 21:13:24zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=4](frmRx=4) Added to rxQueue
debug 2025-01-05 21:13:24zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=3)
debug 2025-01-05 21:13:24zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=194 Len=82]
debug 2025-01-05 21:13:24zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":64704,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":130}], [packetInfo:{"senderShortId":43787,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":196,"lastHopRssi":-51,"lastHopTimestamp":0}], [messageContents=1c5f11610af70041300121a90b032818042100000521010008210a010a21b4840c20011320001420006410006521030069201e6a20026b2000]
debug 2025-01-05 21:13:24zh:controller: Received payload: clusterID=64704, address=43787, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=196, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4447,"transactionSequenceNumber":97,"commandIdentifier":10},"payload":[{"attrId":247,"dataType":65,"attrData":{"type":"Buffer","data":[1,33,169,11,3,40,24,4,33,0,0,5,33,1,0,8,33,10,1,10,33,180,132,12,32,1,19,32,0,20,32,0,100,16,0,101,33,3,0,105,32,30,106,32,2,107,32,0]}}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
debug 2025-01-05 21:13:24z2m: Received Zigbee message from 'Flur GG - Saeule Mitte - Aqara P1 - MQTT -', type 'attributeReport', cluster 'manuSpecificLumi', data '{"247":{"data":[1,33,169,11,3,40,24,4,33,0,0,5,33,1,0,8,33,10,1,10,33,180,132,12,32,1,19,32,0,20,32,0,100,16,0,101,33,3,0,105,32,30,106,32,2,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2025-01-05 21:13:24zhc:lumi: RTCGQ14LM: Processed buffer into data {"1":2985,"3":24,"4":0,"5":1,"8":266,"10":33972,"12":1,"19":0,"20":0,"100":0,"101":3,"105":30,"106":2,"107":0}
debug 2025-01-05 21:13:24zhc:lumi: RTCGQ14LM: unknown key 19 with value 0
debug 2025-01-05 21:13:24zhc:lumi: RTCGQ14LM: unknown key 20 with value 0
[... 174 lines cut ...]
debug 2025-01-05 21:14:09z2m: Device 'Eingang EG - Lichtdecke - MQTT -' requested OTA
debug 2025-01-05 21:14:11zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:14:11zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
debug 2025-01-05 21:14:11zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=6](frmRx=6) Added to rxQueue
debug 2025-01-05 21:14:11zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=4)
debug 2025-01-05 21:14:11zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=195 Len=22]
debug 2025-01-05 21:14:11zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":248}], [messageTag=255]
debug 2025-01-05 21:14:14zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:14:14zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
debug 2025-01-05 21:14:14zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=7](frmRx=7) Added to rxQueue
debug 2025-01-05 21:14:14zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=4)
debug 2025-01-05 21:14:14zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=195 Len=22]
debug 2025-01-05 21:14:14zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":249}], [messageTag=255]
debug 2025-01-05 21:14:17zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:14:17zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
debug 2025-01-05 21:14:17zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=0](frmRx=0) Added to rxQueue
debug 2025-01-05 21:14:17zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=4)
debug 2025-01-05 21:14:17zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=195 Len=22]
debug 2025-01-05 21:14:17zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":250}], [messageTag=255]
debug 2025-01-05 21:14:19zh:ember:uart:ash: <--- [FRAME type=DATA]
debug 2025-01-05 21:14:19zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=4)
debug 2025-01-05 21:14:19zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=1](frmRx=1) Added to rxQueue
debug 2025-01-05 21:14:19zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=4)
debug 2025-01-05 21:14:19zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=195 Len=22]
debug 2025-01-05 21:14:19zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=BROADCAST], [indexOrDestination=65533], [apsFrame={"profileId":0,"clusterId":0,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":251}], [messageTag=255]

m0wlheld, 2025-01-05:
#25587 same device, maybe related

drueppela, 2025-01-05:
Has the update perhaps been withdrawn?

kajapa, 2025-01-05:
I have the same problem with all 5 sensors.

Koenkk, 2025-01-05:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/25587

closed 2025-01-05 by Koenkk

drueppela, 2025-01-09:
@Koenkk : Many thanks for your effort!

Does https://github.com/Koenkk/zigbee2mqtt/issues/25587 have the same root cause? For 2 of 3 of my IKEA TRADFRI motion sensors E1525/E1745 the OTA update still fails.

zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "33559"
z2m: Update of 'Buero EG - IKEA TRADFRI motion sensor E1525/E1745 - MQTT - ' failed (Device didn't respond to OTA request)

Joska59, 2025-01-10:
have a comparable problem. With one of 3 E1745, 2 I was able to update by pressing the pairing button of the motion detector several times after clicking on the update button in z2m, approximately every 3 seconds. Then the update went live and took about 2 hours. With the 3rd motion detector I only have the info -1 as firmware. With this one I get the info in z2m that there is a firmware update for the latest version, but then, like yours, the message that the device does not respond to the OTA Inquiry.

drueppela, 2025-01-15:
@Koenkk:

Could you please advise what to do regarding the pending updates?

Buero EG - IKEA TRADFRI motion sensor E1525/E1745 - MQTT -
30 minutes ago
Update available
Installed version 537011747
Latest version 604241925

Update fails.

BombusAlpinus, 2025-01-30:
> having the same "problem" with this ikea device. update available but fails
> 
> ```
> "update": {
>     "installed_version": 537011747,
>     "latest_version": 604241925,
>     "state": "available"
> }
> ```
> 
> device is working as expected so i'm currently ignoring this update but it pops up every now and then ... and fails again ;)

my fault, took completly new batteries, deleted the device from z2m repaired it and with a bit of luck the update startet...

MihaiKrieger, 2025-02-03:
I found this update pending for my sensors today morning, but it begs the question: Why would I upgrade to 604241925? What benefit does it bring? I found no news on here [the official IKEA patch notes page](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html).

The sensors are still working and I skipped the update.

BombusAlpinus, 2025-02-03:
i was able to update the device. was a bit of a hassel because i had to delete and repair the device and triggering the update right after repairing ...

tomtom-m, 2025-02-24:
> i was able to update the device. was a bit of a hassle because i had to delete and repair the device and triggering the update right after repairing ...

I managed to do the same, but with no hassle. I pressed "reconfigure", and right after that update. The update is very slow, not sure whether that's normal, but it's running.

tmidtboell, 2025-02-27:
This is still an issue.
Hall motion
3 days ago
Update available
Installed version
537011747
Latest version
604241925

SkylightXD, 2025-03-03:
@tmidtboell try this:

Check the charge of the batteries. It does not work at my side with 13%, but new fresh batteries works

1. Remove both batteries of the E1745
2. Press an hold the paring button
3. Insert the frist batterie
4. Start the update process
5. Insert the second batterie
6. Release the pairing button

tmidtboell, 2025-03-05:
> [@tmidtboell](https://github.com/tmidtboell) try this:
> 
> Check the charge of the batteries. It does not work at my side with 13%, but new fresh batteries works
> 
> 1. Remove both batteries of the E1745.
> 2. Press an hold the paring button.
> 3. Insert the frist battery.
> 4. Start the update process in Zigbee2MQTT
> 5. Insert the second battery.
> 6. Release the pairing button.

That worked! Thanks a lot!

BasicWolf, 2025-06-18:
> Check the charge of the batteries. It does not work at my side with 13%, but new fresh batteries works
> 
>     1. Remove both batteries of the E1745.
>     2. Press an hold the paring button.
>     3. Insert the first battery.
>     4. Start the update process in Zigbee2MQTT
>     5. Insert the second battery.
>     6. Release the pairing button.

Thank you, this works. A small update to the procedure: I tried releasing the pairing button once the OTP process started and the process went fine. Note though, that the sensor was lying on the table, facing me and constantly emitting "Occupancy" messages. Perhaps that kept the OTP process alive. Last, but not least: update was painfully slow, it took almost 2 hours to update.

jz6pwc, 2025-07-10:
> That worked! Thanks a lot!

Still not working for me.  Just to double check.  The pairing button is the one with the little link symbol correct?

norulesatall, 2025-07-12:
> Still not working for me. Just to double check. The pairing button is the one with the little link symbol correct?

yes.