Koenkk/zigbee2mqtt #25595: Aqara WXKG13LM fails to publish quadruple state on action property

dergib22, 2025-01-05:
### What happened?

I have noticed that both of my Aqara WXKG13LM buttons are not publishing the property action:quadruple when they are clicked 4 times. I am able to see the actions values of  single, double, triple, quintuple, hold, release, and many when those actions are physically performed.

### What did you expect to happen?

I expect that the property action:quadruple would be published when an Aqara WXKG13LM is physically clicked 4 times.

### How to reproduce it (minimal and precise)

Subscribe to the device's MQTT topic (ex. "zigbee2mqtt/Aqara Button 8")

Click the Aqara button 4 times.

Observe that nothing was published to the device's MQTT topic.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20230507

### Adapter

Slaesh's CC2652RB stick

### Setup

Add-oon HomeAssistant OS in Rpi4

### Debug log

No log entry is generated when clicking the Aqara button 4 times.

m0wlheld, 2025-01-05:
See breaking changes at #24198

m0wlheld, 2025-01-05:
Sounds similar to #25573

dergib22, 2025-03-11:
I just checked today, and my WXKG13LM still does not report the quadruple action, but I might have a thought as to why.
First off, I don't really know what I am looking at, but what I am seeing seems to me like a possible reason why there is no quadruple. 

Looking at the lumi.ts file here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/ddb5c8da6c43f8a602ede8cdd96eb1a4ea4d73db/src/lib/lumi.ts#L2586

It seems to me that the actionLookup for all of the ["WXKG13LM", "WXKG04LM", "WXCJKG11LM", "WXCJKG12LM", "WXCJKG13LM"] models do not get assigned what I suspect would be a 4:"quadruple"

dergib22, 2025-03-17:
https://zigbee.blakadder.com/Aqara_WXKG13LM.html

I am starting to suspect that it does not have a quadruple action, which would make this an issue with the zigbee2mqtt device documentation page.

jusa-a, 2025-05-19:
Facing the same issue with my WXKG13LM mini switch. All other actions work fine except the quadruple click.

closed 2025-07-27 by github-actions[bot]

pabigot, 2026-10-15:
I tracked this down to https://github.com/Koenkk/zigbee-herdsman-converters/pull/2847#issuecomment-891134448 which adds `quintuple` and `many` but not `quadruple`, for reasons that may no longer be true.  I'd be willing to submit a PR to try to fix it, but don't have the skills to build a docker image that incorporates the updated dependency for testing.

pabigot, 2026-10-16:
A [WXKG13LM](https://www.zigbee2mqtt.io/devices/WXKG13LM.html), purchased 2026-10-10 shows up in Z2M with Firware ID Unsupported, and version Not assessed.

Using Wireshark and an nRF52840 Dongle to snoop the OTA zigbee packets  I've confirmed that events are transmitted for 1, 2, 3, 5 button presses, many (6), hold (0) and release (255).  No event is transmitted for four button presses.

So, unless/until that changes there's nothing Z2M can do to provide a `quadruple` action.