Koenkk/zigbee2mqtt #25598: Failed to add devices to group

rccoleman, 2025-01-06:
### What happened?

I seem to have gotten my database into a bad state somehow when I deleted an existing group with Group ID 1, then created a new group with the same name (got assigned Group ID 1 again), and now I can't add any devices to that group.  The error that I get whenever I try to add any of my devices to this new group is

`[2025-01-05 17:27:27] error:    z2m: Failed to add from group (DatabaseEntry with ID '161' does not exist)`

And sure enough, ID 161 is skipped/absent in my `database.db`.  The normal device before it got ID 160, and the new group got ID 162, so I don't know what it's looking for with ID 161.  Weirdly, if I add *another* group, which gets Group ID 28, it gets assigned to ID 163 and I can add devices to it just fine (including the same devices that I couldn't add to the first group).

Here's the database.db snippet:

```
{"id":160,"type":"Router","ieeeAddr":"0x282c02bfffedf9d2","nwkAddr":25458,"manufId":4659,"manufName":"Third Reality, Inc","powerSource":"Mains (single phase)","modelId":"3RSP02028BZ","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,4096,2820,1794],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acPowerDivisor":10,"acVoltageDivisor":10,"rmsVoltage":1162,"acFrequency":60,"acPowerMultiplier":1,"acVoltageMultiplier":1,"acCurrentMultiplier":1,"rmsCurrent":30,"activePower":15,"powerFactor":51}},"seMetering":{"attributes":{"unitOfMeasure":0,"divisor":3600000,"summaFormatting":255,"multiplier":1,"currentSummDelivered":3077524}},"genBasic":{"attributes":{"modelId":"3RSP02028BZ","manufacturerName":"Third Reality, Inc","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":1,"swBuildId":"v1.00.92"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c3d377","endpointID":1}],"configuredReportings":[{"cluster":1794,"attrId":0,"minRepIntval":60,"maxRepIntval":3600,"repChange":0},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":1,"swBuildId":"v1.00.92","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736126454939}
{"id":162,"type":"Group","groupID":1,"members":[],"meta":{}}
{"id":163,"type":"Group","groupID":28,"members":[{"deviceIeeeAddr":"0x282c02bfffeb3bb4","endpointID":1},{"deviceIeeeAddr":"0x282c02bfffeb4567","endpointID":1},{"deviceIeeeAddr":"0xb43a31fffe339ae3","endpointID":1}],"meta":{}}
```
I'll add debug logging in a bit, but I wanted to get this report started first.

### What did you expect to happen?

I expected to be able to add a group and populate it with devices.

### How to reproduce it (minimal and precise)

Hard to say if this is reproducible outside of my specific setup, but I did what I described above.

1. Had a group with group ID 1 that was working
2. Deleted it
3. Created a group with the same name (got ID 1)
4. Tried to add devices to it, which all fail with `z2m: Failed to add from group (DatabaseEntry with ID '161' does not exist)`
5. Create another group
6. Added devices to it without any problem.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

ZStack 20240710

### Adapter

Tubesz POE 2023 (2652P7)

### Setup

Container on an Intel N100

thueske, 2025-01-06:
https://github.com/Koenkk/zigbee2mqtt/issues/25556

thueske, 2025-01-07:
I could cause the problem by adding devices to reused group IDs (happens automatically). If I create a group with an ID that has never existed before, I have no problems.

rccoleman, 2025-01-07:
Yes, I think that's exactly the situation that I'm seeing.

ESP4Ever, 2025-01-11:
When I had problems with groups I seen missing or inconsistent data in config yaml and the database file.
If you have same issue just delete groups from these files (in yaml it shall be like this: groups: {}) and start from the scratch.

MKlitgaard2, 2025-03-10:
Same here. I can create a new group with a new id and sometimes add a single device, but after that I get the same error.

MKlitgaard2, 2025-03-25:
It works for me now, i dont know why, i removed groups from configuration.yaml and restarted.

RezzZ, 2025-03-26:
this morning I could add light 3 of 6 instantly again. its like theres a process in the back hanging and preventing more light to be added before it times out

ArbiterGR, 2025-04-05:
Commenting just to follow the issue, as i'm experiencing the same issues. A solution i found for now, is to use high numbered id groups (eg 90, 91, etc) when creating a group.

Dinth, 2025-06-30:
I know that this one has been closed, but i just set up a completely new Z2M instance, added 6 bulbs to it, without any problem or removing items. Then i have created a group, but when i try to add any of those bulbs to the group, im getting the following error:
```
info 2025-06-30 21:41:02z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_new/bridge/response/group/members/add', payload '{"data":{},"error":"Failed to add from group (DatabaseEntry with ID '8' does not exist)","status":"error","transaction":"huuk5-17"}'
error 2025-06-30 21:41:02z2m: Failed to add from group (DatabaseEntry with ID '8' does not exist)
```
My database.db config looks like this:

[image]

juahan, 2025-07-03:
I'm experiencing this as well, and looks like restarting Z2M always fixes things.

jesperholstjh, 2025-09-14:
I'm having the same issue, though a restart doesn't fix it for me.