Koenkk/zigbee2mqtt #25601: 2.0 Update is gone

GigiPompieru, 2025-01-06:
### What happened?

This morning I have noticed that the Z2M add-on no longer has an icon and the 'yellow' like above it in the add-on store showing it has an update is gone.

The 2.0 update was skipped and when going to the available updates in HA it is no longer there.

Further more, clicking Changelog next to 1.42.0-2 gives: Addon 45df7312_zigbee2mqtt does not exist in the store.

Searching for zigbee2mqtt in the addon store only brings up Proxy and Edge.

So my question is, how can I upgrade to 2.0 in this situation?

### What did you expect to happen?

The 2.0 to be present and available.

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

20240710

### Adapter

20240710

### Setup

HAOS VM

vistvanv, 2025-01-06:
got the same issues, i was getting ready for the update checking configs and modifying stuff. This morning the update prompt for z2m is missing, The icon of the add-on also changed and it is also missing from the add-on store

Trevsweb, 2025-01-06:
I'm on 2.0.0 2 and it's gone for me too. Zigbe2mqtt still working though.

GigiPompieru, 2025-01-06:
> I'm on 2.0.0 2 and it's gone for me too. Zigbe2mqtt still working though.

so the actual issue is that Z2M is gone from the add-on store of HA?

GigiPompieru, 2025-01-06:
it seems there is an issue in the code https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/pull/694
it will get fixed.

Hedin, 2025-01-06:
same issue...

Vidjer, 2025-01-06:
I update yesterday and had the yellow icon. This morning no more yellow icon and same issue

m0wlheld, 2025-01-06:
Btw… the add-on repository is https://github.com/zigbee2mqtt/hassio-zigbee2mqtt

matteos1, 2025-01-06:
also for me same issue. zigbee2mqtt not exist

GigiPompieru, 2025-01-06:
Fixed, just click refresh on the addon page.