Koenkk/zigbee2mqtt #25602: Some Errors

Alexofhiswalnut, 2025-01-06:
### What happened?

Hello, I often get these errors appearing on my system. Do you know where they might be coming from? It's the same with or without cable extension.

```
[2025-01-06 06:48:34] error: 	z2m: Publish 'set' 'state' to 'switch_thermostat_chaudiere' failed: 'Error: ZCL command 0xa4c138bff25524f7/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":61035,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":19},"zclSequence":254,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-06 07:34:59] error: 	z2m: Publish 'set' 'state' to 'Prise_Lum_Comble' failed: 'Error: ZCL command 0xa4c138298a1f45ce/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":55979,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":106},"zclSequence":185,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-06 08:23:42] error: 	z2m: Publish 'set' 'state' to 'Inter_Buandrie' failed: 'Error: ZCL command 0xa4c13884710f04fa/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":27281,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":103},"zclSequence":175,"commandIdentifier":11} timed out after 10000ms)'
```

### Zigbee2MQTT version

1.42.0-2

### Adapter firmware version

7.4.5 [GA]

### Adapter

SonOff Dongle E

### Setup

x86-64

GigiPompieru, 2025-01-06:
I have these as well, as do some other users I have talked to. I get them from light switches with no neutral wire. I don't think this is related to Z2M but to the device going offline and not listening to commands.

Commenting and subscribing to see if anyone has found a way to make it work.

Alexofhiswalnut, 2025-01-06:
They are indeed 220V powered devices, but not all are without neutral wire.

closed 2025-03-15 by github-actions[bot]