Koenkk/zigbee2mqtt #25603: advanced attributes missing in HA

ybizeul, 2025-01-06:
### What happened?

`no_occupancy_since` attribute no longer populated

### What did you expect to happen?

`no_occupancy_since` attribute populated

### How to reproduce it (minimal and precise)

Current config worked before 2.0.0 :
```
  '0x00158d00075f5912':
    friendly_name: motion_escalier_haut
    no_occupancy_since:
      - 30
      - 60
      - 90
      - 120
      - 180
    illuminance_precision: 0
    occupancy_timeout: 5
```

### Zigbee2MQTT version

2.0.0-1

### Adapter firmware version

20220219

### Adapter

zzh!

### Setup

HA Add-on

### Debug log

```
Info 2025-01-06 09:45:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":91,"occupancy":true,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:45:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"occupancy":true,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:45:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:46:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"no_occupancy_since":30,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:46:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"no_occupancy_since":60,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:47:07z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"no_occupancy_since":90,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:47:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"no_occupancy_since":120,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
Info 2025-01-06 09:48:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/motion_escalier_haut', payload '{"battery":100,"device_temperature":24,"illuminance":37,"linkquality":94,"no_occupancy_since":180,"occupancy":false,"power_outage_count":435,"temperature":27,"voltage":3145}'
```

[image]

pejotigrek, 2025-01-06:
I've also noticed, that no occupancy settings in the web-dashboard are empty:

[image]

and when I check those in the `devices.yaml`, all I can see is:
```
  [...]
  no_occupancy_since:
    - undefined: 10
    - undefined: 60
```

don't know if it helps, but maybe it is up tu something.

ybizeul, 2025-01-06:
Oh yes that’s an old one, I think no one expects that to be fixed at this point ! https://github.com/Koenkk/zigbee2mqtt/issues/18083

this issue is different though

omerfaruk-aran, 2025-01-16:
I'm having the same problem. There was no problem before 2.0, but I started experiencing this problem after the update. Device model RTCGQ01LM. I have been using this device for 2 years without any problems by following the procedure in the [link](https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806/6). The fact that I encountered this problem after 2.0 indicates that it is software related. Even though I manually edited the device.yaml file, the problem was not resolved. It automatically detects motion after 55 minutes.

```yaml
'0x00158..':
  friendly_name: MotionSensor1
  occupancy_timeout: 15
  no_occupancy_since:
    - 10
    - 60
```

Josef-Friedrich, 2025-03-03:
zigbee2mqtt 2.1.3

[image]

If I use the webinterface, the `configuration.yaml` looks like:

```yml
devices:
  '0x0017880108672368':
    no_occupancy_since:
      - undefined: 10
      - undefined: 60
```

If I set the configuration manually the messages are sent in the right interval

```yml
devices:
  '0x0017880108672368':
    no_occupancy_since:
      - 10
      - 60
```

ybizeul, 2025-03-04:
This is different and I lost all hope this will ever be fixed. https://github.com/Koenkk/zigbee2mqtt/issues/18083

Adastreia, 2025-04-27:
I still have hope, so much hope!  there not anyone that can fix this?

I have Philips Hue Motionsensors, 4 SML001 and one SML004. They are all very chatty and I suspect it could have  something to do with this. 

If the web-interface is creating this issue, is there a possibility to omit this option from the UI and we will make do with manual configuration?