Koenkk/zigbee2mqtt #25605: events triggered twice for Wireless switch (2-key) - TS0021 on 2.0.0

wjansenw, 2025-01-06:
### What happened?

Since I moved to 2.0.0, and started to use the events, I see strange behavuour:

- when zigbee2mqtt container is started, everything works as expected. Single button, double button: it works OK.
- after a while, zigbee2mqtt sends the events to HA twice

logs after a retart of the container (my button is called button_1, and I'm pressing the 2nd button once - this might make it a bit hard to read)
```
[2025-01-06 11:46:12] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_2_single","battery":0,"last_seen":"2025-01-06T10:46:12.177Z","linkquality":255}'
[2025-01-06 11:46:12] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1/action', payload 'button_2_single'
```

logs after having the container active for a while:

```
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_2_single","battery":0,"last_seen":"2025-01-06T10:45:06.453Z","linkquality":255}'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_2_single","battery":0,"last_seen":"2025-01-06T10:45:06.453Z","linkquality":255}'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1/action', payload 'button_2_single'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1/action', payload 'button_2_single'
```

### What did you expect to happen?

only a single event to be submitted to HA

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

containerized on odroid n2+

wjansenw, 2025-01-06:
Might be relevant (or not):

```
homeassistant:
  enabled: true
  experimental_event_entities: true
  legacy_action_sensor: false
  status_topic: homeassistant/status
  legacy_entity_attributes: false
  legacy_triggers: false
```

aluspl, 2025-01-06:
Thank you Dude! It helps also on ikea tardif (I couldn't update since 1.42 z2m because of double click)

ThrawnJL, 2025-01-06:
I think I am having the same problem but it does not just send the command twice, it sends it 3-4 times each time i click currently

Koenkk, 2025-01-07:
@wjansenw could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

wjansenw, 2025-01-07:
> @wjansenw could you provide the debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

I would love to - but when I do it now; I don't seem to get this behaviour... 
Anyway, I have activated debug and will monitor. I do not use the button that often and the logs have rotated due to some restarts. I will disable rotate to make sure I do not miss the logs if it happens again.

wjansenw, 2025-01-07:
I'm know wondering if it could be because of a router that is unresponsive from time to time. When I was testing, the button was close to that router. Now it is close to another router.
Will see what happens if I can test it close to the malfunctioning router.

ThrawnJL, 2025-01-08:
For me it worked to restart z2m but funny thing is I had tried it before more then once.
Now I’m just watching and see if it comes back.

wjansenw, 2025-01-08:
I grepped the button_1 and provided context of 20 lines. There are messages from other devices in the logs. If you need more - I can provide as well. At [2025-01-08 08:29:22] I see 2 times the same message being received.
At this moment, the button seems not to be linked to the router that is unresponsive sometime.

```
[2025-01-08 08:29:20] debug:    zh:controller: Received payload: clusterID=8, address=62278, groupID=0, endpoint=11, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":123,"commandIdentifier":11},"payload":{"cmdId":4,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2025-01-08 08:29:20] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lampwim', payload '{"brightness":3,"effect":null,"last_seen":"2025-01-08T07:29:20.413Z","linkquality":255,"power_on_behavior":"previous","state":"OFF","update":{"installed_version":16786690,"latest_version":16786690,"state":"idle"},"update_available":false}'
[2025-01-08 08:29:20] debug:    zh:deconz: response received (123)
[2025-01-08 08:29:20] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/lampwim', payload '{"brightness":99,"effect":null,"last_seen":"2025-01-08T07:29:20.413Z","linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"installed_version":16786690,"latest_version":16786690,"state":"idle"},"update_available":false}'
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-01-08 08:29:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-01-08 08:29:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 08:29:22] debug:    zh:deconz:driver: query aps data indication
[2025-01-08 08:29:22] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 73
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 73 srcAddr: 0x774d destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 255
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: response payload: 091706048d0104000100
[2025-01-08 08:29:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 08:29:22] debug:    zh:controller: Received payload: clusterID=61184, address=30541, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":23,"commandIdentifier":6},"payload":{"seq":36100,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"command":{"ID":6,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"activeStatusReport"}}
[2025-01-08 08:29:22] debug:    zh:controller:endpoint: ZCL command 0xa4c138b7b55d9bb2/1 manuSpecificTuya.defaultRsp({"cmdId":6,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":23,"writeUndiv":false})
[2025-01-08 08:29:22] debug:    zh:deconz: no response expected (23)
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 73 srcAddr: 0x774d destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 255
[2025-01-08 08:29:22] debug:    zh:deconz:frameparser: response payload: 091706048d0104000100
[2025-01-08 08:29:22] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 08:29:22] debug:    zh:controller: Received payload: clusterID=61184, address=30541, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":23,"commandIdentifier":6},"payload":{"seq":36100,"dpValues":[{"dp":1,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"command":{"ID":6,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"activeStatusReport"}}
[2025-01-08 08:29:22] debug:    z2m: Received Zigbee message from 'button_1', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":36100}' from endpoint 1 with groupID 0
[2025-01-08 08:29:22] debug:    z2m: Received Zigbee message from 'button_1', type 'commandActiveStatusReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":36100}' from endpoint 1 with groupID 0
[2025-01-08 08:29:22] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_1_single","battery":0,"last_seen":"2025-01-08T07:29:22.844Z","linkquality":255}'
[... cut ...]
[2025-01-08 08:29:22] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/lampwim/set' with data '{"state":"ON","brightness":3}'
[2025-01-08 08:29:22] debug:    z2m: Publishing 'set' 'brightness' to 'lampwim'
[2025-01-08 08:29:22] debug:    zh:controller:endpoint: ZCL command 0x0017880108ff96b6/11 genLevelCtrl.moveToLevelWithOnOff({"level":3,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-08 08:29:22] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0xf346 EP:11 SeqNr. 76 request id: 70
```

wjansenw, 2025-01-08:
It is also receiving messages twice not linked to button clicks (eg, this one in the night):

```
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
```

Koenkk, 2025-01-08:
> It is also receiving messages twice not linked to button clicks (eg, this one in the night):
>
> ```
> [2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type [...]

Please provide the debug log around these lines.

wjansenw, 2025-01-09:
> 
> Please provide the debug log around these lines.

```
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 6
[2025-01-08 03:23:04] debug:    zh:deconz: false, true, false, 10000
[2025-01-08 03:23:04] debug:    zh:deconz: resolve request (6)
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz:driver: query aps data indication
[2025-01-08 03:23:04] debug:    zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 206
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 206 srcAddr: 0x774d destAddr: 0x0 profile id: 0x104 cluster id: 0x0 lqi: 255
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: response payload: 080d0a01002042e2ff2036e4ff2000
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:controller: Received payload: clusterID=0, address=30541, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":13,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":66},{"attrId":65506,"dataType":32,"attrData":54},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-01-08 03:23:04] debug:    zh:controller:endpoint: ZCL command 0xa4c138b7b55d9bb2/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":13,"writeUndiv":false})
[2025-01-08 03:23:04] debug:    zh:deconz: no response expected (13)
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 206 srcAddr: 0x774d destAddr: 0x0 profile id: 0x104 cluster id: 0x0 lqi: 255
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: response payload: 080d0a01002042e2ff2036e4ff2000
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:controller: Received payload: clusterID=0, address=30541, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":13,"commandIdentifier":10},"payload":[{"attrId":1,"dataType":32,"attrData":66},{"attrId":65506,"dataType":32,"attrData":54},{"attrId":65508,"dataType":32,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
[2025-01-08 03:23:04] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":null,"battery":0,"last_seen":"2025-01-08T02:23:04.641Z","linkquality":255}'
[2025-01-08 03:23:04] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":null,"battery":0,"last_seen":"2025-01-08T02:23:04.641Z","linkquality":255}'
[2025-01-08 03:23:04] debug:    zh:deconz:driver: DATA_REQUEST - destAddr: 0x774d EP:1 SeqNr. 207 request id: 108
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 108 status: 0
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DATA_REQUEST RESPONSE - request id: 108 status: 0
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10100110
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...]
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DATA_CONFIRM RESPONSE - destAddr: 0x774d request id: 108 confirm status: 0
[2025-01-08 03:23:04] debug:    zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-08 03:23:04] debug:    zh:deconz: sendZclFrameToEndpoint - message send with transSeq Nr.: 13
[2025-01-08 03:23:04] debug:    zh:deconz: false, true, false, 10000
[2025-01-08 03:23:04] debug:    zh:deconz: resolve request (13)
[2025-01-08 03:23:04] debug:    zh:deconz:frameparser: DEVICE_STATE changed: 10101010
```

Koenkk, 2025-01-10:
Could you provide a screenshot of the z2m frontend -> device -> about page?

wjansenw, 2025-01-10:
See attached screenshot:

wjansenw, 2025-01-10:
Hope this helps. Thanks for checking!

[image]

Koenkk, 2025-01-11:
I'm checking your debug logs again now but I don't see the double action in there, your debug logs:

```
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
[2025-01-08 03:23:04] debug:    z2m: Received Zigbee message from 'button_1', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":66}' from endpoint 1 with groupID 0
[2025-01-08 03:23:04] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":null,"battery":0,"last_seen":"2025-01-08T02:23:04.641Z","linkquality":255}'
[2025-01-08 03:23:04] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":null,"battery":0,"last_seen":"2025-01-08T02:23:04.641Z","linkquality":255}'
```

(note that action is `null`)

I need the debug log of:

```
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_2_single","battery":0,"last_seen":"2025-01-06T10:45:06.453Z","linkquality":255}'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload '{"action":"button_2_single","battery":0,"last_seen":"2025-01-06T10:45:06.453Z","linkquality":255}'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1/action', payload 'button_2_single'
[2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1/action', payload 'button_2_single'
```

jjhelin, 2025-01-12:
I have this same exact issue, upgraded yesterday. However, restarting z2m has no bearing on the issue, ~duplicated/repeated messages occur always. Just like above, the repeated messages are not identical:

`
[2025-01-12 10:35:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/study_switch', payload '{"action":"single","battery":100,"last_seen":"2025-01-12T10:35:23+02:00","linkquality":123,"voltage":3100}'
[2025-01-12 10:35:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/study_switch/action', payload 'single'
`

Grafana Loki aggregation indicated that these are logged at the same millisecond.

It seems that the second message to action subtopic is a "derived" one. However, I have the recommended, vanilla 2.0.0 configuration, where the migration dropped all legacy setting (that were always disabled).

A curious result of this repeating is that my Home Assistant 2024.12.5 executes an MQTT device trigger automation three times, whereas z2m 1.42.0 executed it once (as it's supposed to). I would have expected a double execution (instead of triple), so it will be interesting to check how this changes once the repeated MQTT message are eliminated.

wjansenw, 2025-01-12:
> I need the debug log of:
>
> ```
> [2025-01-06 11:45:06] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/button_1', payload [...]

Aren't they in my previous comment, with different timestamp?

wjansenw, 2025-01-13:
@Koenkk - unless I do not understand which logs you need, I think I provided all. If you need more context around the logs, let me know.

Logs around action: https://github.com/Koenkk/zigbee2mqtt/issues/25605#issuecomment-2577018149
Logs around null-action: https://github.com/Koenkk/zigbee2mqtt/issues/25605#issuecomment-2579497957

Note: it started to behave correctly after a ~10 minutes power outage; still acting correct after 12hrs. So the issue *might* be gone. Will check back in if it appear again.

Koenkk, 2025-01-13:
I must be blind, sorry! Fixed the issue.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

wjansenw, 2025-01-15:
Thanks @Koenkk, I moved to the dev-branch yesterday evening (CEST). I will monitor, as the issue can take (from experience) 24-48hrs before it first appears.

m00n, 2025-01-15:
I'm having the same problem using the :latest-dev Docker image, but with a [Phillips Hue Tap Dial](https://www.zigbee2mqtt.io/devices/8719514440937_8719514440999.html)

etakpte, 2025-01-15:
Hi,

I have the same with IKEA Rodret E2201.

```
[2025-01-15 03:23:33] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":46475,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":12983149,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,3,1]}}
[2025-01-15 03:23:33] debug: 	zh:controller: Received payload: clusterID=6, address=46475, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":3,"commandIdentifier":1},"payload":{},"command":{"ID":1,"parameters":[],"name":"on"}}
[2025-01-15 03:23:33] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-15 03:23:33] debug: 	z2m: Received Zigbee message from 'Nappali-rodret', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
[2025-01-15 03:23:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Nappali-rodret', payload '{"action":"on","battery":100,"identify":null,"linkquality":116,"update":{"installed_version":16777303,"latest_version":16777303,"state":"idle"}}'
[2025-01-15 03:23:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Nappali-rodret', payload '{"action":"","battery":100,"identify":null,"linkquality":116,"update":{"installed_version":16777303,"latest_version":16777303,"state":"idle"}}'
[2025-01-15 03:23:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Nappali-rodret/action', payload 'on'
[2025-01-15 03:23:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Nappali-rodret/action', payload 'on'
```

Koenkk, 2025-01-19:
@etakpte could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

etakpte, 2025-01-19:
> [@etakpte](https://github.com/etakpte) could you provide the debug log of this?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Sure. There is one such evet at [2025-01-15 18:43:05]

[debug.txt](https://github.com/user-attachments/files/18468869/debug.txt)

Koenkk, 2025-01-19:
@etakpte does this also happen after disabling the `legacy_action_sensor` in your `configuration.yaml`?

etakpte, 2025-01-19:
> [@etakpte](https://github.com/etakpte) does this also happen after disabling the `legacy_action_sensor` in your `configuration.yaml`?

Hi,

At the moment it does not happen at all. (i still have   legacy_action_sensor: true)

I was experimenting with the filtered_cache attribute, but now it works with or without the attribute.
I suppose z2m restart which fixed it.

wjansenw, 2025-01-27:
@Koenkk - just updating here that I did not experience the issue anymore since moving to the dev-branch. For me this can be closed - not sure about the others in this thread.

closed 2025-01-28 by Koenkk