Koenkk/zigbee2mqtt #25606: Aqara Smart rotary knob H1 - ZNXNKG02LM - Error input cluster manuSpecificLumi

TDCroPower, 2025-01-06:
### What happened?

Every time I operate the dimmer, I get the following error message in the log...

```
[2025-01-06 11:06:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Schlafzimmer_Aqara_Dimmer_Tisch', payload '{"action":"single","action_rotation_angle":null,"action_rotation_angle_speed":null,"action_rotation_button_state":null,"action_rotation_percent":null,"action_rotation_percent_speed":null,"action_rotation_time":null,"battery":null,"last_seen":"2025-01-06T11:06:27+01:00","linkquality":216,"operation_mode":null,"sensitivity":null,"voltage":null}'
[2025-01-06 11:06:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Schlafzimmer_Aqara_Dimmer_Tisch/action', payload 'single'
[2025-01-06 11:06:27] error: 	z2m: Failed to configure 'Schlafzimmer_Aqara_Dimmer_Tisch', attempt 2 (Error: Device 0x54ef441000a05424 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20241127 - 7.4.5 [GA]

### Adapter

ember

### Setup

Z2M: 2.0.0 as Docker Container

TDCroPower, 2025-01-06:
it also appears without operate with the dimmer...
```
[2025-01-06 11:10:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kinderzimmer_Aqara_Dimmer', payload '{"action_rotation_angle":null,"action_rotation_angle_speed":null,"action_rotation_button_state":null,"action_rotation_percent":null,"action_rotation_percent_speed":null,"action_rotation_time":null,"battery":null,"last_seen":"2025-01-06T11:10:39+01:00","linkquality":236,"operation_mode":null,"sensitivity":null,"voltage":null}'
[2025-01-06 11:10:40] error: 	z2m: Failed to configure 'Kinderzimmer_Aqara_Dimmer', attempt 2 (Error: Device 0x54ef4410009f1af2 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:59:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:144:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/modernExtend.ts:207:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:27)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

TDCroPower, 2025-01-10:
The problem seems to have solved itself after a while and no longer reports any errors in the log.

closed 2025-01-10 by TDCroPower

hrustiks, 2025-01-16:
This error occurs once when restarting Z2M

kkqq9320, 2025-02-14:
I have the same problem. I get an error message within a minute to an hour after z2m starts.
```
[2025-02-14 06:23:48] info: 	z2m: Zigbee2MQTT started!
[2025-02-14 07:05:55] info: 	z2m: Configuring 'Knob Aqara'
[2025-02-14 07:05:56] error: 	z2m: Failed to configure 'Knob Aqara', attempt 1 (Error: Device 0x54ef441000a057d7 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:61:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:146:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:209:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:27)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-02-14 07:05:56] info: 	z2m: Configuring 'Knob Aqara'
[2025-02-14 07:05:56] error: 	z2m: Failed to configure 'Knob Aqara', attempt 2 (Error: Device 0x54ef441000a057d7 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:61:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:146:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:209:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:27)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
[2025-02-14 07:05:57] info: 	z2m: Configuring 'Knob Aqara'
[2025-02-14 07:05:57] error: 	z2m: Failed to configure 'Knob Aqara', attempt 3 (Error: Device 0x54ef441000a057d7 has no input cluster manuSpecificLumi
    at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:61:15)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:146:55)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/modernExtend.ts:209:23)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:27)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

hrustiks, 2025-02-14:
I have the same problem too!