Koenkk/zigbee2mqtt #25608: [Addon] Issue with repository - Zigbee2MQTT missing after 2.0.0 update

FARKIr, 2025-01-06:
### What happened?

After updating to Zigbee2MQTT 2.0.0, the addon shows warning "Add-on Zigbee2MQTT has been removed from the repository". Clicking "Submit" removes the addon completely. The addon disappeared from the repository but installation is still running.

### What did you expect to happen?

The addon should stay in the repository and update process should complete normally without removing the addon.

### How to reproduce it (minimal and precise)

Update to Home Assistant 2025.1.0
See warning about Zigbee2MQTT being removed from repository
DO NOT click Submit as it uninstalls the addon

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20250103.0

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (zstack)

### Setup

Home Assistant Addon

### Debug log

Repair issue
Warning ⸱ Reported by Home Assistant Supervisor

Add-on Zigbee2MQTT has been removed from the repository it was installed from. This means it will not get updates, and backups may not be restored correctly as the supervisor may not be able to build/download the resources required.

Selecting Submit will uninstall this deprecated add-on. Alternatively, you can check [Home Assistant help](https://www.home-assistant.io/help/) and the [community forum](https
[image]
://community.home-assistant.io/) for alternatives to migrate to.

m0wlheld, 2025-01-06:
#25601 

https://community.home-assistant.io/t/add-on-zigbee2mqtt-has-been-removed-from-the-repository-it-was-installed-from/822983

Koenkk, 2025-01-06:
Should be good now!

closed 2025-01-06 by Koenkk