Koenkk/zigbee2mqtt #25609: Xiaomi Model GZCGQ01LM not working after 2.0 update

KurtKeunen, 2025-01-06:
### What happened?

After update i can still see the device and its measurements in Z2M but it is not pushed to mqtt or HA

error 2025-01-06 13:03:03z2m: Failed to configure '0x54ef4410000a323f', attempt 1 (Error: Bind 0x54ef4410000a323f/1 msIlluminanceMeasurement from '0x00124b0021b77b25/1' failed (AREQ - ZDO - bindRsp after 10000ms) at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23) at func (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:423:47) at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZdoInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:428:43) at ZStackAdapter.sendZdo (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:305:16) at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:562:30) at Object.bind (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/lib/reporting.ts:42:9) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/lumi.ts:3003:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:21) at Configure.configure (/app/lib/extension/configure.ts:124:13))

### What did you expect to happen?

the values to be pushed as entities to HA

### How to reproduce it (minimal and precise)

update to 2.0

### Zigbee2MQTT version

2.0

### Adapter firmware version

latest

### Adapter

Electrolama zig-a-zig-ah

### Setup

addon in HA

mrmaximas, 2025-01-06:
i have two GZCGQ01LM works good. please repair one for example.

zajaaczki, 2025-01-07:
Same issue for me GZCGQ01LM but this is old bug. Only "illuminance_lux" entity is available in Home Assistant, second "illuminance" entity not exist in HA. That work OK with "illuminance_lux" in z2m 1.42 after update to 2.0 "illuminance_lux" no longer support and "illuminance" not exist in HA only battery entity left.   

1.42
{
    "battery": 100,
    "illuminance": 20645,
    "illuminance_lux": 116,
    "linkquality": 87,
    "power_outage_count": 17054,
    "voltage": 3209
}

2.0
{
    "battery": 100,
    "illuminance": 128,
    "linkquality": 87,
    "power_outage_count": 17054,
    "voltage": 3207
}

iamelec, 2025-01-07:
I have the exact same issue. Replaced the battery, repaired it to Zigbee2mqtt. It's updates it's parameters in zigbee2mqtt but it doesn't show any entities in Home Assistant.

mrmaximas, 2025-01-07:
Dear friends! Please read this carefully https://github.com/Koenkk/zigbee2mqtt/discussions/24198

KurtKeunen, 2025-01-08:
what fixed it for me was as described on previous comment

add to Z2M configuration.yaml

serial:
  adapter: zstack 

advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

removing and readding the sensor made it work after 5 minutes 
Hope this tl;dr is useful for someone :)

closed 2025-01-08 by KurtKeunen