Koenkk/zigbee2mqtt #25614: E1524/E1810 Trafri remotes no longer work in 2.0

arjenanker, 2025-01-06:
### What happened?

After installing the 2.0 update, all my remotes no longer worked in Home Assistant. Looking at the Z2M logs, nothing happens when I push a button on a remote. Nothing in the logs whatsoever.

Trying to re-pair a remote ends up in the remote not being able to finish the interview with "Error: Interview failed because we can not get active endpoints".

I had to downgrade back to my backup of 1.42.0-2. Remotes worked again, pairing the remote that couldn't complete the interview worked straight away too.

### What did you expect to happen?

I expect my remotes to work like they worked in the previous version.

### How to reproduce it (minimal and precise)

Update to 2.0 and try to use a E1524 remote or try and pair a E1524 remote.

### Zigbee2MQTT version

2.0

### Adapter firmware version

20201026

### Adapter

Texas Instruments LAUNCHXL-CC26X2R1

### Setup

Addon on HA OS VM on Proxmox

m0wlheld, 2025-01-06:
Could be related to breaking changes #24198

arjenanker, 2025-01-06:
Well, all those remotes use *_action triggers. However, the z2m logs showed absolutely nothing when I pushed buttons on those remotes, which seems odd to me. Shouldn't the logs at least say something was triggered?

And, I still can't pair them in 2.0, which seems a seperate issue.

Unfortunately, I can't test any of this since I downgraded since the Z2M addon was removed so I can't update again until it was resolved.

Jopy2104, 2025-01-06:
Having the exact same problem. Can no longer remotely control anything anymore, becasue the _action triggers are all gone from Home Assistant. I can see the actions in Zigbee2MQTT.

eastvan, 2025-01-06:
> Well, all those remotes use *_action triggers. However, the z2m logs showed absolutely nothing when I pushed buttons on those remotes, which seems odd to me. Shouldn't the logs at least say something was triggered?
> 
> And, I still can't pair them in 2.0, which seems a separate issue.
> 
> Unfortunately, I can't test any of this since I downgraded since the Z2M addon was removed so I can't update again until that is resolved.

I can confirm I have the same issue after updating Z2M to 2.0.0-2. 
I am aware of the breaking changes described in #24198, and I was able to modify the existing automations after upgrading to 2.0.0-1 and the remote was working correctly for a day until I moved to 2.0.0 - 2.
As you correctly noted, there are no events in the logs when I press the buttons. The battery is 100%, tried to reset the device a few times without any luck.

eastvan, 2025-01-06:
No events whatsoever. I enabled "permit join" and try to reset the remote. The log entries related to that

[2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' joined
[2025-01-06 13:16:35] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' accepted by handler
[2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' is already in database with different network address, updating network address
[2025-01-06 13:16:35] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-01-06 13:16:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Remote Control', payload '{"battery":90,"identify":null,"last_seen":"2025-01-06T13:16:35-08:00","linkquality":10,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'
[2025-01-06 13:16:35] debug: 	zh:controller: Not interviewing '0x588e81fffeb4e59b', completed 'true', in progress 'false'

No events when I press any buttons after that
I removed the device completely from Z2M and try to join it. These are the log entries 

2025-01-06 13:23:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"hcvl1-4"}'
info 2025-01-06 13:23:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:23:52-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
info 2025-01-06 13:23:52zh:controller: Interview for '0x588e81fffeb4e59b' started
info 2025-01-06 13:23:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:23:52-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
info 2025-01-06 13:23:52z2m: Device '0x588e81fffeb4e59b' joined
info 2025-01-06 13:23:52z2m: Starting interview of '0x588e81fffeb4e59b'
info 2025-01-06 13:23:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b"},"type":"device_joined"}'
info 2025-01-06 13:23:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b","status":"started"},"type":"device_interview"}'
info 2025-01-06 13:23:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138b4b3838a19', payload '{"battery":88,"humidity":43.75,"last_seen":"2025-01-06T13:23:55-08:00","linkquality":109,"temperature":22.37,"voltage":2900}'
info 2025-01-06 13:24:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:24:12-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
info 2025-01-06 13:24:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b"},"type":"device_announce"}'
error 2025-01-06 13:24:12zh:controller: Interview failed for '0x588e81fffeb4e59b with error 'Error: Interview failed because can not get active endpoints ('0x588e81fffeb4e59b')'
error 2025-01-06 13:24:12z2m: Failed to interview '0x588e81fffeb4e59b', device has not successfully been paired

soostdijck, 2025-01-06:
Same issue here, when I look in z2m itself, under the "state" tab for the device I do see it registering presses. (Make sure you press different buttons, because the same button keeps sending the same things, which hides the change).
The info doesn't seem to make it into HASS though.

The logs in the z2m plugin inside HASS shows these logs:
```
[2025-01-06 22:53:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ikea puck woonkamer', payload '{"action":"toggle","battery":90,"identify":null,"linkquality":43,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"}}'
[2025-01-06 22:53:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ikea puck woonkamer/action', payload 'toggle'
[2025-01-06 22:53:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ikea puck woonkamer', payload '{"action":"toggle","battery":90,"identify":null,"linkquality":43,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"}}'
[2025-01-06 22:53:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/ikea puck woonkamer/action', payload 'toggle'
```

The "logbook" inside HASS shows nothing.

lensherm, 2025-01-07:
I can get it to work like so:

[image]

For a list of all available payloads, see here:
[https://www.zigbee2mqtt.io/devices/E1524_E1810.html](url)

Fo3niks, 2025-01-07:
@soostdijck have you activated the legacy triggers option?

See: https://github.com/Koenkk/zigbee2mqtt/issues/25631#issuecomment-2574110162

arjenanker, 2025-01-07:
> No events whatsoever. I enabled "permit join" and try to reset the remote. The log entries related to that
> 
> ```
> [2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' joined
> [2025-01-06 13:16:35] debug: 	zh:zstack:unpi:parser: --- parseNext []
> [2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' accepted by handler
> [2025-01-06 13:16:35] debug: 	zh:controller: Device '0x588e81fffeb4e59b' is already in database with different network address, updating network address
> [2025-01-06 13:16:35] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
> [2025-01-06 13:16:35] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Remote Control', payload '{"battery":90,"identify":null,"last_seen":"2025-01-06T13:16:35-08:00","linkquality":10,"update":{"installed_version":604241925,"latest_version":604241925,"state":"idle"},"update_available":false}'
> [2025-01-06 13:16:35] debug: 	zh:controller: Not interviewing '0x588e81fffeb4e59b', completed 'true', in progress 'false'
> ```
> 
> No events when I press any buttons after that I removed the device completely from Z2M and try to join it. I can no longer join the remote as per below
> 
> ```
> 2025-01-06 13:23:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254},"status":"ok","transaction":"hcvl1-4"}'
> info 2025-01-06 13:23:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:23:52-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
> info 2025-01-06 13:23:52zh:controller: Interview for '0x588e81fffeb4e59b' started
> info 2025-01-06 13:23:52z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:23:52-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
> info 2025-01-06 13:23:52z2m: Device '0x588e81fffeb4e59b' joined
> info 2025-01-06 13:23:52z2m: Starting interview of '0x588e81fffeb4e59b'
> info 2025-01-06 13:23:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b"},"type":"device_joined"}'
> info 2025-01-06 13:23:53z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b","status":"started"},"type":"device_interview"}'
> info 2025-01-06 13:23:55z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138b4b3838a19', payload '{"battery":88,"humidity":43.75,"last_seen":"2025-01-06T13:23:55-08:00","linkquality":109,"temperature":22.37,"voltage":2900}'
> info 2025-01-06 13:24:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x588e81fffeb4e59b', payload '{"last_seen":"2025-01-06T13:24:12-08:00","linkquality":112,"update":{"installed_version":-1,"latest_version":-1}}'
> info 2025-01-06 13:24:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffeb4e59b","ieee_address":"0x588e81fffeb4e59b"},"type":"device_announce"}'
> error 2025-01-06 13:24:12zh:controller: Interview failed for '0x588e81fffeb4e59b with error 'Error: Interview failed because can not get active endpoints ('0x588e81fffeb4e59b')'
> error 2025-01-06 13:24:12z2m: Failed to interview '0x588e81fffeb4e59b', device has not successfully been paired
> ```

Had this exact thing. The remote can't be paired again, had to restore my backup of 1.42 to get my remote to pair again.

Fo3niks, 2025-01-07:
> I suppose the awesome blueprints would need to be reworked in order to be functional with Z2M > 2.0.0

There's a [fork](https://github.com/lsismeiro/awesome-ha-blueprints) that is already working on it. Now that the original creator EPMatt is back there are apparently efforts underway to merge those fixes back into the main repository.

https://github.com/EPMatt/awesome-ha-blueprints/issues/622#issuecomment-2574824759

Hessenpower01, 2025-01-07:
Same here, Remote no longer working under 2.0.0.2, unable to re pair. Downgraded to 1.42 and now working again.

Koenkk, 2025-01-08:
Could you provide the debug log when triggering actions on the device with z2m 2.0.0?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

kalatos86, 2025-01-08:
Same problem here with all the zigbee buttons which are using *_action. After switching Home Assistant legacy action sensors It started to work again. I am wondering why that change and how long this legacy option will be available?

dogmatic69, 2025-01-08:
I have the same problem. I got it working briefly with rejoining the button. I then updated the firmware on the button and it completely died. 

Now Pressing buttons gives no logs in zigbee2mqtt logs. Forcing a rejoin starts but then ultimately fails with the same error about missing endpoints.

dogmatic69, 2025-01-08:
Unbelievable. I've force removed and repaired this button 20 times or more over past 2-3 days. 

Posted the comment above and gave it one more try. It worked. 🤦‍♂️ 

I can now turn my lights on 🎉

bartek75, 2025-01-12:
so, i have the similar/same problem.
After upgrade to: 2.0.0-2
My IKEA remotes stopped working.
It is visible in the system, In HA i can see the new events as triggers, but pressing buttons does not trigger anything.
Me remote is named 'Pilot IKEA' in the log,
In the log i can see:
[2025-01-12 16:58:06] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Pilot IKEA', payload '{"battery":74,"identify":null,"last_seen":"2025-01-12T16:58:06+01:00","linkquality":12,"update":{"installed_version":587728433,"latest_version":604241925,"state":"available"},"update_available":true}'
[2025-01-12 16:58:06] [34mdebug[39m: 	z2m: Received Zigbee message from 'Pilot IKEA', type 'commandIdentifyQuery', cluster 'genIdentify', data '{}' from endpoint 1 with groupID 0
[2025-01-12 16:58:06] [34mdebug[39m: 	z2m: No converter available for 'E1524/E1810' with cluster 'genIdentify' and type 'commandIdentifyQuery' and data '{}'
[45df7312_zigbee2mqtt_2025-01-12T16-00-29.968Z.log](https://github.com/user-attachments/files/18390103/45df7312_zigbee2mqtt_2025-01-12T16-00-29.968Z.log)

dogmatic69, 2025-01-13:
that is not the same problem at all. You need to read the breaking changes notes. This issue is around no events firing at all.

dogmatic69, 2025-01-13:
Based on what you said @bartek75 

> In the log i can see:
> [2025-01-12 16:58:06] �[32minfo�[39m: z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Pilot IKEA', payload [...]

Where as the original ticket states there is **nothing**

> Looking at the Z2M logs, nothing happens when I push a button on a remote. Nothing in the logs whatsoever.

bartek75, 2025-01-14:
so i have made the IKEA remotes work. In my case i had to do:
1. Remove IKEA remote from z2m
2. Restart HA
3. Add remote (allow join and pair remote)
4. Adding was unsuccessful (no endpoints received from device)
5. Force remove again the remote.
6. Add remote again. This time was successful.
7. All buttons work!

arjenanker, 2025-01-17:
I just updated to 2.0.0-2 again.  With "Home Assistant legacy action sensors" enabled in the Z2M sensors. And this time... my remotes seem to work. Did have to restart Home Assistant to get anything to work at all. (And a "an addon was removed from the store" repair thing promptly removed the entire addon even though 1.42 wasn't even installed anymore, so had to do a restore and another update, but that's another matter)

Hessenpower01, 2025-02-04:
Hey all, has somebody with this problem updated to 2.1.0.1. and can you report if it's working again? Thanks

Hessenpower01, 2025-02-10:
Any news on this in release 2.1.1.1?

Bond246, 2025-02-21:
Hello to this round,
i just updated to z2m some weeks ago to 2.0 and also updated it to 2.1 latest (whatever).
All my Ikea buttons where working still fine.

But today i had to change the battery of my E1524 and after that it doesn't reappear.
So i tried re-pairing without deleting.
And i restarted and repaired the button several times with force deletion. 

Nothing.
z2m says that the interview was successful. 
But no button action is provided after that. 

My Coordinator is a conbee (first generation).
But i tried pairing with a ikea-bulb, a innr plug and the coordinator itself.
The button has to do something completely outside of zigbee so binding it directly to a device is not an option for me.

Thanks!

sevorl, 2025-03-17:
I am also waiting for this fix. The _action triggers  are important to be able to use ikea remotes

Koenkk, 2025-03-18:
They should still work, could you provide the debug log when triggering clicking a button and you see no `action` being published?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

AndreKR, 2025-03-19:
Nothing shows up in the debug log and now the devices have pretty much stopped working at all (occasionally the "Last seen" value is updated, but most of the time it is not.

I did some sniffing and when I start pairing on the device (push the button 4 times) I see this:

[image]

That seems like it could be an issue?

AndreKR, 2025-03-19:
Hm, this may or may not have been a red herring. I repeated the pairing and now it successfully paired and for now the actions work. I'm somewhat expecting them to fail in a few days again.

AndreKR, 2025-03-19:
Ok, something is definitely amiss here. I power cycled the remote control (took it off the power supply and put a battery in) and then it stopped working again. I tried re-pairing it and now I'm getting the "PAN full" thing with this one as well.

AndreKR, 2025-03-19:
Ok, something is definitely amiss here. I power cycled the device (took it off the power supply and put a battery in) and the action didn't work anymore. I tried re-pairing it and again got the "PAN full" thing.

Hessenpower01, 2025-03-20:
Thanks @AndreKR for the input: @Koenkk can you work with that? May we offer more info?

Koenkk, 2025-03-20:
@AndreKR 

- Does the pan full also happen when pairing it directly through the coordinator? (you can enable permit join in the frontend for just the coordinator)
- It seems the power cycle makes the device loses it config, could you try to click a button on the remote and immediately after that press the yellow re-configure button and see if it works again?

Bond246, 2025-03-20:
Hello,

i want to add some logs from my setup because what is see is a little different but with the same result.
My setup is based on a conbee stick first generation. This button was the very first one i ever had even before using z2m. It was never perfect but it did a good job at all.

[2025-03-20.22-38-03.log](https://github.com/user-attachments/files/19376267/2025-03-20.22-38-03.log)

I started the logfile just some lines above where i deleted the device from my network and then enabled joining on my coordinator. 
Device-Address is: `0x90fd9ffffe530e81`
But my friendly name is `Schlafzimmer - Bett Button`

I tried to pair the device several times "soft" without deleting it which doesn't worked.
Then i deleted the device (force delete) and paired it to the coordinator. 
Right after that i startet to reconfigure it and while i did that i pushed the button every 1-2s to keep it awaik. 

The only thing i really see while pressing the button is a lot of stuff like that
```
[2025-03-20 22:51:52] debug: 	zh:deconz:driver: DATA_INDICATION - sending read data request - SeqNr. 192
[2025-03-20 22:51:52] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - seqNr. 192 srcAddr: 0xbb70 destAddr: 0xfffd profile id: 0x0 cluster id: 0x0 lqi: 255
[2025-03-20 22:51:52] debug: 	zh:deconz:frameparser: response payload: f256b503ffff2e21000000
[2025-03-20 22:51:52] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-20 22:51:52] debug: 	zh:deconz:frameparser: DATA_INDICATION RESPONSE - Error: Unsupported response reading for cluster ID '0'.
[2025-03-20 22:51:53] debug: 	zh:deconz:driver: send read device state from queue. seqNr: 193
[2025-03-20 22:51:53] debug: 	zh:deconz:frameparser: device state: 10100010
[2025-03-20 22:51:53] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-20 22:51:55] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10101010
[2025-03-20 22:51:55] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
[2025-03-20 22:51:55] debug: 	zh:deconz:driver: query aps data indication
```

Maybe it helps or maybe something with the setup and that conbee stick is bad.

Thanks!

Bond246, 2025-03-25:
Hello,

meanwhile i had the chance to test the same button of a friend but with an older firmware -> exact the same behavior.
And he tested also my button but with another coordinator. He uses the sonoff stick with a TI-chip and was able to get all action of that button without problems.

So i think the problem depends on the deconz / conbee sticks.

sevorl, 2025-03-25:
I use a SLZB-07 Zigbee 3.0 and the action does not trigger anymore with
zigbee2mqtt 2.x.
It works fine with v1.42
So the reason is not the Conbee stick.

|

sevorl, 2025-03-27:
I found the issue for my case. 
I use controllerX and had to adjust my config now

AndreKR, 2025-04-03:
It's annoying me as well. I will investigate more, I just need to bring a computer with the sniffer stick downstairs to my coordinator.

Bond246, 2025-04-05:
At least i can say none of the ikea buttons (will) work anymore.
I bought a RODRET E2201 today more or less just for fun and it has the exact same problem. No action command is exposed through the network.

For the moment i'm waiting for a SLZB-06 to re-configure my complete network on these instead of a conbee.

AndreKR, 2025-04-05:
@Koenkk So, I paired the E1524 directly with the coordinator and for the moment they work. So it seems I'm not really affected by any E1524 issue.

I will now bring them upstairs again, out of range of the coordinator, without resetting them. We'll see what happens.

The STYRBAR one (detected as E2001/E2002) still doesn't work, even when pairing directly with the coordinator. Actually I'm not sure if it's really pairing, it doesn't seem to send anything. It's hard to identify in all the Zigbee traffic around the coordinator. When I started fiddling with it, the "last seen" value updated once and now doesn't do that anymore. The light on the remote control flashes red as if it's pairing, but it doesn't change anything.

That said, I only have a single STYRBAR remote, so this particular one might simply be broken.

Sorry I couldn't be more helpful for the moment.

AndreKR, 2025-04-05:
I'm gonna keep all further discussion of that STYRBAR remote out of this thread. I have opened https://github.com/Koenkk/zigbee2mqtt/issues/27007 to store my observations about the STYRBAR, then this issue here can be solely about the E1524.

AndreKR, 2025-04-06:
@Koenkk I brought the remote control upstairs and here it doesn't work. I did not reset or re-pair it.

If I filter for its network address, this is what I see:

[image]

Each key press generates a couple of these packets. You can see from the timestamps, which packets belong to a single keypress.

If I do not filter, this is what I see (for a single keypress):

[image]

The network addresses in there belong to:
0xe4b9 - the remote control in question  
0xbc11 - the light bulb above me
0x89ae - another light bulb downstairs, closer to the coordinator
0xec1b - doesn't appear in Z2M

By the way, I paired the remote with "Permit join (Coordinator)" as you suggested, but once it has joined it should still be able to roam around the network, right?

One more thing...

Here's again a view filtered by the network address of the remote:

[image]

See that packet in the lower middle, where it says "ZigBee ZDP"? At the same time when that packet appeared, Z2M published a message from the device:
```
[2025-04-06 00:05:22] info:   z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC Top floor Fridge', payload '{"action":null,"battery":34,"identify":null,"last_seen":"2025-04-06T00:05:22.906Z","linkquality":153,"update":{"installed_version":304170354,"latest_version":604241925,"state":"available"}}'
[2025-04-06 00:05:22] info:   z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC Top floor Fridge', payload '{"action":null,"battery":34,"identify":null,"last_seen":"2025-04-06T00:05:22.948Z","linkquality":153,"update":{"installed_version":304170354,"latest_version":604241925,"state":"available"}}'
[2025-04-06 00:05:23] info:   z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/RC Top floor Fridge', payload '{"action":null,"battery":34,"identify":null,"last_seen":"2025-04-06T00:05:23.112Z","linkquality":153,"update":{"installed_version":304170354,"latest_version":604241925,"state":"available"}}'
```

It's not a keypress (action) message, but it is a message and the "last seen" time was updated.

I hope any of that is helpful for you.

Koenkk, 2025-04-06:
Can you send the sniff to me? You can use [this form](https://form.jotform.com/233385867866072) to share a sniff with me.

AndreKR, 2025-04-15:
I didn't keep that file, but I made a new one and submitted it.

There are two button presses in the file. If you set the time format to "Seconds since first captured packet" the first one is at 31.601985 and the second one is at 134.600397, they both start with "Beacon request". The network address of the remote control is 0xE4B9.

By the way, you can also borrow my remote control if you don't own this model. I think IKEA doesn't make them anymore unfortunately. Send me an email if you're interested.

Koenkk, 2025-04-15:
Could you make a sniff what happens when permitting joining and pressing the button?

AndreKR, 2025-04-16:
Uploaded.

In this dump the "Permit join" starts at 9.669706, the first keypress at 25.046264 and the second keypress at 80.031003. There is a third keypress, but I didn't record the exact time, I think around second 120.

Note that the second and third keypress were actually recognized by Z2M, the first one still failed.

Koenkk, 2025-04-16:
It seems the device now properly re-joins after which the button presses show up (I see one at 121.15):

[image]

I've noticed that on rejoin it gets a PAN full:

[image]

Questions:
- What kind of device is 0xe4b9?
- After how much time do the button presses fail again? Any chance you could make a sniff from a working situation until it fails again?

AndreKR, 2025-04-16:
The device seems to re-join only when "Permit join" is on, which kind of seems related, but how is it supposed to work if regular re-joins are required but of course we should not keep "Permit join" on all the time.

0xe4b9 is the remote control itself. The device that said "PAN full" (0x5984) is a Zigbee speaker, something like [this](https://www.aliexpress.com/item/1005003693237953.html)).

I'd say it take about a day to go from working to not working, but I will try if I can find out more. Unfortunately it might very well be that frequent testing prevents the issue. I will try to find out more. If it turns out to be a couple of hours I can probably sniff the whole time. If it's more than a day, that's a bit impractical. Let's see.

bjornfor, 2025-06-17:
[Bad bot](https://drewdevault.com/2021/10/26/stalebot.html).

AndreKR, 2025-07-26:
The issue hasn't happened for me anymore.

At first I had removed a questionable device (the one that says "PAN full" in the captures above) which made the buttons work again after re-pairing. The re-pairing was necessary because I had reset them before.

I have since added that questionable device back and the buttons _still_ work.

I still think there might be something wrong if a misbehaving router can prevent devices from pairing, but that doesn't seem specific to those buttons.

jareware, 2025-08-14:
Update to https://github.com/Koenkk/zigbee2mqtt/issues/25614#issuecomment-3185464393: after letting them sit for like 15 minutes or so, they eventually got their interview status, and have (so far 🤞) worked again!