Koenkk/zigbee2mqtt #25616: Failed to start zigbee-herdsman after update to version 2.0.0-2

koenraadhaedens, 2025-01-06:
### What happened?

[14:17:23] INFO: Preparing to start...
[14:17:23] INFO: Socat not enabled
[14:17:23] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-06 14:17:27] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-06 14:17:27] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-06 14:17:27] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-06 14:17:27] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-06 14:17:27] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-06 14:17:27] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-06 14:17:27] error: 	z2m: Exiting...
[2025-01-06 14:17:27] error: 	z2m: Error: Cannot discover TCP adapters at this time. Specify valid 'adapter' and 'port' in your configuration.
    at findTCPAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:470:15)
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:496:26)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:54)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:38)
    at Zigbee.start (/app/lib/zigbee.ts:69:47)
    at Controller.start (/app/lib/controller.ts:142:13)

Using UZG-01 as controller

my config

data_path: /config/zigbee2mqtt_1
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt_1
  user: xxxxxxx
  password: xxxxxxxxx
serial:
  port: tcp://192.168.0.37:6638
  baudrate: 115200
advanced:
  transmit_power: 20

### How to reproduce it (minimal and precise)

after restore to version 1.42.0-2 from backup all good again

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240519

### Adapter

UZG-01 on ESP32-D0WD-V3

### Setup

HASIO 2024.12.5

Impedanz, 2025-01-06:
Same here with: SLZB-06P7

koenraadhaedens, 2025-01-06:
by the way i have 2 zigbee2mqtt installations on my hassio and the one with the USB dongle was update with no issues and the one with the network zigbee controller has the issue

m0wlheld, 2025-01-06:
See #24198, add `adapter` to your `serial` section

Impedanz, 2025-01-06:
adapter: zstack (add this to yaml in z2m konfig, works directly for me)

benstegink, 2025-01-06:
adding adapter: zstack didn't work for me :( It worked with 2.0.0-1 but isnt' working with 2.0.0-2

melhiore, 2025-01-06:
fixed! Thank you everyone for help.

What was done:

cd /root/homeassistant/zigbee2mqtt
ls
there is:
configuration.yaml

vi configuration.yaml
search for the section "serial"

in my case it looked like that:

serial:
   port: tcp://192.168.102.203:6638

add line right under this:
adapter: zstack

so your serial section looks like that:

serial:
   port: tcp://192.168.102.203:6638
   adapter: zstack

save and quit. Update to 2.0.0-2 as normal

Koenkk, 2025-01-06:
This is one of the Zigbee2MQTT 2.0.0 breaking changes, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364 on how to fix this.

closed 2025-01-06 by Koenkk

koenraadhaedens, 2025-01-07:
This definitely works. I had missed it earlier.