Koenkk/zigbee2mqtt #25619: Zigbee2MQTT 2.0.0-2 and SLZB-06 (Ethernet) zigbee coordinator

melhiore, 2025-01-06:
### What happened?

Currently running 1.42.0-2. Everything works as expected. No issues with managing devices. Upgrading to 2.0.0-1 disabled my zigbee network (restored from full backup taken just minutes before upgrade). Upgrade to 2.0.0-2 seem to be the same issue. I am getting pop-up window stating that add-on ddi not start, do you want o start, clicking "start add-on" does nothing.

### What did you expect to happen?

Zigbee2MQTT normal start after the software update

### How to reproduce it (minimal and precise)

HaOS on Dell Optiplex mini PC (enterprise grade device)
SLZB-06 (Ethernet) zigbee coordinator

Performing software update from 1.14.0-2 to 2.0.0-1 or 2.0.0-2 disables zigbee network

### Zigbee2MQTT version

1.14.0-2

### Adapter firmware version

Firmware version: 20240710; v2.6.8.dev26

### Adapter

SLZB-06 (Ethernet) zigbee coordinator

### Setup

HaOS on Dell Optiplex, no Docker, Generic x86-64

peanut-hu, 2025-01-06:
Same here, log shows this, after installing 2.00-2 from 1.42.0-2.
LOG:
Starting Zigbee2MQTT without watchdog.
[2025-01-06 15:33:43] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-06 15:33:43] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-06 15:33:43] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-06 15:33:43] error: 	z2m: Exiting...
[2025-01-06 15:33:43] error: 	z2m: Error: Adapter 'auto' does not exists, possible options: deconz, ember, ezsp, zstack, zboss, zigate
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:80:19)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
    
    
    When reverting to 1.42.0-2 everything is back to normal.

benstegink, 2025-01-06:
same issue with me too. I had it working with the manual fix with 2.0.0-1, with 2.0.0-2 the add-on keeps crashing and manual fix of adding  adapter: zstack doesn't even seem to work anymore.

clausardal, 2025-01-06:
I have the same problem but just with a SLZB-05

I have tried rebooting EVERYTHING but no luck. I noticed though that my SLZB-05 is not getting an IP address. Under the IP in my router it says "unknown". But it can still see the MAC address. Are others experiencing the same thing with the IP address?

melhiore, 2025-01-06:
fixed! Thank you everyone for help.

What was done:

go to TERMINAL:

cd /root/homeassistant/zigbee2mqtt
ls
there is:
configuration.yaml

vi configuration.yaml
search for the section "serial"

in my case it looked like that:

serial:
port: tcp://192.168.102.203:6638

add line right under this:
adapter: zstack

so your serial section looks like that:

serial:
port: tcp://192.168.102.203:6638
adapter: zstack

save and quit. Update to 2.0.0-2 as normal

melhiore, 2025-01-06:
Closing as resolved

closed 2025-01-06 by melhiore

bco34370, 2025-03-04:
Hi All,
I pretty new here and have since a couple of days the same issue. Connection to SLZB-06 is not working anymore.
I searched the net, found numerous proposed solutions, but none seems to work for me. 
Here's my config:

data_path: /homeassistant/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://localhost:1883
  user: addons
  password: sunoojiaNg4thaeGee8aequeic2Mu6aeheideenguzeBaik2Eifoh8the4eeloo1
serial:
  adapter: zstack
  port: tcp://slzb.local:6638
  baudrate: 115200
  disable_led: false
advanced:
  transmit_power: 20
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

I tried changing the "mqtt://localhost:1883" to "core:mosquito:1833" without success.
I tried changing the slzb.local:6638 to the ip adres of the dongle, without success. And yes, a ping to the slzb.local works. 
I added a couple of lines about legacy stuff that I found in a comment about a slzb-05 not working. Same result.

Here's an excerpt of the log when starting (trying to at least)

[17:08:50] INFO: Preparing to start...
[17:08:50] INFO: Socat not enabled
[17:08:50] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-03-04 17:08:51] info: 	z2m: Logging to console, file (filename: log.log)
[2025-03-04 17:08:51] info: 	z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown)
[2025-03-04 17:08:51] info: 	z2m: Starting zigbee-herdsman (3.2.7)
[2025-03-04 17:08:51] info: 	zh:zstack:znp: Opening TCP socket with slzb.local:6638
[2025-03-04 17:08:51] info: 	zh:zstack:znp: Socket connected
[2025-03-04 17:08:51] info: 	zh:zstack:znp: Socket ready
[2025-03-04 17:08:51] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-03-04 17:08:52] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-03-04 17:09:10] error: 	z2m: Error while starting zigbee-herdsman
[2025-03-04 17:09:10] error: 	z2m: Failed to start zigbee-herdsman
[2025-03-04 17:09:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-03-04 17:09:10] error: 	z2m: Exiting...
[2025-03-04 17:09:10] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)

And obviously all this result in a 502: bad gateway.

Any help would reallly, really be very much appreciated.
TIA,
Bruno

DoWenzl94, 2025-03-06:
> Hi All, I pretty new here and have since a couple of days the same issue. Connection to SLZB-06 is not working anymore. I searched the net, found numerous proposed solutions, but none seems to work for me. Here's my config:
> 
> data_path: /homeassistant/zigbee2mqtt socat: enabled: false master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777 slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5 options: "-d -d" log: false mqtt: server: mqtt://localhost:1883 user: addons password: sunoojiaNg4thaeGee8aequeic2Mu6aeheideenguzeBaik2Eifoh8the4eeloo1 serial: adapter: zstack port: tcp://slzb.local:6638 baudrate: 115200 disable_led: false advanced: transmit_power: 20 homeassistant_legacy_entity_attributes: false homeassistant_legacy_triggers: false legacy_api: false legacy_availability_payload: false device_options: legacy: false
> 
> I tried changing the "mqtt://localhost:1883" to "core:mosquito:1833" without success. I tried changing the slzb.local:6638 to the ip adres of the dongle, without success. And yes, a ping to the slzb.local works. I added a couple of lines about legacy stuff that I found in a comment about a slzb-05 not working. Same result.
> 
> Here's an excerpt of the log when starting (trying to at least)
> 
> [17:08:50] INFO: Preparing to start... [17:08:50] INFO: Socat not enabled [17:08:50] INFO: Starting Zigbee2MQTT... Starting Zigbee2MQTT without watchdog. [2025-03-04 17:08:51] info: z2m: Logging to console, file (filename: log.log) [2025-03-04 17:08:51] info: z2m: Starting Zigbee2MQTT version 2.1.3 (commit #unknown) [2025-03-04 17:08:51] info: z2m: Starting zigbee-herdsman (3.2.7) [2025-03-04 17:08:51] info: zh:zstack:znp: Opening TCP socket with slzb.local:6638 [2025-03-04 17:08:51] info: zh:zstack:znp: Socket connected [2025-03-04 17:08:51] info: zh:zstack:znp: Socket ready [2025-03-04 17:08:51] info: zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload [2025-03-04 17:08:52] info: zh:zstack:znp: Skip bootloader for CC2652/CC1352 [2025-03-04 17:09:10] error: z2m: Error while starting zigbee-herdsman [2025-03-04 17:09:10] error: z2m: Failed to start zigbee-herdsman [2025-03-04 17:09:10] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions [2025-03-04 17:09:10] error: z2m: Exiting... [2025-03-04 17:09:10] error: z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms) at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:113:27) at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29) at Zigbee.start (/app/lib/zigbee.ts:69:27) at Controller.start (/app/lib/controller.ts:142:13) at start (/app/index.js:161:5)
> 
> And obviously all this result in a 502: bad gateway. Edit: Running CORE 2025.2.5, Supervisor 2025.03.0, OS 14.2, Zigbee2MQTT 2.1.3.1.
> 
> Any help would reallly, really be very much appreciated. TIA, Bruno

Same Problem here, when I re-flash the zigbee firmware on my SLZB-06(p7) it works for a few days and than It crashes again.