Koenkk/zigbee2mqtt #25620: ZGP not working if Join not active, then join active ZGP communication works

IntoTheWebNet, 2025-01-06:
### What happened?

After Upgrade to 2.0, permanent Join was gone, which i had activated all the time. I don't know if ZPG was working with deactive Join before version 2.0.

Now i noticed that my ZPG devices not working anymore if join is off. the i switch join on, ZGP communication works as expected.
As ZGP "translator" i'm using a IKEA TRADFRI Driver 30W (i assume).

### What did you expect to happen?

I expect working ZPG

### How to reproduce it (minimal and precise)

Switch on Join, ZGP devices are working.
Switch off Join, ZGP devices are not working.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add on on Home Assistant, running on a virtual maschine on Proxmox

m0wlheld, 2025-01-07:
Whats is „ZGP“?

IntoTheWebNet, 2025-01-07:
ZGP = Zigbee Green Power
https://de.wikipedia.org/wiki/ZigBee#ZigBee_Green_Power_(ZGP)

thomas-schwyn, 2025-01-08:
I am facing this problem as well. 
I run the NabuCasa skyconnect adapter and the Device is PTM 215Z.

chris-1243, 2025-01-13:
ZGP devices works perfectly well and without `permit join` set to ON.

I would remove your faulty device from your network and factory reset it. Then, I would strictly follow the documentation on how to pair it.

```
To pair it, hold the corresponding button for that channel for 7 seconds or more.

Once the device is paired you need to confirm the channel. To do this press A1 and B0 together. Important: don't press any other buttons between this and the pairing.
```

https://www.zigbee2mqtt.io/devices/PTM_215Z.html

https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

The second link will provide very useful information. Take the time to read it.

Basti189, 2025-01-24:
Same issue sinsce dez 2023
https://forum.iobroker.net/topic/71463/ptm-215z-funktioniert-nur-w%C3%A4hrend-des-joining

HWE70, 2025-02-09:
In my case the Problem depends on the used coordinator (Z2M 2.1.1)

- 5 year old Conbee I works without a problem
- Conbee III (most current FW 0x26530900) pairing works, but that's it. Even the channel confirmation when pressing A1 and B0 is not visible in the Z2M log and the PTM is of course not working at all afterwards
- UZG-01 (most current FW 20241001 and CC2652P7 FW 20240710) even worse. Even pairing does not work, no single log entry.

I really wonder how an extremely old coordinator works ok, but most current coordinators which should definitely work, do not. By the way, not ZGP devices work ok in any setup. I also tried different channels an Pan IDs. The results are always the same.

HWE70, 2025-02-09:
OK, UZG-01 does not seem to support green power devices without an external GP proxy. However, Conbee 3 should work, but it doesn't. When setting the Z2M log level to debug the following error is logged whenever pressing a PTM key:

[2025-02-09 17:23:23] debug: 	zh:deconz:frameparser: GP commissioning notification
[2025-02-09 17:23:23] debug: 	zh:deconz:frameparser: GREEN_POWER INDICATION - RangeError: Offset is outside the bounds of the DataView  

This issue was already raised a while a go https://github.com/Koenkk/zigbee2mqtt/discussions/23674

chris-1243, 2025-02-09:
> OK, UZG-01 does not seem to support green power devices without an external GP proxy.

It is expected. This is why there is a need of an external GP proxy.

On the other hand, Silabs chip are able to pair a ZGP device without any GP proxy.

HWE70, 2025-02-09:
> On the other hand, Silabs chip are able to pair a ZGP device without any GP proxy.

I know, I already ordered one. Fingers crossed, that thinks works with the new stick.

chris-1243, 2025-02-09:
It should. I have a Sonoff Dongle-E (EFR32MG21) in a test network and it is able to pair a ZGP debice directly.

Cc2652P7 is a good chip as well. I would have keep it and use a proxy to translate. It depends on your use case as well

HWE70, 2025-02-10:
> It should. I have a Sonoff Dongle-E (EFR32MG21) in a test network and it is able to pair a ZGP debice directly.
> 
> Cc2652P7 is a good chip as well. I would have keep it and use a proxy to translate. It depends on your use case as well

Sonoff Dongle-E arrived today. Works like a charm with the PTM switches (FW ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl)  :-)

Nerivec, 2025-04-06:
Latest-dev includes some more GP cleanup that might help shed some light on this behavior.
From MaxwellJK's logs, it appears the Ikea proxy is replying to the pairing command with `INSUFFICIENT_SPACE`, which would indicate the proxy table of the Ikea proxy is full (or is of size zero).

If you guys could provide some new `debug` logs with "latest latest-dev" of fresh pairings (remove & reset device, restart Z2M, pair device):
- via a GP proxy (permit join on it only)
- via a GP proxy (permit join on it only) after factory-resetting said GP proxy (for Ikea that should be: turn on, then toggle quickly 6 times)
- via the coordinator (permit join on it only)

CC: @fapgomes @MiAutomations

MiAutomations, 2025-04-09:
> f fresh pairings (remove & reset device, restart Z2M, pair device):

I can try share some more logs and information that you need, but first i need to know how can I use the "latest latest-dev" is there any way to test this without impact me Z2M instance that I have running ? 

Thanks

MaxwellJK, 2025-04-12:
> I can try share some more logs and information that you need, but first i need to know how can I use the "latest latest-dev" is there any way to test this without impact me Z2M instance that I have running ?
> 
> Thanks

@MiAutomations `latest-dev` is the docker image tag you have to use to download the latest dev version of the code (assuming you use docker to run your z2m instance)
Otherwise you would need to use the dev branch to execute z2m the way you usually do

CommanderROR9, 2025-04-14:
I just bought one of those Moes self-powering switches and found that it doesn't work properly with Zigbee2mqtt. I should probably have researched this earlier. From what I found here and in the moes specific discussion, there doesn't appear to be official support or even a workaround at this point, right?

MaxwellJK, 2025-04-14:
> I just bought one of those Moes self-powering switches and found that it doesn't work properly with Zigbee2mqtt. I should probably have researched this earlier. From what I found here and in the [...]

did you check [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/19405)? There has been a lot of work done over the last few weeks and it is working for some of us

CommanderROR9, 2025-04-14:
> did you check [this issue](https://github.com/Koenkk/zigbee2mqtt/issues/19405)? There has been a lot of work done over the last few weeks and it is working for some of us

Thanks for pointing that out.
I actually arrived here after reading through most of that thread. So far I haven't found anything over there that worked for me, but I can try going through it again. I tried pairing the Moes switch directly via one of the Tradfri lamps, but that didn't change anything. It pairs fine, then shows that it is unsupported and stops working after pairing mode ends.
I also tried Z2Mqtt "edge" version of the addon in Home Assistant, but that also changed nothing for me.

Nerivec, 2025-04-14:
The Moes will always show as unsupported, because the device doesn't respect the GP spec (non-static source ID), we can't explicitly identify it and have to rely on the auto-generated definition for it to work (which Z2M does not consider "supported"). _That doesn't mean the device doesn't work though._

As for the permit join issue, it's not strictly linked to the Moes. Seems something is not quite right in one of the ZigBee stacks out there, but we have yet to identify that problem more specifically...

If someone is able to sniff the pairing and triggering (with permit on and off) with this issue, that would be great!
https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

choovick, 2025-04-22:
> The Moes will always show as unsupported, because the device doesn't respect the GP spec (non-static source ID), we can't explicitly identify it and have to rely on the auto-generated definition for [...]

@Nerivec 

Is support limitation mentioned here applies to Tuya device here? https://github.com/Koenkk/zigbee2mqtt/issues/27182#issuecomment-2821180745

Should I even attempt a PR (still have to learn things)?

I get that device ID change on every paring, but model and vendors are always the same...
```
    model: 'GreenPower_2',
    vendor: '',
```

Nerivec, 2025-04-22:
If the device uses random source ID, yes, it's the same limitations (model/vendor is not enough to identify the device apart from others for GP).
The device should work though, the auto-generated definition should create a base to react to actions, assuming you're not hit by the permit join troubles...

See https://github.com/Koenkk/zigbee2mqtt/issues/19405#issuecomment-2749278303 if you want some ideas to customize the actions with an external converter (if you don't want the generated mapping that uses the official GP action names).

choovick, 2025-04-22:
Understood... sucks, but I can live with it. Auto-generated definition did not work thou, steps outlined in this comment that worked for me: https://github.com/Koenkk/zigbee2mqtt/issues/27182#issuecomment-2821180745

https://github.com/Koenkk/zigbee2mqtt/issues/27182#issuecomment-2821180745

I don't run dev build thou... just latest HA-Addon build version.

Nerivec, 2025-04-22:
Strange, it's pretty much what the auto-generated definition does... except it wouldn't be `press_1`, 2, 3, it would be the generic GP command names.
_You will have problems with that definition though, because it's not identifying specifically enough. You will not be able to use any other GP device properly (that's what the auto-generated definition fixes: it uses the source ID specific to the current pairing, and since it's re-generated if you re-pair the device, it keeps on working)._
Looks to be an exact copy of the Moes, so the thread I linked above applies the same.

MiAutomations, 2025-05-08:
Hello to all 

I'm using the last Z2M version [2.3.0] , but I still only able to use this device when the "pairing mode" Permit join is running ...

Is there any way to fix this issue? 

Thank you and best regards

tiagores, 2025-06-09:
I am also experiencing the problem with a ZGP device  - MOES switch - where I can only see the state when Join is active. 
I have a Sonof ZBDongle P, I even flashed the latest firmware 20240710, and Z2M 2.4.0 version.
Any hints?
Thanks in advance.

CommanderROR9, 2025-06-09:
> I am also experiencing the problem with a ZGP device - MOES switch - where I can only see the state when Join is active. I have a Sonof ZBDongle P, I even flashed the latest firmware 20240710, and Z2M 2.4.0 version. Any hints? Thanks in advance.

Sadly this seems to be a dead end. I haven't heard anything about this being worked on or any progress being made. My Moes ZGP Switch is now back in "storage" and I will check back wvery few Months to see if anything has changed.

choovick, 2025-06-11:
@tiagores 
**OfTopic**
Managed to get it working without re-paring, official step did not work for it, it was not seamless . Had to flash old IEEE address onto new dongle (as secondary) and modify database.db 1st line to re-match eeeAddr, of old network.
coordinator_backup.json also had to be adjusted to network settings of the SkyConnect. bit painful, but worked.

toxuin, 2025-06-18:
Is that a software limitation or a hardware quirk? Really not looking forward to moving to a whole new coordinator 😅 
Facing the same issue: ZT-LP-ZEU2S-WH-MS with TI coordinator, paired through a single IKEA device, works fine only when in pairing mode, and doesn't work at all when in normal mode.
Can we maybe have a band-aid of adding a semi-always-pairing mode where a pairing mode is always active with a certain device but will refuse to actually pair devices? 😆

tylwright, 2025-11-17:
Please keep open. Thanks!

chris-1243, 2026-07-30:
In order to avoid latencies, you should always pair your PTM through a proxy like a light bulb or a smart plug. As you are using `zstack`, it is not possible to pair a ZGP device directly to the coordinator unfortunately.

> Failure default response: clusterID=33 cmdId=1 status=INSUFFICIENT_SPACE

The memory table of the devices tested was full and they could not accept more information. Try to wait some hours (a full day would be better) and hope those devices will clean the memory. Pair again your PTM via those plugs and see how it goes.

CherAlban, 2026-08-17:
I have two EnOcean PTM 215Z which worked fine for more than two years, but suddenly both show the same "Only while permit join is enabled, all actions work correctly" behavior.

Home Assistant
Zigbee2MQTT 2.13.0
SMLIGHT SLZB-06, firmware v3.3.5.dev1
CC2652P with ZStack
Radio firmware 20260311

I don't know exactly when this started, I applied all updates when available.