Koenkk/zigbee2mqtt #25622: Z2M starts, but no frontend, no module data through MQTT ( only online/offline info)

kalon33, 2025-01-06:
### What happened?

Z2M starts, lists my adapter and outputs all modules names in info log. In info log, Z2M status is also output to MQTT, and I see it though a MQTT explorer. As debug, it also starts to output data from modules, as expected. But frontend doesn't start, and modules output are not sent to MQTT. It happened suddenly, as the first time it happened after a couple of days, and now everytime.

### What did you expect to happen?

MQTT data flows correctly, and frontend is started.

### How to reproduce it (minimal and precise)

No idea. It happened suddenly after a few days, and I can't restart it properly now.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Dockerised, with network_mode: host

### Debug log

[log.log](https://github.com/user-attachments/files/18321135/log.log)

m0wlheld, 2025-01-06:
Are you talking about Zigbee2MQTT or the Home Assistant Zigbee2MQTT Add-On? For the later, there are similar issues posted, e.g. https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/688#issue-2768872324

scubieman, 2025-01-06:
I had same issue... See this...

https://github.com/Koenkk/zigbee2mqtt/discussions/24198
https://github.com/Koenkk/zigbee2mqtt/discussions/24364

kalon33, 2025-01-06:
> Are you talking about Zigbee2MQTT or the Home Assistant Zigbee2MQTT Add-On? For the later, there are similar issues posted, e.g. [zigbee2mqtt/hassio-zigbee2mqtt#688 (comment)](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/688#issue-2768872324)

Zigbee2MQTT, I have a dockerised HA install, so no add-ons.

kalon33, 2025-01-06:
> I had same issue... See this...
> 
> #24198 #24364

I already have
```
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_06d493f96345ed11a792c88f0a86e0b4-if00-port0
  adapter: zstack
```
In my configuration.yaml and no legacy option.

If you exclude devices and passwords/tokens, I have that:  
```
mqtt:
  server: mqtt://127.0.0.1:1883
  user: jeedom
  password: [redacted]
  base_topic: zigbee2mqtt2
  include_device_information: true
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_06d493f96345ed11a792c88f0a86e0b4-if00-port0
  adapter: zstack
frontend:
  enabled: true
  port: 8080
  host: 192.168.0.2
  auth_token: [redacted]
advanced:
  log_level: debug
  channel: 25
  ext_pan_id:
[redacted]
  transmit_power: 12
devices:
[a lot of devices]
ota:
  zigbee_ota_override_index_location: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
availability:
  enabled: true
homeassistant:
  enabled: true
version: 4
```

kalon33, 2025-01-06:
> A new version 2.0.0.2 is released that seems to fix this.

For the HA add-on, but I don't think that is relevant for standalone setup in Docker. There's no new release in github releases nor a new Docker image.

scubieman, 2025-01-06:
> > I had same issue... See this...
> > #24198 #24364
> 
> I already have
> 
> ```
> serial:
>   port: >-
>     /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_06d493f96345ed11a792c88f0a86e0b4-if00-port0
>   adapter: zstack
> ```
> 
> In my configuration.yaml and no legacy option.
> 
> If you exclude devices and passwords/tokens, I have that:
> 
> ```
> mqtt:
>   server: mqtt://127.0.0.1:1883
>   user: jeedom
>   password: [redacted]
>   base_topic: zigbee2mqtt2
>   include_device_information: true
> serial:
>   port: >-
>     /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_06d493f96345ed11a792c88f0a86e0b4-if00-port0
>   adapter: zstack
> frontend:
>   enabled: true
>   port: 8080
>   host: 192.168.0.2
>   auth_token: [redacted]
> advanced:
>   log_level: debug
>   channel: 25
>   ext_pan_id:
> [redacted]
>   pan_id: [redacted]
>   transmit_power: 12
> devices:
> [a lot of devices]
> ota:
>   zigbee_ota_override_index_location: https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
> availability:
>   enabled: true
> homeassistant:
>   enabled: true
> version: 4
> ```

I am not seeing.... advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false

kalon33, 2025-01-06:
> I am not seeing.... advanced: homeassistant_legacy_entity_attributes: false homeassistant_legacy_triggers: false legacy_api: false legacy_availability_payload: false

Because file has already been migrated by Z2M to version 4 as indicated by the `version` key. But that doesn't changes anything to add these keys and restart.

m0wlheld, 2025-01-06:
Is the frontend's `host` IP correct?

kalon33, 2025-01-06:
> Is the frontend's `host` IP correct?

yes

mundschenk-at, 2025-01-06:
What do the logs say?

philippemezzadri, 2025-01-07:
> What do the logs say?

They don't say much actually to tell us what's happening.
Here is some logs : [log.log](https://github.com/user-attachments/files/18328672/log.log)
Very similar to the ones from @kalon33.

giedriuskilcauskas, 2025-01-07:
I have exactly same issue with addon. In debug logs it looks like zigbee data coming in, but frontend doesn't start and data doesn't end up in HA.

mundschenk-at, 2025-01-07:
@philippemezzadri According to those logs, you are not running Z2M 2.0.0 but 1.42.0 (the previous release).

philippemezzadri, 2025-01-07:
@mundschenk-at 
Sorry,  i uploaded the wrong log file .
This is the one:  [log.log](https://github.com/user-attachments/files/18335983/log.log)

As you can see, line 919 i tried to switch on a light from HA, and then we can see some errors:

```
[2025-01-07 16:52:30] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Entrée/set' with data '{"state":"ON"}'
[2025-01-07 16:52:34] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xb4e3f9fffedf2b0a:43263/1 (0,3,1)
[2025-01-07 16:52:34] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":43263,"destendpoint":1,"srcendpoint":1,"clusterid":513,"transid":27,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,87,11,10,0]}}
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,255,168,1,1,1,2,27,0,30,5,24,87,11,10,0,48]
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-07 16:52:34] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,233,1,27,52]
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,233,1,27,52]
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [233,1,27] - 52
[2025-01-07 16:52:34] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":233,"endpoint":1,"transid":27}
[2025-01-07 16:52:34] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-07 16:52:34] debug: 	zh:zstack: Data confirm error (0xb4e3f9fffedf2b0a:43263,233,3)
[2025-01-07 16:52:34] debug: 	zh:zstack: Wait 2000ms
[2025-01-07 16:52:37] error: 	z2m: Failed to call 'Frontend' 'stop' (TypeError: Cannot read properties of undefined (reading 'close')
    at /opt/zigbee2mqtt/lib/extension/frontend.ts:132:52
    at new Promise (<anonymous>)
    at Frontend.stop (/opt/zigbee2mqtt/lib/extension/frontend.ts:132:15)
    at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:368:17)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:233:9)
    at stop (/opt/zigbee2mqtt/index.js:172:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:180:13))
[2025-01-07 16:52:37] debug: 	z2m: Saving state to file /opt/zigbee2mqtt/data/state.json
[2025-01-07 16:52:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-01-07 16:52:37] info: 	z2m: Disconnecting from MQTT server
```

TanerCRB, 2025-01-08:
Hi 
I have the same issue in the HA addon it seems that communication between the coordinator and devices is working but frontend is not starting to get 502 bad gateway, I was struggling with this problem for more than 2 days and after the coordinator FW updated to the news one from Koenkk it starts to work for about 24hours but today I have the same issue, it seems that after zigbe2mqtt update I did not do the restart of HA and it works but yesterday I did update because of adding new ESP device and after this restart zigbee2mqqt stops working and I can not make it work again without downgrading to 1.42.0-2

After downgrading to 1.42.0-2 everything is working fine. 

My configuration: 
```
data_path: /config/zigbee2mqtt 
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  user: mqtt_user
  password: [seecret]
  server: mqtt://192.168.8.164:1883
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_464ac0dd0560ec11bd312c5f25bfaa52-if00-port0
  adapter: zstack
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
```
The log file 
[45df7312_zigbee2mqtt_2025-01-08T08-30-55.202Z.log](https://github.com/user-attachments/files/18344433/45df7312_zigbee2mqtt_2025-01-08T08-30-55.202Z.log)

Koenkk, 2025-01-09:
https://github.com/Koenkk/zigbee2mqtt/pull/25693 may have fixed this, could you check?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

MSL-DA, 2025-01-09:
> #25693 may have fixed this, could you check?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I have tested and the problem is solved

antst, 2025-01-10:
> > #25693 may have fixed this, could you check?
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.
> 
> I have tested and the problem is solved

Do you happen to have some TRVs or some other devices which have time internally?

antst, 2025-01-10:
> #25693 may have fixed this, could you check?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

When can we expect new release of Z2M, so this fix will get into proper HA add-on, which is not based on dev branch?

chefkoch, 2025-01-10:
> #25693 may have fixed this, could you check?
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I had same issues described here by other users using my Dongle-E (ember) and [switching to dev ](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge#updating-the-edge-add-on) resolved the problem. Thanks for the fix and all the hard work and support theses days with the v2 migration.

philippemezzadri, 2025-01-10:
> Do you happen to have some TRVs or some other devices which have time internally?

fix tested and the problem is solved.
I have 6 Danfoss TRVs.

indi81, 2025-01-10:
> > Do you happen to have some TRVs or some other devices which have time internally?
> 
> fix tested and the problem is solved. I have 6 Danfoss TRVs.

I also have 4 Danfoss TRV, model eTRV0103.

MrAdam, 2025-01-22:
Had the same issue, and also have (7) Danfoss Ally TRV's.
Moved to development build, where issue is gone and I can use Z2M again.

closed 2025-02-02 by Nerivec